milesburton/Arduino-Temperature-Control-Library #27: Does not work on low temperatures, always show -127C

Nalorokk, 2015-02-09:


milesburton, 2015-02-09:
Can you confirm which version (commit) you are using from Git. This was actually recently resolved

Nalorokk, 2015-02-09:
Still exists for me in latest commit 146cfdd2d7

RobTillaart, 2015-07-04:
@Nalorokk 
Does this problem still exist? otherwise this issue can be closed.

milesburton, 2015-07-08:
I believe this is actually still outstanding. https://github.com/milesburton/Arduino-Temperature-Control-Library/tree/tommi-tuovila-temperature-fix resolved the issue but I don't believe it was ever confirmed by the author so I opted to avoid merging.

Are you able to test this in your setup and let us know if it works. If it does, it'll be quick merge

milesburton, 2015-09-23:
Closing, assuming not a problem

closed 2015-09-23 by milesburton

rubberduck203, 2017-10-17:
I know it's an old issue, but I'm actually experiencing this problem with a MAX31820.
Does anyone remember the cause or solution?

rubberduck203, 2017-10-17:
This happens if you wire VDD and GND backwards.
The pinout in the datasheet for the MAX31820 is wrong.

[image]