milesburton/Arduino-Temperature-Control-Library #29: Alarm bytes unnecessarily restricted

fbtoc, 2015-02-13:
In the methods setHighAlarmTemp(const uint8_t* deviceAddress, char celsius) and
setLowAlarmTemp(const uint8_t* deviceAddress, char celsius), celsius is restricted by the if statement:

if (celsius > 125) celsius = 125;
else if (celsius < -55) celsius = -55; 

Per the data sheet, the alarm temp is formed by the concatenation of the High and Low Alarm bytes so the per byte test doesn't accomplish the stated goal.

Also, per the data sheet, if you aren't using the alarm function the two bytes are fully available for user use.  In my case, I am using them for shorthand device id's.

My vote is to remove the if statements, ask users to make sure the concatenation of bytes represents a valid temperature if they are using the alarm function, and not enforce any constraints if the bytes are for user data.

RobTillaart, 2015-02-13:
Interesting thought to use the alarm bytes for other purposes, however I disagree with your proposal as it breaks current behaviour. Limiting the alarm temperature to valid values makes quite some sense.

To implement your idea I propose add two new methods to use the two bytes as user data

```C
void setUserData(const uint8_t* deviceAddress, int16_t data);
int16_t setUserData(const uint8_t* deviceAddress);
```

Documentation should make clear that these reuse the alarm bytes,

opinion?

fbtoc, 2015-02-14:
Rob,
      
      Adding the two new methods is the way to go.  For the user data
      type I'd go with unsigned.  I definitely like treating the data as
      a 16 bit rather than dealing with a high and low byte.
      
      I re-read the data sheet and now realize that TL and TH encode a
      low and high temperature limit.  I somehow had morphed that fact
      into believing that TH was the high byte of the alarm temperature
      and TL the low byte just as it is for reading current temperatures
      and was therefore confused as to how the current code could be
      useful - my bad.  
      
      BTW,  the library works great, thanks for creating/maintaining it.
      
      I did have one issue with the library semantics that's clearly not
      worth changing but might be worth mentioning in an example.  I'm
      doing a project where the number of sensors might change
      dynamically, and with code that's sort of like the tester
      example.  From what I see in the examples and based on the name, I
      naively put the begin method in my setup code and temperature
      retrieval code in a forever loop, not  appreciating that to the
      extent I was using device indices, even with a call to
      getDeviceCount, that wasn't going to play well with adding devices
      unless I rebooted.  Easy enough to figure out what was going on
      but certainly mysterious for a while.  The point being you don't
      want to call begin as a part of initialization and forget it,  you
      really want to call it whenever the number of devices changes (if
      you're using device indexes of course).  At least that's the
      picture I see now.  Point me in the right direction if I'm
      misssing something.
      
      Fred
      
      
      
    
    On 2/13/2015 11:24 AM, Rob Tillaart
      wrote:
    
    
      Interesting thought to use the alarm bytes for other purposes,
        however I disagree with your proposal as it breaks current
        behaviour. Limiting the alarm temperature to valid values makes
        quite some sense.
      To implement your idea I propose add two new methods to use the
        two bytes as user data
      
        void setUserData(const uint8_t* deviceAddress, int16_t data);
int16_t setUserData(const uint8_t* deviceAddress);
      
      Documentation should make clear that these reuse the alarm
        bytes,
      opinion?
      —
          it on GitHub.

RobTillaart, 2015-02-14:
"  I did have one issue with the library semantics that's clearly not
      worth changing but might be worth mentioning in an example. "
please create separate issue for that, otherwise things get mixed up, lost, ...etc.

RobTillaart, 2015-07-04:
Created PR #33

milesburton, 2015-07-08:
PR requested ready. Please reopen this issue if you still see this problem once the merge is complete

closed 2015-07-08 by milesburton