milesburton/Arduino-Temperature-Control-Library #31: How many DS18B20 ?

kluczyk11, 2015-06-30:
Hi,

When I connect  ten  DS18B20  , Adruino is getting strange readings. The readings  are changening randomly for a second. Exapmle 23,4 degree in one room  to -127 degree, and back to  23,4. After  this the next DS18B20 do the same in another room . The readings are not changenign at the same moment on all DS18B20. Readings are changening room by room randomly.  When I disconnect  tenth DS18B20, everything  is going back to normal. I have checked:
- every  DS18B20 one by one   - works 
-  try tree different power supply with ten DS18B20 - problem remains

I is working fine only with maximum nine DS18B20

Please help me with my issue

Thanks

Pawel

Scott216, 2015-06-30:
How are they wired?  Are you using in parasitic mode?  I would try adding a short delay between each reading.  Also, you could check for -127 and if that's what you got, then read it again.

kluczyk11, 2015-06-30:
Thanks for answer , three  wires , no  parasitic mode , readings every 15 minutes and this is strange. Becouse when reading is good (every 15 minutes)  for nine DS18B20  and I connect tenth DS18B20 the readings are changening every second or 2 seconds. If something is wrong this strange reading should be every 15 minures.Should I set readings for example every 8 minutes for  three DS18B20  , every 7 minutes for another   three DS18B20........ Becouse now Arduino tries to read all ten DS18B20  every 15 miuntes.

Scott216, 2015-06-30:
Does your code read all 10 at the same time?  If so, I'd add a delay between each one; 100mS should be enough.  I'm using 50 mS and that works pretty good.

kluczyk11, 2015-06-30:
Thanks, I will change the code and check.

RobTillaart, 2015-06-30:
More sensors implies the line is longer and you might need a smaller pull up resistor.
What value do you use now?

kluczyk11, 2015-06-30:
I have 4,7 k ohm.

RobTillaart, 2015-06-30:
Try a 3K3 or 2K2 when you have 10 sensors

kluczyk11, 2015-07-02:
I tried 2,2 k ohm and it works perfect. Thanks a lot !!!!!!!

closed 2015-07-02 by kluczyk11