milesburton/Arduino-Temperature-Control-Library #36: Can't set resolution with setResolution()

tk5ep, 2015-07-15:
Hi,

I use the requestTemperatures() and getTempCBIndex() functions.

Ahead of that, i tried to set the resolution with setResolution(12), but it seems that it doesn't work. It should set all devices...

I used the "single" example and modified it for 12 bits resolution and that worked. This example makes use of the deviceaddress.

After that, my sketch displays OK in 12 bits resolution.

Any clue ?

RobTillaart, 2015-07-15:
Can you print  getDeviceCount(void);  just before setResolution();

What does it return?

tk5ep, 2015-07-16:
Hi  Rob,

getDeviceCount()  returns 0.

Here a part of my sketch:

#include "FreqPeriod.h"
#include <LiquidCrystal.h>
#include <OneWire.h>                  
#include <DallasTemperature.h>      

#define ONE_WIRE_BUS 10                         // Pin 10 One Wire

LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

OneWire oneWire(ONE_WIRE_BUS);                  
DallasTemperature DS18B20(&oneWire); 

double lfrq;
long int pp;
float vitesseVent, vitesseVentMax;
float tempExt, tempExtMax, tempExtMin;
char buffer[15];

void setup() {
  Serial.begin(115200);
    FreqPeriod::begin();

lcd.begin(16,2); 
lcd.print("Station meteo");
delay(2000);
lcd.clear();

 DS18B20.setResolution(12);
 DS18B20.requestTemperatures();
 tempExt = DS18B20.getTempCByIndex(0); 

tempExtMax=tempExt;
tempExtMin=tempExt;
}

void loop() {

//  Serial.println (DS18B20.getDeviceCount());
 DS18B20.setResolution(9);
 DS18B20.requestTemperaturesByIndex(0);                    // Send the command to get temperatures
  tempExt = DS18B20.getTempCByIndex(0);            // on récupère les temps. par index de capteurs
  
  if (tempExt < tempExtMin) {
    tempExtMin = tempExt;
    }
                
  if (tempExt > tempExtMax) {
    tempExtMax = tempExt;
    }

robtillaartfei, 2015-07-16:
I think your sketch is missing 
```C++
DS18B20.begin();
```
call before the setResolution

give it a try

tk5ep, 2015-07-16:
Hi Rob,

You were perfectly right !
How could i have overlook this ??

Sorry for bandwidth and thanks for the help, all works perfectly now.

Patrick.

RobTillaart, 2015-07-16:
Everyone is blind for his/her own code ;)

@milesburton 
please close issue

tk5ep, 2015-07-17:
Thanks again and keep up the good job !

Cheers,

closed 2015-07-17 by tk5ep