milesburton/Arduino-Temperature-Control-Library #47: The

JoakimSoderberg, 2015-12-26:
Note that the example https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Multibus_simple/Multibus_simple.ino does not work unless you have changes in `OneWire.h` that is not yet merged in that lib.

Specifically this Pull Request has not yet been merged:
https://github.com/ntruchsess/arduino-OneWire/pull/2

JoakimSoderberg, 2015-12-27:
Ok, I now realise that the mentioned repository I linked to is not the official one. The arduino lib community is very confusing :(

As I understand it, this is the official repository: https://github.com/PaulStoffregen/OneWire

But the issue still applies. The example requiers the changes by @prashantdev in `OneWire.h` to work

2mik, 2016-02-06:
The reason that OneWire class does not contain a default parameterless constructor. I think Multibus_simple.ino is outdated

2mik, 2016-02-06:
I've started the example after the following changes:
#include <OneWire.h>
#include <DallasTemperature.h>

//int oneWirePins[]={3,7};//OneWire DS18x20 temperature sensors on these wires
//const int oneWirePinsCount=sizeof(oneWirePins)/sizeof(int);

#define ONE_WIRE_BUS0 2
#define ONE_WIRE_BUS1 3
#define ONE_WIRE_BUS2 4

const int oneWirePinsCount = 3;

//OneWire ds18x20[oneWirePinsCount];
OneWire oneWire0(ONE_WIRE_BUS0);
OneWire oneWire1(ONE_WIRE_BUS1);
OneWire oneWire2(ONE_WIRE_BUS2);
OneWire ds18x20[] = {oneWire0, oneWire1, oneWire2};
DallasTemperature sensor[oneWirePinsCount];

prashantdev, 2016-02-07:
The whole reason for my writing this code (and the empty constructor) is to
have exactly one place to make a change in sensors - add a new sensor,
change a pin, remove a sensor. Change in one place, and don't worry about
code breaking elsewhere.

So, while I could have started out by doing what you did, my liking for a
better/cleaner code forced me to update OneWire. Multibus_simple derives
from my attempt to write better code in my .ino files.

2mik, 2016-02-07:
Thank you for the very usefull library

milesburton, 2016-05-24:
Closing issue for now. Feel free to reopen if any new information appears

closed 2016-05-24 by milesburton

Bouffski, 2016-09-06:
@2mik Would it be possible to publish the complete -revised- example sketch? I have two 1820's, one is recognized, the other isn't...
And me knowledge is just (too) little... (and not only on this subject, helas)
BTW thank you very very much, I'm almost home!

2mik, 2016-09-07:
Oh, it was so far ago, sorry.
May be my project that uses this library, helps you. See [the link](https://github.com/2mik/arduino-2mik/blob/master/net_th_meter/net_th_meter.ino)