milesburton/Arduino-Temperature-Control-Library #52: Failure when temperature is a multiple of 16.0 deg C

dave-p, 2016-03-26:
My application contains the following (pseudocode) loop which is called once per second:

  if (sensors.isConversionAvailable(chSensor)) {
    ch.temp = sensors.getTempC(chSensor);
    if (ch.temp != DEVICE_DISCONNECTED) {
      . . .
    }
    sensors.requestTemperatures();
  }

This code fails if the temperature reading is exactly 16.0 degrees C - isConversionAvailable() always returns FALSE.
Looking at the library code, isConversionAvailable() reads all of the scratchpad registers and returns TRUE if register 0 (temperature LSB) is non-zero. However a temperature of 16.0C is stored as 0x0100 in the scratchpad so the LSB will be zero and FALSE will be returned.
I have replaced isConversionAvailable() by the following, which follows the device datasheet by issuing a single read slot to test the data line, and my code now works correctly.

bool DallasTemperature::isConversionAvailable(uint8_t* deviceAddress)
{
	// Check if the clock has been raised indicating the conversion is complete
  _wire->reset();
  uint8_t b = _wire->read_bit();
  _wire->reset();
  return (b == 1);
}

chrwei, 2016-03-30:
current code does select the address first, so it is per sensor.

would "return scratchPad[0] + scratchPad[1];" work?  I'm about to try it, was just doing research first and saw this.  basically, if the first 2 bytes are both zero it's zero, otherwise it's truthy

dave-p, 2016-03-31:
Adding (or ORing) scratchPad[0] and scratchPad[1] would still fail at 0
degrees C; whether that was a problem would depend on the application.

I've been running more tests. It seems that isConversionAvailable() 
returns TRUE if there has ever been a conversion completed on the 
device, it says nothing about whether the most recent conversion request 
has completed.

The library header file used to define another function 
isConversionComplete() but there was no code for this function in the 
library and the definition was removed by commit 45db7e7 on 2014-04-10. 
I've added the following:

bool DallasTemperature::isConversionComplete()
{
   uint8_t b = _wire->read_bit();
   return (b == 1);
}

and this works as I would expect, it returns FALSE for as long as there 
is a conversion in progress on the bus. I don't have enough spare 
hardware to test it with multiple sensors on the bus but I would expect 
to only return TRUE when all conversions were complete.

muscatelle, 2016-05-23:
I'd just like to confirm the problems with isConversionAvailable() and the success of using the isConversionComplete() code above with a single device.

Thanks so much

milesburton, 2016-05-24:
I've never been entirely certain why the body of that method was removed. It must have been a mistake. Thanks for your comments

milesburton, 2016-05-24:
I've restored isConversionComplete for the time being. However, it sounds like isCompleteAvailable may has a bug that needs investigating

vallon, 2016-05-28:
I came here checking if anybody else is having problems with the temperature blocking code, and it seems I am not alone.

Here is what I can show (non-parasite power, one sensor, leaving all optional behavior at its default):

* Set resolution to 12 bits
* Call getTempC() which reads scratch and maps into float.  You get the most recent temp (Temp1) reading (sitting in scratch).
* Use micros() to time a call to requestTemperaturesByAddress.  It returns too soon.
* Call getTempC() which reads the old temp (Temp1).  The temperature calculation is still in progress.
* Repeatedly call getTempC() until it changes.  If it does change, it takes about 70 ms (datasheet says max is 750ms).

I believe the proper way to poll for conversion on a self-powered bus is:

```
_wire->reset();
_wire->select(target);
_wire->write(STARTCONVO, parasite);
if (parasite) {
    delay(maxConversionTime(resolution)+1); // +1 for slop
    _wire->depower(); // write(*, true) left bus power on
} else {
    while (_wire->read_bit() == 0) {
        // Somebody is still busy
        // yield() here?
        // timeout at 1.1 * 750ms (recover from errant sensor)
    }
}
```

I think isConversionAvailable is completely bogus.  The comment leads me to believe it is a misunderstanding of the read time slots:

```
bool DallasTemperature::isConversionAvailable(const uint8_t* deviceAddress){

    // Check if the clock has been raised indicating the conversion is complete
    ScratchPad scratchPad;
    readScratchPad(deviceAddress, scratchPad);
    return scratchPad[0];

}
```

The comment sounds like it means to implement "_wire -> read_bit() != 0", but is reading the scratchpad instead.

BTW, this is also why I get an 85C at startup (powerup).  The scratchpad init temp is 85.0C.  The getTempC() call is run before the first temperature conversion is complete, and the init temp is returned.

The following client code correctly blocks (not for parasite power, though):

```
    dalTemp.setWaitForConversion(false);
    dalTemp.requestTemperaturesByAddress(addr); // takes 19ms
    while (!dalTemp.isConversionComplete()) // takes 80ms
        yield();
    float temp = dalTemp.getTempC(addr);
```

I can/will probably clean up the code now that I see that other people are having problems/concerns with the temperature blocking.

Also, while I only have one temperature sensor, I believe this would also apply to multiple sensors.  If reset / skip / write(STARTCONV0) is run, each (every) sensor will pull the bus low in response to the read time slots until they each have completed their conversion.  Once the last (slowest) has completed, it will no longer pull the bus low, and isConversionComplete() will return true.

Comments?

vallon, 2016-05-28:
I went crazy and did a lot of cleanup.  I forked this repo and pushed back my changes into my github account (first fork on github!).  There are many commits but each should be easily digestible.

A few thoughts:

1) Is CheckForConversion feature (true/false) useful?  My 12-bit sample takes about 70ms when polling, but worst-case time is 750ms.  Seems like there would be no reason to disable CheckForConversion, and removing it would eliminate/simplify some code.

2) There are a few places where _wire->reset() is called after the command is complete.  It seems to only waste time since every command always starts with a reset() anyway, and I don't think devices are any "better off" after a reset.

3) If bus power is used, and check for conversion is always true, then since parasite == false, the maxBitResolution is unnecessary.  Thus, begin() doesn't need to poll bitResolution if !parasite.

4) The crc check (crc(buf, count-1) == buf[count-1]) can be alternately written (crc(buf, count) == 0) which makes for more readable code.  I didn't touch any of those - there are a few instances.

@milesburton - Let me know if you want to see any of this split out into separate independent commits, create a separate issue, or whatever.

milesburton, 2017-02-16:
@vallon Sorry the epic delay. Do you want to create a PR and I'll go through it?

milesburton, 2017-12-23:
Some good changes on your fork. I've raised a PR https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/103/files

Will need to review the changes given the time that's elapsed but it's useful to see the comparison.

shajek, 2018-01-22:
i have this problem either. I stucked on  0°C 16°C 32°C. How can i fix it finally ?

RobTillaart, 2018-01-22:
@shajek 
Did you use the latest version of the library? 
The fix of dave-p is in the latest version and according to him it should work.

shajek, 2018-01-22:
i look at it, i use PlatformIO with this library, i have latest 3.7.7 i see now update to 3.7.8 .. i check if it works :) thank you

shajek, 2018-01-22:
eww.. i use in my code
`if (ds18b20.isConversionAvailable(ds18b20Address))`

but is not usable for now .. is good use ?
`if (ds18b20.isConversionComplete())`

RobTillaart, 2018-01-22:
Yes that is the one that should work.

milesburton, 2018-01-24:
Going to close this for now. Shout me if there's any issues

closed 2018-01-24 by milesburton