milesburton/Arduino-Temperature-Control-Library #54: Wrong measurements when having two sensors on the same line

viktak, 2016-04-21:
hi,
I have been using your library for a few days now, and it's been working fine. But now I added another sensor chip to the circuit and, although the code recognizes both chips, the measured temperatures are about twice as much as they should be.
This is the code I am using:

definitions:
`struct thermometer{
  DeviceAddress deviceAddress;
  float measuredTemperatureC;
}thermometer[32];
uint8_t oneWireDevicesCount;
`

using it:
`        Serial.print("Requesting temperatures... ");
        sensors.requestTemperatures(); // Send the command to get temperatures
        Serial.println("Done.");

        for (size_t i = 0; i < oneWireDevicesCount; i++) {
          Serial.print("Device ");
          Serial.print(i);
          Serial.print(" temperature:\t");
          thermometer[i].measuredTemperatureC = sensors.getTempC(thermometer[i].deviceAddress);
          printTemperature(thermometer[i].measuredTemperatureC);
        }`
This is a sample output from it:
`Requesting temperatures... Done.
Device 0 temperature:   Temp C: 91.62   Temp F: 196.93`

Even if I unplug one of them without power cycling the device, I get correct readings from the one that stayed online:
Requesting temperatures... Done.
Device 0 temperature:   Temp C: 91.62   Temp F: 196.93
Device 1 temperature:   Temp C: 89.56   Temp F: 193.21
Requesting temperatures... Done.
Device 0 temperature:   Temp C: -127.00 Temp F: -196.60
Device 1 temperature:   Temp C: 21.75   Temp F: 71.15
Requesting temperatures... Done.
Device 0 temperature:   Temp C: 96.00   Temp F: 204.80
Device 1 temperature:   Temp C: 86.00   Temp F: 186.80
You can clearly see that the middle sample is the one where one of the sensors were missing and the other one gave correct reading.

Any idea what might cause this?

viktak, 2016-04-21:
I actually figured it out: I was using the devices in parasite power mode. Once I gave them power (just to try out all possibilities) it was working. Then, I guess this is a bug. :(

Anyway, I'd appreciate your comments on this issue. It's not a big problem for me, this library makes development for these chips a breeze.

RobTillaart, 2016-04-21:
hypothesis:
If bits are clocked in incorrectly e.g. by starting too early. it can happen that a bitpattern is shifted 1 or more bits. (had this with another lib/sensor) A bit shift of 1 bit equals a multiplication by 2 and a bitshift of 2 equals a multiplication by 4. This latter could explain the difference between 21.75C and 86.00C (== 21.5 x 4)

That said, I do not recall seeing this problem on the Arduino forum. 
disclaimer: No experience with the parasite power mode

viktak, 2016-04-21:
Yes, this makes sense. As per my second comment, I can now easily repro the same (faulty) behaviour.

mattdreimer, 2016-05-20:
If you are using these chips in parasite power mode ground should be shorted to VDD(pin3) and you will need a 4.7kohm (if you have a lot of devices or a long bus you may need to play with this value) resistor between power and the dq line see this page for an image http://www.tweaking4all.com/wp-content/uploads/2014/03/ds18b20-parasite-power.jpg

viktak, 2016-05-20:
@mattdreimer  Thank you, this indeed works. Thanks!

closed 2016-05-20 by viktak