milesburton/Arduino-Temperature-Control-Library #57: warning: unused parameter 'deviceAddress'

Krosik, 2016-06-21:
sample code
```
#include <DallasTemperature.h>
void setup() {
}
void loop() {
}
```

when I open IDE 1.6.9 and compile code first time I always see this error

```
C:\Users\User\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:781:6: warning: unused parameter 'deviceAddress' [-Wunused-parameter]

 void DallasTemperature::defaultAlarmHandler(const uint8_t* deviceAddress){}
```
When I compile it again, there is no error. This error is showing always only after first compilation when IDE is opened.

Can you fix this?

RobTillaart, 2017-12-22:
@Krosik 
That function has no implementation, so the warning is correct in itself. The second time you compile, the compiler see no chances in the object file generated for the library, so it does not recompile it.

A look at the code shows that the defaultAlarmhandler is only set with one of the constructors.  

@milesburton 
I propose to remove the defaultAlarmHandler() as it adds no value. If people need an empty alarmHandler they can create it themselfs. If you agree I will create a PR for it.

milesburton, 2017-12-22:
Hi Rob,

Agreed, let's remove it. Feel free to raise a PR, or I can do it.

Miles

RobTillaart, 2017-12-22:
PR #100 fixes this

closed 2017-12-22 by milesburton