milesburton/Arduino-Temperature-Control-Library #58: after resolution change is required sensors.begin()

Krosik, 2016-06-21:
Hello,

after I triple checking for this, I found the bug that after using of function sensors.setResolution the resolution is change and with sensors.getResolution you will get new resolution, but for some reason sensor's delay is not changed.

For example sensor has default 12 bit, after changing to 9 bit I'm receiving only 9 bit precision but it take 750 ms.

After you will reset Arduino, then this is fixed and it take around 110 ms.

If you will test this, please note, when you open Serial console in Arduino IDE, it will always reset Arduino = you will start receiving correct delay.
So to verify this issue you need open different serial console, or you need have serial cable without possibility to reset Arduino.

For now I can solve this by adding init command
```
sensors.begin();
...
sensors.setResolution(insideThermometer, 9);
sensors.begin();
```

Strange is, when I change resolution to 9 bit, sensors.begin and then to 12 bit, Im receiving 12 bit resolution number and it take only around 110 ms.

Please investigate, thank you.

webbes, 2016-08-03:
#61 Should fix your issue. By default, the setResolution(deviceAddress, value) now recalculates the global resolution which is used when reading all values.

RobTillaart, 2017-12-22:
@milesburton 
#61 is merged so this one can be closed

closed 2017-12-23 by milesburton