milesburton/Arduino-Temperature-Control-Library #60: Write to eeprom only when current value is different

webbes, 2016-07-31:
People might call "setResolution(12)" in setup or loop.
Devices that deepSleep by resetting themselves, might setup very often. This could kill the eeprom of the sensors.

The "setResolution(const uint8_t* deviceAddress, uint8_t newResolution)" method should/could check the current device resolution before actually changing it.

Same goes for every method that writes to eeprom.

milesburton, 2016-07-31:
Hmm, it's a good point. My only concern is the additional complexity of doing so.

webbes, 2016-08-01:
I tried to use git to make the changes and then a pull request, but I don't understand anything about git... because of that I decided to attach my changes... it might save you some work...

updated:
setResolution(const uint8_t* deviceAddress, uint8_t newResolution)
setUserData(const uint8_t* deviceAddress, int16_t data)
setHighAlarmTemp(const uint8_t* deviceAddress, char celsius)
setLowAlarmTemp(const uint8_t* deviceAddress, char celsius)

I know it is not the right way, but at least I tried to give back to the community :-)

[DallasTemperature.txt](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/393980/DallasTemperature.txt)

webbes, 2016-08-01:
In the mean while I figured out GIT. I'll test my changed thoroughly and then commit a pull request...

webbes, 2016-08-01:
#61 Fixed the issues

closed 2016-08-01 by webbes