milesburton/Arduino-Temperature-Control-Library #64: Error on reboot

rally7, 2016-08-29:
I have a Pro Micro ATmega32U4 that has a simple sketch that simply reads the temp from A DS18B20.  The sketch uploads and runs fine the first time printing out the correct temp until you restart or reset the Arduino.  Then the temp prints out as -196.60  I have tried multiple sensors with the same result. 

The DS18B20 is connected Red to +5v, Black to Ground and Yellow (signal) is connected to Red by 4.7K resistor and to pin 6 on the Mini.

Thanks for any help!

Bill

#include <DallasTemperature.h>
#include <OneWire.h>

OneWire oneWire(6);
DallasTemperature sensors(&oneWire);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("Program Start");
 
   //initialized temp sensors
  sensors.begin();
 
}

void loop() {
  // put your main code here, to run repeatedly:
 
  sensors.requestTemperatures();
  float temp;
  temp = sensors.getTempFByIndex(0);
  Serial.print("Now sending ");
  Serial.println(temp);

}

MarkVeldt, 2016-09-26:
Hi Bill,

You have to wait for the conversion to be finished, see the WaitForConversion examples at the end of the code for more explanations.

Also -196.6 is defined in the header file that should also give a hint.
(#define DEVICE_DISCONNECTED_F -196.6)

In parasitic mode the function requestTemperatures() always returns directly.

Happy programming.

RobTillaart, 2017-12-22:
@rally7 
Is this issue still valid?

milesburton, 2017-12-23:
Please reopen if this is still an issue. I presume you've solved it during the last year ;)

closed 2017-12-23 by milesburton