milesburton/Arduino-Temperature-Control-Library #66: Compiling in INO environment (command-line Arduino dev)

marcusj, 2016-09-24:
I had some problems compiling the cpp code within the ino environment and had to explicitly cast out some of the const specifications.  Diff:

`79c72
<     return (_wire->crc8(deviceAddress, 7) == deviceAddress[7]);
---
>     return (_wire->crc8((unsigned char*)deviceAddress, 7) == deviceAddress[7]);
121c114
<     _wire->select(deviceAddress);
---
>     _wire->select((unsigned char*)deviceAddress);
149c142
<     _wire->select(deviceAddress);
---
>     _wire->select((unsigned char*)deviceAddress);
160c153
<     _wire->select(deviceAddress);
---
>     _wire->select((unsigned char*)deviceAddress);
173c166
<     _wire->select(deviceAddress);
---
>     _wire->select((unsigned char*)deviceAddress);
353c346
<     _wire->select(deviceAddress);
---
>     _wire->select((unsigned char*)deviceAddress);
834c827
`

milesburton, 2017-01-28:
Thanks, I'll take a look

Patroklo, 2017-02-11:
For noobs like me, if you are having right now errors compiling INO (for example with the Arduino IDE) files with this library, change the extension of the file to .PDE and it will work fine.

RobTillaart, 2017-12-22:
@marcusj 
Is this issue still valid?

milesburton, 2017-12-23:
As per @RobTillaart, reopen if this is still an issue

closed 2017-12-23 by milesburton