milesburton/Arduino-Temperature-Control-Library #70: Ethernet crashes when calling sensors.begin()

sanchezluciano, 2017-01-26:
Hi,

I've recently been dealing with an issue which appears to be in the DallasTemperature library. The problem occurs when I call sensors.begin() in the startup() function. I basically have a SQL database set up that is storing values to display later on a website. [(EXAMPLE HERE)](http://www.icreateproject.info/uncategorized/arduino-save-data-to-database/)

The moment I un-comment the sensors.begin() line in my code, the program stops connecting to my server and does not send any more data. If I comment out the line, I can send data to my server just fine. 

Is there an issue with this library and the Ethernet library I am not aware of? I've looked around and can't find a solution. 

Thanks,
Luciano

milesburton, 2017-01-28:
Morning, are you able to paste the offending code?

It's possible you're using one of the pins the ethernet library is using. Alternatively OneWire could be using one of the interrupts.

The ethernet library by it's very nature is quite demanding of the Arduino. It'll uses quite a few pins and interrupts 

Have you tried using different pints just to rule out the obvious

sanchezluciano, 2017-01-30:
Sorry for getting back so late. I never received a notification from the post. 

The current setup I have does not require any pins for the Ethernet library to run and upload data to my SQL server. 
I have used various prints in my code to see where it is getting hung up and in this case the code halts right before the if statement and doesn't continue to loop. Please note this only occurs when I am using sensors.begin(). If I do not specify that line or comment it out, that is when I have issues. 

Below is code I just put together using the example from the SQL database sketch and my current program. I have not tested it on an actual Arduino since I do not have access to it at the moment but I will when I get home later. I'm suspect that the code below will have the same issue though. 

Thanks for the help,
Luciano

`#include <OneWire.h>
#include <DallasTemperature.h>
#include <SPI.h>
#include <Ethernet.h>

// Dallas Temperature Setup
const int temperaturePin  = 53;
OneWire oneWire(temperaturePin);
DallasTemperature sensors(&oneWire);
DeviceAddress t1 = { 0x28, 0xFF, 0xB5, 0xB6, 0x70, 0x16, 0x05, 0xDB };

byte mac[] = {
  0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
 
// Enter the IP address for Arduino, as mentioned we will use 192.168.0.16
// Be careful to use , insetead of . when you enter the address here
IPAddress ip(192,168,0,16);

char server[] = "192.168.0.11"; // IMPORTANT: If you are using XAMPP you will have to find out the IP address of your computer and put it here (it is explained in previous article). If you have a web page, enter its address (ie. "www.yourwebpage.com")

// Initialize the Ethernet server library
EthernetClient client;

void setup() {
 
  // Serial.begin starts the serial connection between computer and Arduino
  Serial.begin(9600);
  sensors.begin();//WHEN I UNCOMMENT THIS, THE CODE STOPS WORKING, WHEN I COMMENT, IT WORKS.
 
  // start the Ethernet connection
  Ethernet.begin(mac, ip); 
}

void loop() {
  float temp1Val = calculateTemp(t1);//Get temperature in F of t1 sensor
  Serial.println(F("NOT CONNECTED: before if statement\n"));
  // Connect to the server (your computer or web page)  
  if (client.connect(server, 80)) {
    Serial.println(F("CONNECTED: Inside if\n"));
    client.print("GET /write_data.php?"); // This
    client.print("value="); // This
    client.print(temp1Val); // sent temp1Val value
    client.println(" HTTP/1.1"); // Part of the GET request
    client.println("Host: 192.168.0.11"); // IMPORTANT: If you are using XAMPP you will have to find out the IP address of your computer and put it here (it is explained in previous article). If you have a web page, enter its address (ie.Host: "www.yourwebpage.com")
    client.println("Connection: close"); // Part of the GET request telling the server that we are over transmitting the message
    client.println(); // Empty line
    client.println(); // Empty line
    client.stop();    // Closing connection to server

  }
  else {
    // If Arduino can't connect to the server (your computer or web page)
    Serial.println(F("NOT CONNECTED: inside else:"));
    Serial.println("--> connection failed\n");
  }
 
  // Give the server some time to recieve the data and store it. I used 10 seconds here. Be advised when delaying. If u use a short delay, the server might not capture data because of Arduino transmitting new data too soon.
  Serial.println(F("Before Delay\n"));
  delay(10000);
}

float calculateTemp(const unsigned char *tempSensor) {
  sensors.requestTemperatures();
  return sensors.getTempF(tempSensor);
}`

sanchezluciano, 2017-02-08:
Any ideas why this might be happening? Is it also happening to anyone else? 

Luciano

muldereric, 2017-02-09:
Because of the high pin numbers I am assuming you are using an Arduino Mega. The ethernet shield 2 uses pins 50, 51 and 52. The manual says that pin 53 has to stay configured as output else the spi interface won't work. There's your problem.

muldereric, 2017-02-09:
From arduino.cc:
Arduino communicates with both the W5500 and SD card using the SPI bus (through the ICSP header). This is on digital pins 10, 11, 12, and 13 on the Uno and pins 50, 51, and 52 on the Mega. On both boards, pin 10 is used to select the W5500 and pin 4 for the SD card. These pins cannot be used for general I/O. On the Mega, the hardware SS pin, 53, is not used to select either the W5500 or the SD card, but it must be kept as an output or the SPI interface won't work.

RobTillaart, 2017-12-22:
@milesburton 
Seems a valid explanation, no discussion anymore so can be closed.

closed 2017-12-23 by milesburton