milesburton/Arduino-Temperature-Control-Library #71: #define CONFIGURATION conflicts with other libraries

Nzbuu, 2017-02-09:
```
#define CONFIGURATION   4
```

conflicts with other libraries that use the same string "CONFIGURATION" in other places. For example: in https://github.com/marvinroger/homie-esp8266/blob/develop/src/HomieBootMode.hpp, CONFIGURATION is part of an enumeration.

Leads to compiler errors such as: "error: expected identifier before numeric constant".

milesburton, 2017-02-16:
Good spot. We should name space this.

Nzbuu, 2017-04-08:
Is there a good reason why the scratchpad locations need to be in the header file? Would it be better to just move them into the source file? Under what circumstances would any other module need to use the scratchpad locations?

I can do this and raise a pull request, if you like.

milesburton, 2017-04-11:
Go ahead @Nzbuu

PR #75 (2017-04-11): Move internal constants to DallasTemperature.cpp
Moved the following constants, since the user should not need to use them:

* Onewire commands
* Scratchpad locations
* Device resolution (handled in setResolution and getResolution)

Closes #71
closed 2017-04-11 by milesburton