milesburton/Arduino-Temperature-Control-Library #74: Why the temperature from DS18B20 by using the library is always zero？

zhoufusong, 2017-03-29:
Why the temperature from DS18B20 by using the library is always zero？ hope your reply.

jiapei100, 2017-07-31:
Same here... 
In my case,  **sensors.getDeviceCount()** always returns 0 .

Cheers
Pei

RobTillaart, 2017-07-31:
Do you have a pull up resistor on the dataline? e.g. 4K7 between +5V and dataline.
 If not the communication is less reliable (longer wire length needs smaller resistor).
Do you have a proper 5Volt? please check with a voltmeter.

RobTillaart, 2017-07-31:
@zhoufusong   Sorry cannot read the link you posted, but schematic looks ok.

RobTillaart, 2017-12-22:
@milesburton 
There is no reaction from the "questioneers" so apparently the problems are solved,
please close

closed 2017-12-23 by milesburton

Spreeuw, 2018-09-19:
I have the same issue, and I don't quite understand why:
`
  numberOfDevices = sensors.getDeviceCount(); // returns 0
  numberOfDevices = 3;
  for (int i = 0; i < numberOfDevices; i++) {
    if (sensors.getAddress(tempDeviceAddress, i)) {
      sensors.setResolution(tempDeviceAddress, TEMPERATURE_PRECISION);
    }
  }
`
so `sensors.getDeviceCount()` returns 0 yet I can setResolution and do `sensors.requestTemperatures()` and `sensors.getTempCByIndex(i)` without problems. Any ideas? I do this on an ESP32 (Lolin D32 Pro), so not sure if this is somehow related to #114

chrwei, 2018-09-19:
this issue about the temperature returning zero, not the number of devices.  you should make a new issue,