milesburton/Arduino-Temperature-Control-Library #77: Exemple Multiple.pde  Assign address manually

kamahat, 2017-04-24:
Using method :  Assign address manually
Ligne : 42: error: assigning to an array from an initializer list
   insideThermometer = { 0x28, 0xFF, 0x19, 0x71, 0xC1, 0x15, 0x01, 0x55 };
                     ^
Ligne : 43: error: assigning to an array from an initializer list
   outsideThermometer   = { 0x28, 0xFF, 0xFD, 0xAE, 0xC1, 0x15, 0x01, 0x0B };

The array can only be declared when you initialize it. It cannot be declare 1st and initialize after, it need to be declared 1 more time.
should be : 

// arrays to hold device addresses
DeviceAddress insideThermometer, outsideThermometer;
[...]
// Assign address manually. The addresses below will beed to be changed
  // to valid device addresses on your bus. Device address can be retrieved
  // by using either oneWire.search(deviceAddress) or individually via
  // sensors.getAddress(deviceAddress, index)
**DeviceAddress** insideThermometer = { 0x28, 0xFF, 0x19, 0x71, 0xC1, 0x15, 0x01, 0x55 };
**DeviceAddress** outsideThermometer   = { 0x28, 0xFF, 0xFD, 0xAE, 0xC1, 0x15, 0x01, 0x0B };

RobTillaart, 2017-12-22:
@kamahat 
Is this issue still valid? 
Which examples are involved or not compiling?

kamahat, 2017-12-23:
i've  not tested all the exemple. I've meet, and AFAIK the issue is still there, with multiple\Multiple.pde

RobTillaart, 2017-12-23:
@kamahat 
The PR fixes the example Multiple.pde , 
thanks for your help so far

milesburton, 2017-12-23:
Thanks Rob

closed 2017-12-23 by milesburton