milesburton/Arduino-Temperature-Control-Library #81: Improper implementation of operator new/delete

EmilyBjoerk, 2017-05-25:
I just saw this and thought I'd mention it.

The `operator new/delete` as found in [DallasTemperature.cpp:828](https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/DallasTemperature.cpp#L828) is incorrect. According to the C++ specification of [operator new/delete](http://en.cppreference.com/w/cpp/memory/new/operator_new) the operators should only allocate and free the memory regions.

The constructor and destructor calls are inserted by the compiler. As the `operator delete` is calling the destructor explicitly, this will cause a double destruction of the object. Which is undefined behaviour. However as the destructor is trivial it is *probably* benign in that it won't cause any serious issues.

If this is some kind of a work around for a compiler but then I believe that it should be clearly stated.

milesburton, 2017-12-23:
This wasn't the case when the original library was written. If that's changed now, fantastic. We can remove it in the next PR. Let me know if you have any thoughts.

closed 2017-12-23 by milesburton

EmilyBjoerk, 2017-12-23:
This aspect of the c++ standard has always been this way. It's nothing new
in recent times. Note that I'm not talking about what the compiler does,
but rather what the standard says. Which hopefully is the same thing but
bugs happen.