milesburton/Arduino-Temperature-Control-Library #83: Multibus_simple example fails to compile.

CvW, 2017-07-03:
Hi there, 

The Multibus_simple example that ships with this library fails to compile. The error code is as follows: 

`Multibus_simple:7: error: no matching function for call to 'OneWire::OneWire()'
 OneWire ds18x20[oneWirePinsCount];
                                 ^
In file included from D:\Arduino\libraries\DallasTemperature\examples\Multibus_simple\Multibus_simple.ino:1:0:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire/OneWire.h:269:5: note: candidate: OneWire::OneWire(uint8_t)
     OneWire( uint8_t pin);
     ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire/OneWire.h:269:5: note:   candidate expects 1 argument, 0 provided
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire/OneWire.h:254:7: note: candidate: constexpr OneWire::OneWire(const OneWire&)

 class OneWire
       ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire/OneWire.h:254:7: note:   candidate expects 1 argument, 0 provided
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire/OneWire.h:254:7: note: candidate: constexpr OneWire::OneWire(OneWire&&)
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire/OneWire.h:254:7: note:   candidate expects 1 argument, 0 provided

Multibus_simple: In function 'void setup()':
Multibus_simple:22: error: 'class OneWire' has no member named 'setPin'
     ds18x20[i].setPin(oneWirePins[i]);
                ^
Using library OneWire at version 2.3.2 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\OneWire 
Using library DallasTemperature at version 3.7.6 in folder: D:\Arduino\libraries\DallasTemperature 
no matching function for call to 'OneWire::OneWire()'`

I suspect that OneWire has changed its function names and that this library needs to be updated?

RobTillaart, 2017-07-27:
Hi CvW,

Please change line 5:

**OneWire ds18x20[oneWirePinsCount];**
to
**OneWire ds18x20[oneWirePinsCount] = { 3, 7 };**

and line 22:

**ds18x20[i].setPin(oneWirePins[i]);**
to
**ds18x20[i].set_pin(oneWirePins[i]);**

Give it a try

RobTillaart, 2017-07-27:
@milesburton
Above changes does also work with latest OneWIre from [https://github.com/PaulStoffregen/OneWire](https://github.com/PaulStoffregen/OneWire)

provided a pull request.

milesburton, 2017-07-27:
Thanks all

closed 2017-07-27 by milesburton