milesburton/Arduino-Temperature-Control-Library #85: Why sensors.getDeviceCount() always returns 0?

jiapei100, 2017-07-31:
As title:

Why sensors.getDeviceCount() always returns 0?
How to debug the issue?

Cheers

RobTillaart, 2017-07-31:
@milesburton 
Discussed already here #74 so this is a duplicate (can be closed).

closed 2017-07-31 by milesburton

timothy3001, 2019-08-03:
I think there's a lot of confusion going on. Issue #74 is about the TEMPERATURE being always 0. This one is about the DEVICE COUNT always being 0. I actually encounter the same problem here currently, also using an ESP32 like @Spreeuw. My code is basically the same as his, I can read the temperature but I'm not getting a device count. Are there other issues I didn't see yet which are not referenced? 

Should we maybe further investigate this? This might be an issues with ESP32s. 

In case these are necessary information: I wired the DS18B20 to the ESP32 using 5V (VIN of the ESP32) for the VIN of the sensor, connected the 3.3V of the ESP32 via an 4.7K resistor to the Data line and used Pin 14 for Data. Ground is of course connected as well.

TazerReloaded, 2019-08-06:
I found this issue because I had the same problem, but with an Arduino Mega.
Hardware wired correctly, 5V supply with 4.7K Pull-up to 5V on the data bus.
The solution for me was very simple: I forgot the call to "sensors.begin();", which resets the bus and waits for devices. Please check if you're missing that call too. Otherwise, it might really be a problem with ESP32s, because now everything works correctly on my Arduino.

JopWerff, 2019-11-19:
I have also this problem sensors.getDeviceCount() on ESP32, but it is different.
Sometimes returning 0 but other times returning the correct value 3.
Moreover, mostly when returning 0, also the durance of sensors.requestTemperatures() is only 2 or 3 miilis, where it should be much more, dependent of chosen bit-resolution - with sensors.setResolution(9 - 12) -  like this output of the serial monitor:

```
Start ESP32 Multiple DS18B20 Sensors - versie 1.0
ESP32 IDF-versie: v3.2.3-14-gd3e562907

Locating devices...Found 3 devices.

Resolution= 9  -  deltaT= 96 ms.
Sensor 1 : 16.00°C  |  t2= 29 ms.Resolution= 10  -  deltaT= 190 ms.
Sensor 2 : 17.50°C  |  t2= 45 ms.Sensor 1 : 16.00°C  |  t2= 29 ms.
Sensor 3 : 85.00°C  |  t2= 61 ms.Sensor 2 : 17.00°C  |  t2= 45 ms.
Resolution= 10  -  deltaT= 190 ms.
Sensor 1 : 16.00°C  |  t2= 29 ms.
Sensor 2 : 17.00°C  |  t2= 45 ms.
Sensor 3 : 17.00°C  |  t2= 61 ms.
Resolution= 11  -  deltaT= 377 ms.
Sensor 1 : 16.00°C  |  t2= 29 ms.
Sensor 2 : 17.12°C  |  t2= 45 ms.
Sensor 3 : 17.12°C  |  t2= 61 ms.
Resolution= 12  -  deltaT= 752 ms.
Sensor 1 : 16.19°C  |  t2= 29 ms.
Sensor 2 : 17.12°C  |  t2= 45 ms.
Sensor 3 : 85.00°C  |  t2= 61 ms.

```
Here deltaT is counting millis while executing sensors.requestTemperatures().

But when sensors.getDeviceCount() after a reset is returning 0, then mostly this deltaT is 2 or 3 ms. Besides that my 3th sensor DS18B20 is giving many times also error value 85.00.

Regards,
  Jop

DMension, 2019-11-23:
Hello,
Same problem here, with ESP32 (PU 4.7K)
code/output below:

//numberOfDevices = sensors.getDeviceCount(); --> always returns zero
  numberOfDevices=2; --->hardcoded

for(int i=0;i<numberOfDevices; i++){
    // Search the wire for address
    if(sensors.getAddress(tempDeviceAddress, i)){
      Serial.print("Found device ");
      Serial.print(i, DEC);
      Serial.print(" with address: ");
      printAddress(tempDeviceAddress);
}

Output:
Locating devices...Found 2 devices.
Found device 0 with address: 28C47E45920202B9
Found device 1 with address: 288A6945921302DA
Temperature for device: 0
Temp C: 19.31 Temp F: 66.76
Temperature for device: 1
Temp C: 19.62 Temp F: 67.32

RobTillaart, 2019-11-26:
@DMension 
Can you give it a try with 3K3 or even 2K2 PU resistors? (e.g. use 2x 4K7 in parallel)
That would pull up the (3.3V) voltage faster, giving a better square wave.

DMension, 2019-11-27:
@RobTillaart 
Thanks for the suggestion, but it didn't work. I tried with a 3.3K and 2K.  Tested first with the number of devices still hard coded to make sure it could read the temp val with the lower resistor (it could in both cases) then, uncommented sensors.getDeviceCount(), but both time it came back with 0...

RobTillaart, 2019-11-27:
@DMension 
I tried to recreate your problem 
Used IDE 1.8.10
Used DTCL version 3.7.9 
Used an ESP32 Dev module (powered via an USB hub)
Used two DS18's to pin 14 
Used 3V3 to power the sensors
Used a 4K7 resistor as PU
Used wire length 10 cm, 
Used source example "multiple.ino", 
Used no parasite power
Nothing else connected

My ESP32 does provide 3.29V at the pins of the DS18's
At first I got no data indicated by -127C
Shaking the wires/sensors in the breadboard and a restart ==> it worked as it should. 

```
Requesting temperatures...DONE
Device Address: 28660D7A0200007F Temp C: 21.44 Temp F: 70.59
Device Address: 28C7E379020000E4 Temp C: 21.37 Temp F: 70.47
```

So in practice I cannot recreate the problem except for bad wiring/breadboard. 
Maybe the oneWire device search is sensitive

RobTillaart, 2019-11-27:
@DMension @JopWerff 
Which source do you use?

RobTillaart, 2019-12-04:
In issue #141  the version of the onewire library is mentioned as possible cause.

jackthese, 2019-12-04:
Posted this in another thread, but thought it might be useful here...
My previous testing was with oneWire 2.3.5 installed, according to the library manager. So here's what I did:
(1) I went to the link you provided and hit the clone button, then extracted the download to the libraries folder inside my Arduno folder in Documents. That created a folder called OneWire-master.
(2) There was also a folder in libraries called just OneWire, and I removed it, and restarted the Arduino IDE and opened the test sketch. The library manager listed OneWire 2.3.3 as installed, and I did not update it to 2.3.5, which was listed as available.
(3) I ran my test sketch with an ESP32, and it worked correctly!
(4) I then used the library manager to install 2.3.5, which it did, and then recompiled and ran, and the error happened, just like before. Now, here's where it gets weird...
(5) I tried to use the library manager to reinstall 2.3.3, but it did not change the version, nor did it give me an error message. Closing and restarting the IDE changed nothing, and the program still had the error.
(6) I then tried to install v 2.3.0, but it installed 2.3.3 instead, and the program worked again!

Please note that I'm new at all this, and the library manager is somewhat of a mystery for me.... Thanks for getting this to work--now I just have to remember not to upgrade to 2.3.5...Jack

ba58smith, 2020-05-16:
> In case these are necessary information: I wired the DS18B20 to the ESP32 using 5V (VIN of the ESP32) for the VIN of the sensor, connected the 3.3V of the ESP32 via an 4.7K resistor to the Data line and used Pin 14 for Data. Ground is of course connected as well.

Maybe this is what you meant, but the 4.7k resistor should be a pull-up resistor, between any point in the data line and 3.3V (or 5V), NOT in-line between the data leg of the sensor and pin 14.

I'm sure that doesn't have anything to do with getDeviceCount() always returning 0. Which, by the way, has anyone found a solution for? I have that problem with an ESP32, but not with ESP8266.

jackthese, 2020-05-18:
So, I went back and repeated my steps I outlined above on Dec. 4, to answer
> I'm sure that doesn't have anything to do with getDeviceCount() always returning 0. Which, by the way, has anyone found a solution for? 
Yes, I do have a solution, sort of. Specifically, I
(1) Verified that, with OneWire 2.3.5 installed,  the call "uint8_t mycount = myDallasSensors.getDeviceCount();" always returned zero , even with Dallas sensors connected.
(2) Verified that with OneWire 2.3.3 installed, the same call correctly returned the number of sensors connected (This is the solution, if you really need one).
(3) Verified that the Library Manager malfunctioned when I requested that OneWire 2.3.3 be installed instead of 2.3.5. The requested did nothing, and 2.3.5 was left in place, and there was no error message.
(4) Verified that the Library Manager malfunction when I requested that OneWire 2.3.0 be installed over 2.3.5: It installed OneWite 2.3.3 instead (which I really wanted in this case).

So, there are two problems here:
(1) OneWire 2.3.5 has an error in it that was not present in 2.3.3, and 
(2) The Library Manager malfunctions as I describe above.

Since both of these problems have existed for months on end now, I don't believe anyone is working to correct these matters. Welcome to trying to use an ESP32 in Arduinoland...

ba58smith, 2020-05-19:
@jackthese - thank you! I'm going to look at the changes between OneWire 2.3.3 and 2.3.5 to see if there's anything obvious. But ultimately, I'll probably just work around the lack of a reliable getDeviceCount() for ESP32.

jackthese, 2020-05-20:
> But ultimately, I'll probably just work around the lack of a reliable getDeviceCount() for ESP32.

Yes, that's what I'm doing: I use 2.3.5, and on developmental efforts I just loop through a maximum number of possible sensors, looking for valid addresses. When I finalize an implementation, I hard-wire the specific addresses for the sensors being used. Always works. Thanks for your feedback on this.

I'm new at all this--is there a place I should report the issue about the library manager install failures I described above?

ba58smith, 2020-05-20:
> Yes, that's what I'm doing: I use 2.3.5, and on developmental efforts I just loop through a maximum number of possible sensors, looking for valid addresses. When I finalize an implementation, I hard-wire the specific addregesses for the sensors being used. Always works. Thanks for your feedback on this.

DallasTemperature::getAddress() returns false if there are no more sensors found. So you can put that inside a while() loop to get as many sensors as it finds, and when it returns false, exit the while() loop. So you don't need to look at the maximum possible number of sensors - just getAddress() until it finds no more.
> 
> I'm new at all this--is there a place I should report the issue about the library manager install failures I described above?

Maybe here? https://github.com/arduino/Arduino/issues

jackthese, 2020-05-22:
Ah, you've detected an old FORTRAN programmer disability! I wrote my first program at Carnegie Tech in 1963 in a language called ALGO (precursor to Algol). It had a "for" loop in it. It ran on a Bendix G10 computer that used vacuum tubes, not transistors. At least there were no go to statements in it.

Having admitted that, I do use while statements where they make sense (for me). I'm not an expert at the Arduino C++ variant, for sure. Is there a way to dynamically allocate memory for the sensor addresses as I find them, so I don't have to pick a maximum number for that?

ba58smith, 2020-05-22:
> <snip> Is there a way to dynamically allocate memory for the sensor addresses as I find them, so I don't have to pick a maximum number for that?

You don't need to worry about allocating memory, for the OneWire addresses, or anything else in your Arduino code, for that matter. It just happens. Here's the while() loop I'm using to find the address of each detected sensor - it works for any number.

```
  bool check_again = true;
  uint8_t sensor_index = 0;
  while (check_again) {
    if(sensors->getAddress(addr, sensor_index)) { //addr is DeviceAddress
      std::copy(std::begin(addr), std::end(addr), std::begin(owda)); //see below for what owda is
      known_addresses.insert(owda); //see below for known_addresses
      char addrstr[24];
      owda_to_string(addrstr, owda);
      debugI("Found OneWire sensor %s", addrstr);
      sensor_index++;
    }
    else check_again = false;
  }
```

owda is one of these:
const OWDevAddr null_ow_addr = {0, 0, 0, 0, 0, 0, 0, 0};

known_addresses is one of these:
std::set<OWDevAddr> known_addresses;

Hope this is helpful.

jackthese, 2020-05-22:
> Hope this is helpful.

It will be. I see a number of places where I can improve my implementation. For many years I used compilers that had no index checking on arrays during execution, and I'm still paranoid about explicitly allocating memory I might use. Here's my code as an fyi. I'll be modifying it as I learn a bit about what I see in your code.
(this code is from a .h tab in the Arduino IDE)
```
#include <OneWire.h>
#include <DallasTemperature.h>                   // Temp sensor library
#define NUM_DAL 3
DeviceAddress FoundSensors[NUM_DAL];             // used in code to read sensor addresses
OneWire myOneWire(ONE_WIRE_BUS);                 // Setup onewire instance
DallasTemperature myDallasSensors(&myOneWire);   // create DallasTemperature instance`

```
(this code is in a .cpo tab):

```
void SetUpDallas() {
  Serial.printf("\nOne Wire Bus set up to pin %d", ONE_WIRE_BUS);
  myDallasSensors.begin();                                        // initialize Dallas instance

  uint8_t mycount = myDallasSensors.getDeviceCount();             // see if getDeviceCount works
  Serial.printf("My Device Count from getDeviceCount: %d\n", mycount);

  Serial.print("Searching for Dallas sensor addresses by index\n");
  DeviceAddress tempaddr;
  NumDallasActive = 0;
  for (uint8_t j = 0; j < NUM_DAL; j++) {                         // loop to find addresses of sensors connected
    if (myDallasSensors.getAddress(tempaddr, j)) {                // true if sensor found
      NumDallasActive++;
      AssignSensor(FoundSensors[j], tempaddr);
      Serial.printf("Sensor %d address: ", j);
      printAddress(FoundSensors[j]); Serial.println();
      myDallasSensors.setResolution(FoundSensors[j], DalResBits); // set and confirm resolution to 12 bits
      uint8_t ReturnedRes = myDallasSensors.getResolution(FoundSensors[j]);
      float DegFBit = 1.8 * 256 / pow(2, DalResBits);             // 12 bits: temps to better than 0.1 deg F
      Serial.printf("Resolution set and confirmed to %.3f DegF/bit\n", DegFBit);
    }
  }
  myDallasSensors.requestTemperatures();                          // initiate call for temps
}

void AssignSensor(DeviceAddress Dest, DeviceAddress Source) {     // move a DeviceAddress
  for (uint8_t j = 0; j < 8; j++) {
    Dest[j] = Source[j];
  }
}

void printAddress(DeviceAddress devAddr) {                        // print address formatted for cut and paste
  Serial.print("0x");
  for (uint8_t i = 0; i < 8; i++) {
    if (devAddr[i] < 16) Serial.print("0"); Serial.print(devAddr[i], HEX);
    if (i < 7) Serial.print(", 0x");
  }
}

```

ba58smith, 2020-05-22:
That obviously works, but if you check the return value of getAddress(), you don't have to know NUM_DAL.

RobTillaart, 2021-10-06:
@omski 
Thanks for your code, if I understand correctly you implemented an ```uint getDeviceCount()``` outside the library. [similar to the approach proposed by @ba58smith  of 20 may 2020 above ]

Your algorithm main worker is the line
```cpp
while (sensors.getAddress(tempaddr, NumDallasActive)) ...
```
As long as an index [0.. n] returns an address the search continues in the end resulting in a number of DS18B20 devices.
Code wise it seems correct (I did not verify it working) , however a few remarks should be placed. 

```sensors.getAddress(tempaddr, NumDallasActive)``` does a search from the start for every call.  This results in a O(n^2) algorithm where the loop in the library in ```begin()``` is O(n). For small numbers no problem, for larger numbers it might be. 

As the ATCL library still holds the variable **devices** (==0) internally, and uses this in a number of functions e.g. ```setResolution()``` these functions still do not work as intended. 

A solution to the problem COULD be integrating your **getDeviceCount()** into the library into ```sensors.begin()``` e.g. if the compiler detects an ESP platform by means of an **#ifdef**. Then the internal **devices** variable could be set and the functions will work as intended. It would not solve the root cause of the problem, it would be a work around solution but it would make the functions working again.

That said it is good to see some code that is working, and the fact that this is working could help to pinpoint the root cause. 

Thanks,

omski, 2021-10-06:
Yes I used the approach from above. I tried to reimplement the method as your proposed but for some reason it did not work. If I have some more time to spend I will have a closer look at it.
After lots of frustration I just came up with this code which did the work for me as I needed a device which can handle a variable amount of sensors and displays the measured values.
[image]

gitkomodo, 2021-10-07:
> I ran into this also and solved it. Maybe this will be helpful for someone: https://github.com/omski/DS18B20_TTGO

Hmm, your algorithm
 
```c++
while (sensors.getAddress(tempaddr, NumDallasActive)) ...
```

doesn't seem to be much different from the begin() procedure. That procedure checks a few more things, but I think it should come up with at least one device if your alternative works.

I suspect the ESP or sensor(s) aren't ready/powered-up when begin() is called. Consequence could be that the call to reset_search() and/or first search() on the OneWire is missed/fails. If this **theory** is correct, a delay before the call to begin() might solve the issue.

gitkomodo, 2021-10-08:
The double call to begin() was also mentioned in [this comment](https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/168#issuecomment-763158681) on DallasTemperature issue #168.

The root of the problem seems to be in het OneWire library. A more fundamental solution seems to be in [this comment](https://github.com/PaulStoffregen/OneWire/issues/57#issuecomment-450151098) on OneWire issue PaulStoffregen/OneWire#57.

If I remember correctly there will not be any specific changes to OneWire to support ESP. As there are multiple DallasTemperature issues for this problem, it might be worth it to add the solution above to the library.

rvdende, 2023-01-08:
I ran into this issue with the Arduino ZERO, or actually the Wemos clone board. It has the ARM M0 chip. In anycase, after looking at the adafruit feature m0 docs I found that the only pin I could get OneWire to work on is pin 5.

```c
#define ONE_WIRE_BUS 5
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
```

Also make sure to use the 3.3v power source not the 5v on the ARM boards, and yes you have to use the 4.7k ohm pullup between data and power.

EmileSpecialProducts, 2023-06-07:
I have done this to resolve it:

  //sensorscount=sensors.getDeviceCount(); // this dus not work 
  sensorscount=0;
  while ( sensors.getAddress(sensorsdeviceAddress, sensorscount))
  {
    sensorscount++;
  }

parsdc, 2024-07-22:
**deviceCount** is the answer

```
   sensors.requestTemperatures();  
    while (tempC[0] =! (-127) ){
      tempC[0] = sensors.getTempCByIndex(0);
      deviceCount++;
    }
    deviceCount--;
```