milesburton/Arduino-Temperature-Control-Library #86: int16_t DallasTemperature::millisToWaitForConversion(uint8_t)' is private

jeffersonpenteado, 2017-08-15:
Whenever I try to compile this message it appears, Can someone help?

arduino 1.8.3
dallas temperature 3.7.6
onewire 2.3.3

RobTillaart, 2017-08-15:
Which code are you trying to compile? For which hardware platform?

mfalkvidd, 2017-10-02:
Related to https://github.com/milesburton/Arduino-Temperature-Control-Library/pull/72

Akubi, 2017-10-28:
Version 3.7.7 is available but Arduino still look for wersion 3.7.6 ???

mfalkvidd, 2017-10-28:
@Akubi 3.7.6 is the latest release according to https://github.com/milesburton/Arduino-Temperature-Control-Library/releases

Akubi, 2017-10-28:
Yes sure. But the source version on master is the 3.7.7
Version 3.7.7 (master) solve the #72 issue, but on release we still have the 3.7.6

mfalkvidd, 2017-10-29:
@Akubi the Arduino IDE only looks for released versions.

Akubi, 2017-10-29:
Yes this is the problem. Why do not put the last version as released ?

milesburton, 2017-12-23:
Fixed it latest release

closed 2017-12-23 by milesburton