milesburton/Arduino-Temperature-Control-Library #87: Reading temperature from multiple sensors - Question/Feature request

lazarvgd, 2017-09-04:
Hi, Is there any option to read temperature from two or more sensors using just one pin on arduino?

I have found [the solution](https://www.tweaking4all.com/hardware/arduino/arduino-ds18b20-temperature-sensor/), but with this library it is far easier.

I don't want to be rude, but if the library has or will has this feature that would be awesome.

Thanks

RobTillaart, 2017-09-04:
Yes the library has this feature support multiple ds18's on one datapin.
Have a look at the example  https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Multiple/Multiple.pde 
It shows two sensors on one pin.

lazarvgd, 2017-09-04:
Thank you very much, this was very helpful. :+1:

closed 2017-09-04 by lazarvgd