milesburton/Arduino-Temperature-Control-Library #88: Reading problem / sensor sensitivity

lazarvgd, 2017-09-04:
Hi, I am using this library and I have observed that it has very good precision. The problem is occurring if I heat sensor up with hand and then release it to cool down to room's temperature, it takes about 7-10 mins.

For example, I have heated up to 30 deg. C with hand and measured time until sensor reads 26 deg. C. The time was ~10min.

Can you please tell me is this normal or I am doing something wrong? With code from question #87  I got instant measurements after releasing the sensor and leave it to room temperature.

thanks

RobTillaart, 2017-09-05:
There are different ds18b20 bodies (> 10) which all have a different heat capacity as the different sizes have more or less mass causing some latency when heating up / cooling down. E.g. the watertight reacts slower than the "normal" one as the watertight enclosure also works as isolation.  check the [datasheet](https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf) if it mentions details.

The speed of cooling depends on the difference in temperature of the environment and the temperature of the sensor itself. You can see this if you plot the values of a cooling down / heat up in a graph.

lazarvgd, 2017-09-05:
Thanks Rob. Sorry about this question.

closed 2017-09-05 by lazarvgd