milesburton/Arduino-Temperature-Control-Library #90: ESP8266 difficulties

MrTrick, 2017-10-25:
I'm having some trouble getting the library to read a DS18B20+PAR with a Wemos D1 Mini. (ESP8266EX)
The code (a close copy of the 'simple' example) works fine if run instead on an Arduino Pro Micro.

On the Wemos D1 Mini I just get:
`Starting temperature read...done. Temperature is -127.00
Starting temperature read...done. Temperature is 85.00
Starting temperature read...done. Temperature is -127.00
Starting temperature read...done. Temperature is 85.00
...`

If I detach the 4.7K pullup resistor, no change.
If I attach the pullup resistor to 5V instead of 3.3V I instead get:
`Starting temperature read...done. Temperature is 127.94
Starting temperature read...done. Temperature is 127.94
...`

Code is below.
(The +PAR variant is ONLY capable of parasitic mode, and pin 3 is NC)

Can you please advise? I do not know what the temperatures indicate.

`#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port D6 on the Wemos D1
#define ONE_WIRE_BUS D6
// Setup a oneWire instance to communicate with any OneWire devices
OneWire oneWire(ONE_WIRE_BUS);
// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);
/*
 * The setup function. We only start the sensors here
 */
void setup(void)
{
  // start serial port
  Serial.begin(115200);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // Start up the library
  sensors.begin();
  sensors.setWaitForConversion(false);
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  if (1!=sensors.getDeviceCount()) {
    Serial.print("Did not find 1 device, found "); 
    Serial.println(sensors.getDeviceCount());
    return;
  }
  if (!sensors.isParasitePowerMode()) {
    Serial.println("Expected parasite mode");
    return;
  }

  Serial.print("Starting temperature read...");
  sensors.requestTemperatures();
  delay(1000);
  Serial.print ("done. Temperature is ");
  float T = sensors.getTempCByIndex(0);
  Serial.println(T);
}`

RobTillaart, 2017-10-25:
Is this Wemos D1 Mini a 3.3V device? If so the 3.3V might be too low (e.g 3.0 V effective) to "feed" the sensor reliably. Can you measure the actual voltage?
According to the datasheet 3.3V should be sufficient.
https://datasheets.maximintegrated.com/en/ds/DS18B20-PAR.pdf

Still a bidi levelconvertor like  https://www.sparkfun.com/products/12009  might help to

MrTrick, 2017-10-25:
It's 3.3V with a 5V=>3.3V regulator, so both voltages are available on board.  
(Pins are 3.3V logic with ~12mA source capacity, and are 5V tolerant)

I measured the steady voltage of the regulator to be 3.3V. 
Capturing the signal to check for sag during that communication will be tricky, but I'll give it a try.

Also the test is just on breadboard, wire length is <10cm.

MrTrick, 2017-10-26:
I amended my code to just show raw scratchpad data;
```
void loop(void)
{ 
  Serial.print("Starting temperature read...");
  sensors.requestTemperaturesByAddress(addr);
  delay(1000);
  Serial.println("done.");
  //--------------------------------------------------------
  Serial.print("Raw scratchpad is: ");
  if (!sensors.isConnected(addr,pad)) { Serial.println("ERROR bad connection"); return; }  
  char buf[27];
  sprintf(buf,"%02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X:%02X",pad[0],pad[1],pad[2],pad[3],pad[4],pad[5],pad[6],pad[7],pad[8]);
  Serial.println(buf);
  //--------------------------------------------------------
}
```

# 5V Behaviour
 
With a 5V pullup I get a constant value;
```
Raw scratchpad is: FF:07:4B:46:7F:FF:01:10:2F
Starting temperature read...done.
Raw scratchpad is: FF:07:4B:46:7F:FF:01:10:2F
...
```
Not sure what that value is supposed to be, but it doesn't change when the thermometer is held. (as I saw on the pro micro platform) 

I've tried looking at the voltage of the data line with a DSO:  [5V Pullup](https://i.imgur.com/rhX4P1N.gif)  
The pullup voltage during the conversion time stays at 5V, I think there's not a power supply problem.

# 3.3V Behaviour

If I use a 3.3V pullup I get stranger behaviour;
3 times in a row through loop() I get "ERROR: Bad Connection", then the 4th time returns that same scratchpad value as above, and the cycle repeats.

The DSO shows the data line's voltage is at a steady ~1.6V during those 3 bad reads, then it jumps up to 3.3V for the one that reads successfully. [3.3V Pullup](https://i.imgur.com/g6wZmfe.gif)

I'm getting very confused!

MrTrick, 2017-10-26:
More testing, found something that works; reducing the pullup resistor value.
Not sure exactly what's going on, but it's working properly now. Thanks for your library. :-)

[image]

[image]

[image]

closed 2017-10-26 by MrTrick

RobTillaart, 2017-10-26:
By lowering the value of the resistor the DS18B20 gets more power/energy to load itself. Apparently with 1K5 it gets enough to do the actual measurement, the math and IO internally to give a decent response.