milesburton/Arduino-Temperature-Control-Library #92: Only one DS18B20 is read

tatulaiot, 2017-11-06:
Hi, I have been struggling for several days , cannot find the solution to this issue : 
Device ESP8266 (3.3v) USB and externally powered reading 2 Dallas DS18B20, they work in single mode ( one at the time ) tested with 10K, 4K7, 3K3 , 2K2 pull up resistors. I am not using parasite mode. 
The only improvement I found was to place these 2 lines before initializing : 
        digitalWrite(2, HIGH); // <- incantation code: same number as OneWire port
        pinMode(2, OUTPUT); //<- incantation code: same number as OneWire port

With those to lines I managed to discover 2 deivces but only one of them is read.
The first one reply its memory address, and proper data. The second one report 000000000 memory address and of course (as per library definition) temp -127 . 
Also tested with Method 1. 

I have tried almost everything but no success to make it work with 2 sensors. 

This is the declaration. 

#include <OneWire.h>
#include <DallasTemperature.h>

#define MAX_TEMP_DEVIATION 10
#define MAX_DS18B20_SENSORS 5

/// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature.
DallasTemperature sensors(&oneWire);

// arrays to hold device addresses
DeviceAddress insideThermometer, outsideThermometer;

and the init function ...  

int init_DS18B20(void){

        digitalWrite(2, HIGH); // <- incantation code: same number as OneWire port
        pinMode(2, OUTPUT); //<- incantation code: same number as OneWire port
        Serial.print("Locating devices...");
        sensors.begin();
        Serial.print("Found ");
        Serial.print(sensors.getDeviceCount(), DEC);
        DS18B20_sensor_config.sensors_conunt = sensors.getDeviceCount();
        Serial.println(" devices.");
        // report parasite power requirements
        Serial.print("Parasite power is: ");
        if (sensors.isParasitePowerMode()) Serial.println("ON");
        else Serial.println("OFF");
        // Assign address manually. The addresses below will beed to be changed
// to valid device addresses on your bus. Device address can be retrieved
// by using either oneWire.search(deviceAddress) or individually via
// sensors.getAddress(deviceAddress, index)
// Note that you will need to use your specific address here
//insideThermometer = { 0x28, 0x1D, 0x39, 0x31, 0x2, 0x0, 0x0, 0xF0 };

// Method 1:
// Search for devices on the bus and assign based on an index. Ideally,
// you would do this to initially discover addresses on the bus and then
// use those addresses and manually assign them (see above) once you know
// the devices on your bus (and assuming they don't change).
  //      if (!sensors.getAddress(insideThermometer, 0)) Serial.println("Unable to find address for Device 0");

// method 2: search()
// search() looks for the next device. Returns 1 if a new address has been
// returned. A zero might mean that the bus is shorted, there are no devices,
// or you have already retrieved all of them. It might be a good idea to
// check the CRC to make sure you didn't get garbage. The order is
// deterministic. You will always get the same devices in the same order
//
// Must be called before search()
oneWire.reset_search();
// assigns the first address found to insideThermometer
if (!oneWire.search(insideThermometer)) Serial.println("Unable to find address for insideThermometer");

// show the addresses we found on the bus
        Serial.print("Device 0 Address: ");
        printAddress(insideThermometer);
        Serial.println();
        delay (1000);
        Serial.print("Device 1 Address: ");
        printAddress(outsideThermometer);
        Serial.println();

// set the resolution to 9 bit (Each Dallas/Maxim device is capable of several different resolutions)
        sensors.setResolution(insideThermometer, 9);
        sensors.setResolution(outsideThermometer, 9);

        Serial.print("Device 0 Resolution: ");
        Serial.println(sensors.getResolution(insideThermometer), DEC);
        Serial.print("Device 1 Resolution: ");
        Serial.println(sensors.getResolution(outsideThermometer), DEC);
        Serial.println();

        return 0;
}

RobTillaart, 2017-11-07:
Some questions:
- Do the sensors work with a standard 5V Arduino board?
- Can you post the output of the program above?
- Do the sensors get hot?

80stech, 2017-12-12:
Maybe wires are too long ?

milesburton, 2017-12-23:
Nothing heard, reopen if it's still a problem

closed 2017-12-23 by milesburton