milesburton/Arduino-Temperature-Control-Library #94: multiple sensors mixed readings

80stech, 2017-12-12:
I am using 4  DS18B20  to control my fridge and using getTempCByIndex to read them every 5 seconds.  Everything works great except that every so often (maybe 10 -30 min.)  I get a reading from the next in line sensor instead of the one being read.  Thanks :)

RobTillaart, 2017-12-12:
You might need to check your wiring. 
You should fetch all addresses and hard code every address with location, more reliable.

80stech, 2017-12-12:
Thanks Rob, :)  I was going originally going to hard code the addresses but I took the easy way out! ;)  I tried a few other things last night and found that the reading can come from either neighboring address, not only the one after it.  I am not an expert, but  I am having trouble wrapping my head around how that can even happen!

RobTillaart, 2017-12-12:
By far the easiest way, if you have enough pins, is to put every sensor on its own pin. Then the addresses are unambiguous.

80stech, 2017-12-13:
Problem solved!  kinda....  Loading the addresses with getAddress into getTempC with a DeviceAddress variable solves the problem! :)  but I can't get it working by putting the address into getTempC manually?? Thanks for the help! :) Cheers

closed 2017-12-22 by milesburton