milesburton/Arduino-Temperature-Control-Library #101: requestTemperatures() doesn't seem to work with multiple sensors in parasite mode

sidoh, 2017-12-22:
Hi,

Thanks for this great library!

I'm running into an interesting problem where the `requestTemperatures()` method (the one without specifying a device index or address) causes incorrect readings.  

Setup:

1. Two DS18B20s in parasite mode (i.e., Vin pin connected to Gnd)
2. 4.7 KΩ resistor connecting data line and +5V
3. ESP8266 chip, data line connected to D1

What I've noticed:

1. Parasite mode is detected accurately
1. Device IDs are read correctly
1. When in parasite mode, using `requestTemperatures()` and then reading temps with `getTempC(index)` produces funky readings.  Seems like one of the sensors reads 80 C, the other reads 40 C.  The correct reading is ~29 C.
1. The exact same sketch works when I disable parasite mode (e.g., connect the Vin pin to +5V instead of Gnd).  Both sensors read ~29 C.
1. Using `requestTemperaturesByAddress(addr)` on each individual sensor produces the correct results (both sensors reading ~29 C).

Happy to help debug if I can, but thought I'd toss up an issue in case this is a known bug or I'm doing something silly.

milesburton, 2017-12-22:
Hmm could be a voltage drop between polls.

You could try and drop the resistor value and see what effect that has.

sidoh, 2017-12-22:
Interesting.  Dropped to a 2.2 KΩ and it works!  Good call.

I think I'll stick with polling individual addresses since it seems a little more robust.

Obviously not a software issue, though.  Thanks very much for the quick response :)

closed 2017-12-22 by sidoh