milesburton/Arduino-Temperature-Control-Library #106: Can not compile example with ver. 3.7.6

bill-orange, 2018-01-26:
I tried compiling my weather station sketch with the new ver. 3.7.6 and got errors so I retested with the simple example sketch and received generally the same errors.  Ver 3.7.5 works fine both in my sketch and the example.  I am using the latest board manager 2.4.0.  Perhaps that is the source of the problem.

Advice?

Here is the output of the compiler:

```
Archiving built core (caching) in: C:\Users\William\AppData\Local\Temp\arduino_cache_442829\core\core_esp8266_esp8266_nodemcu_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_366d5b04ba07ad9d4558d29603ae96db.a
libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::depower()':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:163: multiple definition of `OneWire::reset()'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:163: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::search(unsigned char*, bool)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:197: multiple definition of `OneWire::write_bit(unsigned char)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:197: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::read_bit()':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:225: multiple definition of `OneWire::read_bit()'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:225: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::write(unsigned char, unsigned char)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:249: multiple definition of `OneWire::write(unsigned char, unsigned char)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:249: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::write_bytes(unsigned char const*, unsigned short, bool)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:263: multiple definition of `OneWire::write_bytes(unsigned char const*, unsigned short, bool)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:263: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::read()':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:277: multiple definition of `OneWire::read()'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:277: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::read_bytes(unsigned char*, unsigned short)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:287: multiple definition of `OneWire::read_bytes(unsigned char*, unsigned short)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:287: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::select(unsigned char const*)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:296: multiple definition of `OneWire::select(unsigned char const*)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:296: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::skip()':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:308: multiple definition of `OneWire::skip()'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:308: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::depower()':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:314: multiple definition of `OneWire::depower()'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:314: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::reset_search()':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:328: multiple definition of `OneWire::reset_search()'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:328: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::OneWire(unsigned char)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:145: multiple definition of `OneWire::OneWire(unsigned char)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:145: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::OneWire(unsigned char)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:145: multiple definition of `OneWire::OneWire(unsigned char)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:145: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::target_search(unsigned char)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:343: multiple definition of `OneWire::target_search(unsigned char)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:343: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::search(unsigned char*, bool)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:368: multiple definition of `OneWire::search(unsigned char*, bool)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:368: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::crc8(unsigned char const*, unsigned char)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:526: multiple definition of `OneWire::crc8(unsigned char const*, unsigned char)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:526: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::crc16(unsigned char const*, unsigned short, unsigned short)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:568: multiple definition of `OneWire::crc16(unsigned char const*, unsigned short, unsigned short)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:568: first defined here

libraries\DallasTemperature\OneWire.cpp.o: In function `OneWire::check_crc16(unsigned char const*, unsigned short, unsigned char const*, unsigned short)':

C:\Users\William\Documents\Arduino\libraries\DallasTemperature/OneWire.cpp:562: multiple definition of `OneWire::check_crc16(unsigned char const*, unsigned short, unsigned char const*, unsigned short)'

libraries\OneWire-master\OneWire.cpp.o:C:\Users\William\Documents\Arduino\libraries\OneWire-master/OneWire.cpp:562: first defined here

collect2.exe: error: ld returned 1 exit status
```

RobTillaart, 2018-01-26:
You have two times the OneWire library installed. The cause is probably because it is now incuded with the DT lib. Move one outside the library path and try again.

milesburton, 2018-01-29:
An additional OneWire library has been included. We'll reconcile this shortly. Apologies.

closed 2018-01-29 by milesburton