milesburton/Arduino-Temperature-Control-Library #108: Addition of OneWire.cpp/h files creates link issue with existing code.

davidlehrian, 2018-01-29:
After updating to the latest release version of the DallasTemperature for the Arduino IDE my application could no longer link.  I get the following error:  

OneWire.cpp.o (symbol from plugin)*: In function OneWire::write(unsigned char, unsigned char)
(.text+0x0)*: multiple definition of OneWire::write(unsigned char, unsigned char)
OneWire.cpp.o (symbol from plugin)*: (.text+0x0): first defined here

A little looking around shows that OneWire.cpp/h were added to the DallasTemperature library.  I see there were a couple of new architectures defined and that is great, but they should be added to the OneWire library https://github.com/PaulStoffregen/OneWire instead of forking and adding them to this library.

milesburton, 2018-01-29:
+ @PaulStoffregen

PaulStoffregen, 2018-01-29:
Agreed, a duplicate copy of OneWire inside this library is just asking for compile errors when the Arduino IDE detects which libraries are needed and tries to compile both this lib and OneWire.

stickbreaker, 2018-02-02:
@PaulStoffregen I agree, This library should not have `OneWire()` inside it. It should just document that it is required and point a link to the master at [PaulStoffregen/OneWire](https://github.com/PaulStoffregen/OneWire)

Chuck.

PaulStoffregen, 2018-02-02:
Or direct people to use the Library Manager to install it the easy way.

per1234, 2018-02-07:
I agree that it's best not to have a duplicate copy of the OneWire library in this library. If for some reason it is not possible to immediately remove it there is a quick fix for the compilation error caused by having both this library and the OneWire Library installed. Simply change the order of the includes in the example sketches:
```arduino
#include <DallasTemperature.h>
#include <OneWire.h>
```
Including OneWire.h after DallasTemperature.h will cause this library's copy of OneWire to have include priority, as intended, thus the error is fixed.

milesburton, 2018-02-16:
All agreed. Resolved throughhttps://github.com/milesburton/Arduino-Temperature-Control-Library/releases/tag/3.8.0

Will leave this as a pre-release should there be any faults

aufdenkampe, 2018-03-05:
Thanks for the fix to this with release 3.8.0. It allowed us to resolve our issue https://github.com/EnviroDIY/Libraries/issues/6

closed 2018-03-28 by milesburton