milesburton/Arduino-Temperature-Control-Library #111: Wrong temperature on low (< 5C) degrees

Marvo2011, 2018-03-15:
Hello,

the library return wrong values at low (< 5C) degrees. ist show values like 50C but it should 1 or 2C. All about 5C is working correctly.

Thank for Help

RobTillaart, 2018-03-15:
Very strange that nobody noticed this for you. 
How many sensors did you try?
What is the exact type of sensor?
Did this happen with the parasitic mode or normal mode?
Do you use a correct pull up resistor?
Does it also happen with the demo sketches?
What platform do you use?

Marvo2011, 2018-03-15:
I have try 2 sensors same problem here.
The sensor type is the DS18B20 (with long cable and water proof).
I have use the normal mode.
I have use a 4.7kO pull up resistor at 3V
Yes, also in the demo sketches.
The Arduino IDE, and all is connected to a ESP8266.

RobTillaart, 2018-03-15:
Try a 2K2 pull up, at 3V the sensor is near the edge of its working range.
Check datasheet to be shure DS18B20 works at 3V

Marvo2011, 2018-03-15:
I will try the 2K2 resistor, the datasheet says -55°C to +125°C; VDD = 3.0V to 5.5V

RobTillaart, 2018-03-15:
So 3V is at the edge as I thought. 
Can youmeasure the voltage with a DMM?

milesburton, 2018-03-15:
I agree with Rob, coupled with the high value resistor and the resistance of the wire @3v I'm surprised you're getting a correct value at room temperature

closed 2018-05-31 by milesburton