milesburton/Arduino-Temperature-Control-Library #112: getTempFByIndex

Tannoo, 2018-04-09:
This is giving me the same reading (20) as getTempCByIndex. What am I doing wrong?

```c++
  sensors.requestTemperatures();
  water_temp = (sensors.getTempCByIndex(0));
  Serial.print("Temperature is: "); 
  Serial.println(water_temp);
```

Tannoo, 2018-04-09:
```C++
  sensors.requestTemperatures(); // Send the command to get temperature readings 
  water_temp = (sensors.getTempCByIndex(0));
  Serial.print("Temperature is: ");
  Serial.println(water_temp);
```

Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18

VS

```C++
  sensors.requestTemperatures(); // Send the command to get temperature readings 
  water_temp = (sensors.getTempFByIndex(0));
  Serial.print("Temperature is: ");
  Serial.println(water_temp);
```

Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18

Both are the same.. What gives?

RobTillaart, 2018-04-09:
Very strange, as it is not the wole sketch I can only guess. (Better discuss whole sketch on Arduin.cc forum)

Can you do the test this way?
```C++
sensors.requestTemperatures(); // Send the command to get temperature readings 
  water_temp = (sensors.getTempFByIndex(0));
  Serial.print("Temperature F is: ");
  Serial.println(water_temp);
  water_temp = (sensors.getTempCByIndex(0));
  Serial.print("Temperature C is: ");
  Serial.println(water_temp);
```
What platform are you using?

Tannoo, 2018-04-09:
Arduino. I can do that.

Okay, I have something else going on... I'm not sure what.

This is the function in the larger program and the only spot for this temp reading:
```C++
void readTemp() {
  sensors.requestTemperatures(); // Send the command to get temperature readings 

  water_temp = (sensors.getTempFByIndex(0));
  Serial.print("Temperature Farenhiet is: ");
  Serial.println(water_temp);
  water_temp = (sensors.getTempCByIndex(0));
  Serial.print("Temperature Celcius is: ");
  Serial.println(water_temp);
  safe_delay(100);
}
```
I'm still getting this output and I don't have a `Serial.print("Temperature is: ");` anywhere in my program now.

Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18
Temperature is: 18

Tannoo, 2018-04-09:
I am going to restart Arduino IDE and see what happens.

milesburton, 2018-04-09:
Best to move this to the Arduino forums, this is intended for issues with the library itself

closed 2018-04-09 by milesburton

Tannoo, 2018-04-09:
BTW, I did find that the Mega2560 I was using, had 'lost it's marbles'. Needs bootloader re-loaded.

Tannoo, 2018-04-09:
This library is working fine with another processor. Thank you for helping.