milesburton/Arduino-Temperature-Control-Library #113: ESP8266 parasitic DS18B20 reset during conversion (85°C)

LeoDJ, 2018-05-06:
### Description
I was running two `DS18B20+PAR` temperature sensors without any problems for the last two years on an ESP-01 module running on 3.3V using a 4.7K pullup resistor.

Now, wanting to develop a new firmware I started the development with new versions of everything (Arduino IDE and libraries), but now, with the exact same hardware setup that worked flawlessly for two years, the sensor always returns 85°C (error code for reset during conversion). I suspect that somehow, I don't know how, the ESP does not leave the data pin powered anymore.

To test this, I recompiled and flashed my old firmware that was running fine for two years and lo and behold, it still returned 85°C. Trying to isolate the issue, I flashed the old binary of the firmware and it returned the correct temperatures again.

To isolate the issue further, I tried numerous versions of the OneWire (2.3 to 2.3.2) and [DallasTemperature](https://github.com/milesburton/Arduino-Temperature-Control-Library) (3.7.3 to 3.7.9) libraries, but to no effect. I am still getting 85°C all the time.

I also tried to compile it using Arduino IDE and PlatformIO, but both yielded the same issue still.
Trying on another ESP8266 (Wemos D1 mini and nodeMCU) did not change anything.

Slowly I am suspecting that the issue lies in the ESP8266-Arduino framework updating something or a change in the compiler, which now breaks keeping the data pin configured as output during the conversion.

A workaround, of course, is to use a lower value pullup resistor so that the sensors take power through that instead of the powered microcontroller pin. (One parasitic sensor requires 1K, two 470 Ohms) But this is not a solution.

I will update this issue in the following days by testing on other platforms (older Arduino IDE, older Arduino-ESP8266 version, different OS) and hooking a oscilloscope up to the data line to see if the issue really lies in voltage sag during conversion.
Also, I want to try out the library fork from [orgua](https://github.com/orgua/OneWire) to see if the issue happens in his library too.

So, to conclude: I can rule out a hardware isse, I think it's not an issue with library versions and I suspect something in the ESP8266-Arduino framework or Arduino IDE (compile options).

I am looking forward to help fix this issue soon. :)

### Steps To Reproduce Problem
Try to read temperatures with an ESP8266 and a parasitic DS18B20 with 4.7K pullup. Using the newest Arduino IDE, ESP8266-Arduino, OneWire and DallasTemperature versions.

### Hardware & Software

**Board:** ESP8266 (any module, tried ESP01, Wemos D1 mini, nodeMCU
No shields
**Arduino IDE version:** 1.8.5
**Version info & package name (from Tools > Boards > Board Manager):** 2.3.4 (but tried all other versions)
**Operating system & version:** Windows 10
**Any other software or hardware?** Dallas DS18B20+PAR connected with a 4.7K pullup

### Arduino Sketch
Happens with every sketch, not limited to this one ([Simple Example](https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Simple/Simple.pde))
```cpp
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

/*
 * The setup function. We only start the sensors here
 */
void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // Start up the library
  sensors.begin();
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  // call sensors.requestTemperatures() to issue a global temperature 
  // request to all devices on the bus
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  // After we got the temperatures, we can print them here.
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  Serial.print("Temperature for the device 1 (index 0) is: ");
  Serial.println(sensors.getTempCByIndex(0));  
}
```

### Errors or Incorrect Output
It returns 85°C all the time.

Looking forward to help fix this issue soon.

I already posted [this issue on the OneWire repository](https://github.com/PaulStoffregen/OneWire/issues/58), but I received no reaction whatsoever. I hope there will be more feedback here.

alex1702, 2018-05-16:
i have the same issue. i use a wemos d1 mini pro.

alex1702, 2018-05-16:
my temporary solution:
```c
ds18b20.setWaitForConversion(false);
ds18b20.requestTemperatures();

pinMode(DS18B20_PIN, OUTPUT);
digitalWrite(DS18B20_PIN, HIGH);
delay(750);
Serial.println(ds18b20.getTempCByIndex(0));  
```

now i get the right temperature.

LeoDJ, 2018-05-16:
Yes, I tried that too and it is a minimally less dirty workaround...
I would still like to fix it in the library, but like I already said [here](https://github.com/PaulStoffregen/OneWire/issues/58#issuecomment-386846027), I have no idea what could be broken...

milesburton, 2018-05-31:
Hi @LeoDJ, that' sounds rather strange though I did see some precision issues myself - though I discovered they were related to power issues.

I'm guessing this is related to the updated timings in both libraries. I'd ordinarily use a lower resister value (as you've demonstrated). 4.7 sounds a bit high for 3.3 - if memory serves that runs it below the operating voltage (cap charging excluded)

@PaulStoffregen Notifying you. I need to get my gear out to test the issue that's been raised.

Thanks for the thorough post. I'll hunt around for my ESP + Maxin ICs and see if I can find out what the issue is.

PaulStoffregen, 2018-05-31:
I do not directly support ESP8266.  I can not get involved in this one.

milesburton, 2018-05-31:
Thanks @PaulStoffregen for your expedient response. We'll do some digging

LeoDJ, 2018-05-31:
I also noticed, that when I use the standard `pinMode` function, like [alex used above](https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/113#issuecomment-389638589) but at the end of the OneWire `write`, the GPIO will get set as an output. Maybe the implementation of the `DIRECT_MODE_INPUT` and `DIRECT_MODE_OUTPUT` in the [OneWire library](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h#L94) is not correct anymore. I could not find it in the Arduino-ESP8266 framework for comparison

mmithril, 2019-04-04:
See #130.

gitkomodo, 2020-02-28:
Maybe this is related to a timing issue described in (and possibly resolved by) this commit  https://github.com/stickbreaker/OneWire/commit/79a534e80371313e3db287f725e3bf3db10f3ead  It is related to ESP32 so may be a different issue, but I thought it might be worth to mention it.

pstolarz, 2020-06-01:
@LeoDJ have you solved your issue? If not, could you check [my fork](https://github.com/pstolarz/Arduino-Temperature-Control-Library) of the DallasTemperature lib and let me know about results?  This fork doesn't use OneWire as 1-wire service but my alternative of this library. Maybe it will help. Pull-up resistor shouldn't be issue here if proper strong pull up is implemented by the service lib. Unfortunately I don't have ESP8266 to check it by myself.

CptHolzschnauz, 2020-11-21:
I have the same issue with SAMD21 on Arduino MKR.  With a new hardware revision (No changes on Pullup etc) the problem came but i don't understand why.

With the lib i receive 85 or 127, exept for resolution 8,9,10 then i get the right value on pin PA22 (Digital) but still 127 on PA04 (Analog). 
If i use
pinMode(DS18B20_PIN, OUTPUT);
digitalWrite(DS18B20_PIN, HIGH);
delay(750);
it works.

If i use the examples from https://www.pjrc.com/teensy/td_libs_OneWire.html everything works perfect in any variation.

All with the same OneWire lib.

I'm thankfull for a hint of the pro's, the problem is sooo deep in the registers...

CptHolzschnauz, 2020-11-22:
No matter how weird it gets...
In my case it's the sensors hardware....
2 original DS18B20: One of them works perfectly, no issues. The other one behave like i mentioned in the post above:
The sample sketch from the OneWire lib  works, with the DallesTemp lib it works with resolution 9 and 10..

Whatsoever : If someone comes to this post in desperation (like me) : CHECK (AND REPLACE) THE SENSORS, it may be the hard, not the software

john-tomazis, 2020-12-04:
Hi guys
just to share my experience with the DS18B20 . I had same problems with the sensor and wrong readings like, -175 degrees and high temperatures jumping up to 50 ,80 ,or 100 degrees and also no readings at all .
i was using the index method to read it, but when i changed to address reading method  things got much better.
I am using 14 sensors one wire bus with about 2meters cable . I have changed the software a bit with some if statements to keep my readings within range and everything works fine

CptHolzschnauz, 2020-12-04:
Same here. With the OneWire temperature example from the sketchbook it work with every sensor without hard coding the adresses. With the DallasTemp Lib and the index method it's hardware related. Some seonsor do, some don't. 
I use four OneWire lines and i don't know how many sensors the user will connect to each line so i have to scan every bus and count and index and so on. With the DallasTemp Lib it's easy, i don't have the nerve to recode without the lib..
Conclusion:
- Find a supplier with proper sensors and everything works fine - Guys, thx for the work on the DallasTemp lib!
- It's hard to solve that issue for the DallasTemp lib because it's a flaky error and hardware related.
- If you know the numbers and the adresses of the sensors i would read out the data with the onewire lib directly without the DallasTemp lib.

closed 2022-06-03 by milesburton