milesburton/Arduino-Temperature-Control-Library #119: Delay Check for Parasitic Powering and Setting Resolution

gfvalvo, 2018-08-24:
Hi,
Regarding the ‘DallasTemperature::begin()’ method. I’m wondering if perhaps the following lines:
`  if (validAddress(deviceAddress)) {
    if (!parasite && readPowerSupply(deviceAddress))
      parasite = true;

    bitResolution = max(bitResolution, getResolution(deviceAddress));

    devices++;
    if (validFamily(deviceAddress)) {
      ds18Count++;
    }
  }`

Should be rearranged as:
`  if (validAddress(deviceAddress)) {
    devices++;
    if (validFamily(deviceAddress)) {

      if (!parasite && readPowerSupply(deviceAddress))
        parasite = true;

      bitResolution = max(bitResolution, getResolution(deviceAddress));
      ds18Count++;
    }
  }`
Thus, the check for parasitic powering and setting of the temperature resolution will be postponed until we’re sure that the currently located device is actually a member of the DS18... family. This could be important if other types of devices are also on the same OneWire bus.

RobTillaart, 2018-08-27:
Seems valid to me. Can you make a PR for it?

RobTillaart, 2020-02-11:
#150 should be merged first

commit ce610fe9 (2020-06-30): fix #119 Call readPowerSupply and getResolution only for DS18xxx devices.
Other onewire devices may not support these calls.
closed 2020-06-30 by milesburton