milesburton/Arduino-Temperature-Control-Library #121: blockTillConversionComplete overflow when millis() is less than delms

johnsthompson, 2018-09-12:
I've had an issue using this library with an DB18B20 on the ESP8266 chipset.  It exhibits itself when you call DS18B20.requestTemperatures very soon after startup.  On the ESP8266 the millis() restarts at 0 every time you reset.  My goal is to take a reading and go back to sleep as soon as possible.  I was wondering why DS18B20.setWaitForConversion(true / false) had no effect in my code...

With reference to the code block below if millis() is less than delms I believe that there is an overflow bug.  millis() is an unsigned integer and it overflows.  **millis() - delms** is always greater than **now** and execution continues regardless of what isConversionComplete() is.

// Continue to check if the IC has responded with a temperature
void DallasTemperature::blockTillConversionComplete(uint8_t bitResolution) {
	int delms = millisToWaitForConversion(bitResolution);
	if (checkForConversion && !parasite) {
		unsigned long now = millis();
		**while (!isConversionComplete() && (millis() - delms < now))**
			;
	} else {
		delay(delms);
	}
}

I changed the while statement to:

while (!isConversionComplete() && (millis() - now < delms))
   yield();

Note that I think (please correct me if I'm wrong) this statement change also correctly protects against millis() wrap-around.

I added the yield(); because on long 12-bit conversions circa. 750ms the ESP8266 watchdog soft-resets without it.

RobTillaart, 2018-09-13:
The wrap around of millis occurs after ~50 days  but still worth to check this issue.

suppose millis() wraps at 1000 ( to make the analysis easier) and that delms = 10

scenario
```
unsigned long now = millis();      // now = 995 - close near wrap around value of millis()

(millis() - delms < now)
995 - 10 = 985 < 995 => true
996 - 10 = 986 < 995 => true
997 - 10 = 987 < 995 => true
...
004 - 10 = 994 < 995 => true
005 - 10 = 995 < 995 => false	=> so fall through
006 - 10 = 996 < 995 => false	=> so fall through
...
009 - 10 = 999 < 995 => false	=> so fall through
010 - 10 = 0 < 995 => true 		=> !!!!
...
994 - 10 = 984 => true ...

(millis() - now < delms)
995 - 995 = 0 < 10 => true
996 - 995 = 1 < 10 => true
997 - 995 = 2 < 10 => true
...
004 - 995 = 9 < 10 => true
005 - 995 = 10 < 10 => false		=> so fall through
006 - 995 = 11 < 10 => false		=> so fall through
...
009 - 995 = 14 < 10 => false		=> so fall through
010 - 995 = 15 < 10 => false		=> so fall through
...
994 - 995 = 999 < 10 => false 
```
So indeed the original code fails when we have a wrap around of millis() and the time between two subsequent calls to the test condition is so large that millis() has become larger than **delms**.
As **delms** is at least ~90 milliseconds, the chance the existing code fails is pretty low.
That said your proposed code is correct and better than the existing code.

The addition of yield() in the while loop should be no problem as far as I can see.

Can you make a pull request?

adams13x13, 2019-04-14:
One should replace
```
		unsigned long now = millis();
		while (!isConversionComplete() && (millis() - delms < now))
			;
```
by
```
		unsigned long start = millis();
		while (!isConversionComplete() && (millis() - start < delms))
			;
```
Whether **millis()** overflows or not, **(millis() - start)** will be still calculated correctly as long as **delms** is not more than 49 days which is uncommon for a cable connection. ;-)

For example if start=4294967291 (which corresponds to -5 if considered as a signed integer), after 20 milliseconds millis() will give 15; 15-4294967291 = 20 (mind the overflow!) - correct value. We can always compare it with "delms".

commit 7ed29994 (2020-02-10): Calculate timeout against millis() correctly

Fixes #121.
closed 2020-02-10 by milesburton