milesburton/Arduino-Temperature-Control-Library #126: Erroneous temperature readings

porkyneal, 2019-01-24:
Apologies if this has been covered before, I have spent a long time using google to find others with this issue but no exact match.

I have deployed a number of sensors with an ESP8266, some extended some not. However, occasionally when in a noisy environment I get the odd erroneous temperature reading. See the data below, specifically the 79.5 and 79.75 readings.

`
-15.6,
-16,
-15.9,
-16.3,
79.75,
-16.4,
-16.5,
-16.6,
-16.9,
-17,
-16.9,
-16.8,
-16.9,
-16.6,
-16.8,
-16.4,
79.5,
-16.3,
-16.1,
-16,
`

What is interesting, is that every erroneous reading is around that 79 / 80 c mark.

`79.81,
80.56,
80.56,
79.94,
80.13,
80.06,
84.99,
80.25,
80.19,
80,
80.06,
80.31,
80.38,
80,
80.38,
79.88,
80.25,
80.13,
80,
80.31,
80.19,
80.13,
80`

The probe is checked every 10s and I am effectively chucking out the disconnect results (i.e. 85C).

Probe is powered by 3.3v and 4.7k resistor used, 3 wires not parasitic mode. I have around 30 devices deployed, with only 1/2 showing this issue at the moment. 

I assume, rightly or wrongly, that the CRC should be kicking any communication errors out. So is this a probe fault, comms issue or other?

Does anybody have any potential solution? I would like to fix this without hardware changes (i.e. resistor) as the device is not easily accessible.

RobTillaart, 2019-01-24:
Are those temperatures Celsius or Fahrenheit?
Can you post the core of your code where you check the CRC?

porkyneal, 2019-01-24:
Celcius. Is the CRC not checked in the library? 

`bool DallasTemperature::isConnected(const uint8_t* deviceAddress,
		uint8_t* scratchPad) {
	bool b = readScratchPad(deviceAddress, scratchPad);
	return b && (_wire->crc8(scratchPad, 8) == scratchPad[SCRATCHPAD_CRC]);
}`

I have noticed the new function to check for a scratchpad with all zeroes, which I do not have installed. But if I was suffering this issue, would I not get the same temperature output every time?

RobTillaart, 2019-01-24:
The CRC is 8 bit so theoretically if a CRC error occurs it has a 1 in 256 chance it does not detect it. So it definitely does not catch all communication errors.

I notice the difference between your temperature -16 and the readout 80 is 96, which indicates that (at least) two bits have flipped. 96 = 64 + 32. 

can you add some debug print lines in the isConnected() function of the library?

```bool DallasTemperature::isConnected(const uint8_t* deviceAddress,
		uint8_t* scratchPad) {
	bool b = readScratchPad(deviceAddress, scratchPad);
	return b && !isAllZeros(scratchPad) && (_wire->crc8(scratchPad, 8) == scratchPad[SCRATCHPAD_CRC]);
}```

==>

```bool DallasTemperature::isConnected(const uint8_t* deviceAddress,
		uint8_t* scratchPad) {
	bool b = readScratchPad(deviceAddress, scratchPad);
        Serial.print("DBG:\t");
        Serial.print(_wire->crc8(scratchPad, 8), HEX);
        Serial.print("\t");
        Serial.print(scratchPad[SCRATCHPAD_CRC], HEX);
	return b && !isAllZeros(scratchPad) && (_wire->crc8(scratchPad, 8) == scratchPad[SCRATCHPAD_CRC]);
}```

This shows the CRC and it might give an idea why it does match even with wrong temperature.

porkyneal, 2019-01-24:
The device is remote, so I am just connecting the library with my MQTT logging service. Will report back when I have the device updated. Cheers.