milesburton/Arduino-Temperature-Control-Library #127: Devices are never issued 0xB8 (recall eeprom) command

m-elias, 2019-01-25:
The short:
Nowhere in the DallasTemperature library does the RECALLSCRATCH (0xB8) command get used. The library assumes that the EEPROM contents never change after power up. I believe 0xB8 should be issued before reading the scratchpad. 

The long:
I recently discovered I have some knock-off DS18B20 sensors that, I presume, have faulty EEPROM or no EEPROM. Either way, I cannot write to the EEPROM on these particular devices. This lead me to suspect my code was somehow bad so I tried to use the DallasTemperature library's setHighAlarmTemp & setLowAlarmTemp functions instead of my own direct OneWire.write commands. At first, I thought it was working until I realized the changes were not surviving a power cycle on my faulty devices.

The reason I thought the DallasTemperature library was working is that it successfully writes to the device's scratchpad and issues the 0x48 command (copy scratchpad to EEPROM), but then when checking if it was successful, it reads the scratchpad (which has the correct values) without first copying the EEPROM to the scratchpad.

The Arduino library manager says I'm at 3.8.0 but DALLASTEMPLIBVERSION shows 3.7.9.
OneWire library 2.3.3.

m-elias, 2019-02-01:
I'm not sure what the best solution is. Copying the EEPROM to the scratchpad every time the scratch is read would create a lot of extra overhead as the scratchpad is frequently read when writing to the High/Low alarm setpoints (other functions may also frequently read the scratchpad but the setHighAlarmTemp & setLowAlarmTemp functions were the only ones that I closely examined). Maybe a "copyEepromToScratchpad" function needs to be added and the user needs to be made aware of the necessity to expressly call it?

everslick, 2019-07-28:
maybe only verify the very first write to the eeprom?

RobTillaart, 2020-02-11:
The main purpose of the EEPROM is that after a power down the scratchPad of the DS18 is loaded with the last known values. The datasheet states that the DS18 EEPROM can be written 50K times which is roughly 14 times per day for 10 years so a lot but not endless.

Current implementation of **writeScratchPad(...)** will always write new values to EEPROM. If the user does three calls setResolution(). setHighAlarmTemp() and setLowAlarmTemp() there can be **THREE** writes/copies to EEPROM.  

The good news is that these functions do not update the EEPROM if the new value equals the current one. 

The bad news is that current implementation to set a new resolution is rather inefficient. Patched the DallasTemperature.cpp file with some Serial.println(__FUNCTION__); to see the calls, one sensor connected. Comments are mine.
```
setResolution(addr, 9); 
---
// if (getResolution(deviceAddress) == newResolution)
getResolution  // to check if new == old res, so good optimization
isConnected  
readScratchPad

// if (isConnected(deviceAddress, scratchPad)) {
isConnected       // 
readScratchPad  // double
writeScratchPad   // that is what we want to achieve --> writes to EEPROM

//  if (!skipGlobalBitResolutionCalculation && (bitResolution > newResolution)) {
getResolution
isConnected
readScratchPad
getResolution    // called twice due to using the MAX macro... ==> performance bug ==> new issue.
isConnected
readScratchPad
```
So setResolution() needs attention although this might become a speed versus size issue. I will investigate this. 
Furthermore I will make an issue (and PR) for the max bug in setResolution() first. Same bug exists in begin() but as that code is only executed once the impact is less.

Back to the original issue, a void recallEEPROM(addr) function seems an appropriate addition. 
It could look like code below, note that timing is to be investigated.
```
void DallasTemperature::recallEEPROM(addr) {
	_wire->reset();
	_wire->select(deviceAddress);
	_wire->write(RECALLSCRATCH);
        delay(10);
        if (parasite) {
		activateExternalPullup();
		delay(10);
		deactivateExternalPullup();
	}
	_wire->reset();
};

gitkomodo, 2020-02-13:
Because the number of times the EEPROM can be written I think values should only be saved when explicitly requested and *not* every time new values are written to the ScratchPad. As it is now *every* resolution change leads to an EEPROM update.

The data sheet states that the 'recall operation happens automatically at power-up'. So while recall isn't called from the library, saving the configuration parameters isn't useless.

I would call the functions saveConfiguration and recallConfiguration to express *what* is saved/recalled rather than *where* it is saved to/recalled from.

Regarding the timing after a Recall EEPROM command the data sheet details the same behaviour as seen after a temperature conversion request. So the timing could be:

`
while(!isConversionComplete()) yield();
`

Because of the broader use of the function it might be renamed to isCommandComplete(). And maybe it's good to add a max waiting time to prevent infinite blocking and/or add a boolean to the recall function to indicate if blocking till completion is wanted.

RobTillaart, 2020-02-14:
> Because the number of times the EEPROM can be written is limited I think values should only be saved when explicitly requested and not every time new values are written to the ScratchPad. As it is now every resolution change leads to an EEPROM update.

That would explicitly break the current behavior of the library.

[thinking...]
How about a flag **autoSaveConfiguration** which is default true, That would keep the behavior the same unless the user explicitely sets it to false. It would not break the interface and it would allow one to save the EEPROM writes 

> I would call the functions saveConfiguration and recallConfiguration to express what is saved/recalled rather than where it is saved to/recalled from.
Agree,   [RECALL EEPROM comes from datasheet]

so the proposal would become to extend the interface with
```
void DallasTemperature::recallConfiguration(addr) 

void DallasTemperature::saveConfiguration(addr)

void DallasTemperature::setAutoSaveConfiguration(addr)

bool DallasTemperature::getAutoSaveConfiguration(addr)
``` 

And a refactor to use saveConfiguration() internally in  saveScratchPad()

gitkomodo, 2020-02-14:
Again (like in the other issue we both commented on) you're right about not breaking the current behavior. I support your solution using a flag autoSaveConfiguration that is true by default.

I'm not sure if that flag should be kept for each address like your interface proposal suggests. Wouldn't it be enough to just keep one flag that is used for every device on the wire?

Something else... do we need recall/saveConfiguration() without address to apply the action to all devices on the wire? And matching ...ByIndex functions? It feels like a lot of additions but on the other hand matches the function families for other commands.

RobTillaart, 2020-02-14:
> I'm not sure if that flag should be kept for each address like your interface proposal suggests. Wouldn't it be enough to just keep one flag that is used for every device on the wire?

Yes, one autoSaveConfiguration to rule them all, is better. Simple to implement, minimal memory impact and still allows the developer to switch it on/off for a specific sensor.

> do we need recall/saveConfiguration() without address to apply the action to all devices on the wire? 
Yes, would make it more complete.

> And matching ...ByIndex functions? It feels like a lot of additions but on the other hand matches the function families for other commands.
Makes sense, they are minimal wrappers with additional getAddress(idx, addr) call.

Make the return value bool to see if the calls succeeded?

so the proposal would become to extend the interface with
```
// returns true if succeeded, false otherwise
bool DallasTemperature::recallConfiguration(index) 

// returns true if succeeded, false otherwise
bool DallasTemperature::saveConfiguration(index)

// if addr == NULL all devices are recalled.
// returns true if succeeded, false otherwise
bool DallasTemperature::recallConfiguration(addr == NULL) 

// if addr == NULL all devices are recalled.
// returns true if succeeded, false otherwise
bool DallasTemperature::saveConfiguration(addr == NULL)

// returns value of the autoSaveConfiguration flag
bool DallasTemperature::setAutoSaveConfiguration()

// returns value of the autoSaveConfiguration flag
bool DallasTemperature::getAutoSaveConfiguration()
```

gitkomodo, 2020-02-15:
I don't think setAutoSaveConfiguration(argument bool flag!) needs a return value as I can't think of a reason why this wouldn't succeed. For the save/recall functions I can imagine a return value is useful, but I wouldn't add too much overhead to actually check if the operation succeeded. So if you mean success = no errors, then I agree.

That makes me think... there can be various reasons why an operation fails (disconnected, unexpected response, etc). A boolean doesn't give too much information about the error, but adding extra info can be too much overhead.

Somewhat related: not all functions that could have a return value indicating success/failure have one. Would changing return void to bool for those be a problem?

gitkomodo, 2020-02-16:
I implemented first versions of the functions. While scanning the datasheets of different sensor models I noticed some things that raise questions:

- All supported sensor models except DS18B20 mention a way to check if Copy Scratchpad is completed (similar to temperature conversion and recall scratchpad commands). DS18S20/DS28EA00 datasheets describe it in text and flowchart, DS1822/DS1825 don't describe it in text but show it in flowchart. All datasheets state writing to EEPROM should take max 10ms. Current implementation just waits 10ms, but previous implementation waited longer (see comments in code).

- I proposed function names like save/recall 'configuration' but this term is only used in the datasheets for one of the bytes saved to EEPROM, some models don't even have this configuration register. So maybe it's not the best name... Someone come up with a good name/word for Th & Tl registers, user bytes 1 & 2 and configuration register?

Current first work in branch https://github.com/gitkomodo/Arduino-Temperature-Control-Library/tree/gitkomodo-eeprom needs (a lot of) testing, especially for parasite powered mode I'm not experienced.  All feedback is welcome, most interested in:
1. If Copy Scratchpad (to EEPROM) and Recall Scratchpad (from EEPROM) work in your setup.
2. Parasite powered mode results.
3. Typical times for Recall Scratchpad (from EEPROM) to complete (in both normal en parasite power modes).

RobTillaart, 2020-02-16:
(2 days away and a lot happened)  👍 

> I don't think setAutoSaveConfiguration(argument bool flag!) needs a return value as I can't think of a reason why this wouldn't succeed. 

Good point as it only sets a flag.

> For the save/recall functions I can imagine a return value is useful, but I wouldn't add too much overhead to actually check if the operation succeeded. So if you mean success = no errors, then I agree.

Mean reason I can think of why these could fail if an address/sensor is not reachable.

> That makes me think... there can be various reasons why an operation fails (disconnected, unexpected response, etc). A boolean doesn't give too much information about the error, but adding extra info can be too much overhead.

Make it an int and use [0 == ok,   -1 == FAIL]  plenty of room for extension in future, not all need to defined / implemented in first version.

> Somewhat related: not all functions that could have a return value indicating success/failure have one. Would changing return void to bool for those be a problem?

No as no existing app would use it yet. However one has to think over the semantics of the error codes over the whole set of functions. 

in short: go for int return values.

RobTillaart, 2020-02-16:
>I implemented first versions of the functions. 

👍 I will have a look

>All datasheets state writing to EEPROM should take max 10ms. Current implementation just waits 10ms, but previous implementation waited longer (see comments in code).

Follow datasheet and let it be max 10ms 

> Someone come up with a good name/word for Th & Tl registers, user bytes 1 & 2 and configuration register?

saveConfigRegisters() recallConfigRegisters();  or  saveRegisters() recallRegisters(); or just  save(), and recall().  saveScratchpad(), recall scratchPad() 
MakePersistent()  ?

saveScratchpad(), would fit in the read / write scatchpad names 

> parasite mode
equal less experience, ... but you raise good questions

RobTillaart, 2020-02-16:
First look at the naming / signatures

```
  // Sends command to one device to save configuration from scratchpad to EEPROM by index
  // Returns 0 if no errors were encountered, otherwise indicates failure
  int saveScratchPadByIndex(uint8_t);
  
  // Sends command to one or more devices to save configuration from scratchpad to EEPROM
  // if nullptr or no address is given all devices are saved.
  // Returns 0 if no errors were encountered, otherwise indicates failure
  int saveScratchPad(const uint8_t* deviceAddress == nullptr);
  
  // Sends command to one device to recall configuration from EEPROM to scratchpad by index
  // Returns true if no errors were encountered, false indicates failure
  int recallScratchPadByIndex(uint8_t);
  
  // Sends command to one or more devices to recall configuration from EEPROM to scratchpad
  // if nullptr or no address is given all devices are recalled.
  // Returns 0 if no errors were encountered, otherwise indicates failure
  int recallScratchPad(const uint8_t* deviceAddress == nullptr);
  
  // Sets the autoSaveScratchPad flag
 inline void setAutoSaveScratchPad (bool b) { autoSaveScratchPad = b; };
  
  // Gets the autoSaveScratchPad flag
  inline bool getAutoSaveScratchPad(void) { return autoSaveScratchPad; };

  // used to determine if scratchPad will be saved to EEPROM on every scratchpad write
  bool autoSaveScratchPad;

(and somewhere in the constructor too...)
```
opinion?

Also needed are 2 examples
1. save recall ScratchPadByIndex manually
2. save recall scratchPad manually

RobTillaart, 2020-02-16:
A look in your code https://github.com/gitkomodo/Arduino-Temperature-Control-Library/blob/gitkomodo-eeprom/DallasTemperature.cpp
```
bool DallasTemperature::recallConfiguration(const uint8_t* deviceAddress) {
...
if (millis() - start < 200) return false; // Datasheet doesn't specify typical/max duration
```
should be > 200   

and I think 20ms is more than enough (e.g. EEPROM in 24LC series have max 5ms latency)

=====

In writeScratchPad you can use the  autoSaveConfiguration flag directly as it is a private member of the class.  (if you declare them inline in the .h file as proposed  above the compiler can do this automagically)
```
void DallasTemperature::writeScratchPad(const uint8_t* deviceAddress,
		const uint8_t* scratchPad) {
...
  if (autoSaveConfiguration)
    saveConfiguration(deviceAddress);
  else
    _wire->reset();
}
```

gitkomodo, 2020-02-16:
Yes, saveScratchPad/recallScratchPad/autoSaveScratchPad sounds best. For now I'll keep return values boolean as this matches all other functions in the library that indicate success. Maybe differentiating error codes is something for a future project (but I don't really need it).

I did some timing tests on recalling the scratchpad from EEPROM. My sensor reports the 'done bit' within a millisecond, so the timeout in this function can be way shorter than my initial 200ms. Right now I'm thinking of a 10ms timeout as recalling will probably always be faster than saving (which is specified in the data sheet).

Regarding parasite mode, is there an alternative to using a p-channel mosfet? I don't have any of those laying around...

m-elias, 2020-02-16:
> Regarding parasite mode, is there an alternative to using a p-channel mosfet? I don't have any of those laying around...

I've never used a MOSFET in parasite power mode on an Arduino. I just connect data and ground, and I believe technically the VCC pin is supposed to be tied to ground in that mode although it works without that.

gitkomodo, 2020-02-16:
> if (millis() - start < 200) return false; // Datasheet doesn't specify typical/max duration
> 
> should be > 200
> 
> and I think 20ms is more than enough (e.g. EEPROM in 24LC series have max 5ms latency)

Right! Correcting doesn't change timings because the first bit read already indicates 'done'. Changing condition, leaving 10ms for now as mentioned above.

> In writeScratchPad you can use the autoSaveConfiguration flag directly as it is a private member of the class. (if you declare them inline in the .h file as proposed above the compiler can do this automagically)

Using the flag directly now. Declaring inline gives me compiler errors.

>I've never used a MOSFET in parasite power mode on an Arduino. I just connect data and ground, and I believe technically the VCC pin is supposed to be tied to ground in that mode although it works without that.

Really, data to ground? I didn't expect that. What to do with the strong pullup pin configured in the library? Can you list all connections to make for parasite mode? That would help a lot.

m-elias, 2020-02-16:
Data to Arduino pin
Gnd to gnd
VCC to gnd
The strong pull-up pulls up on the data line to keep the internal parasite capacitor charged while the temperature conversion is happening. In my applications, the Arduino just sets the pin to output high during the pull-up period.

m-elias, 2020-02-16:
Forgot to mention you still need the pull-up resistor on the data pin.

gitkomodo, 2020-02-16:
Thank you @m-elias now I understand! ;-) I guess I overlooked parasite mode without _external_ pullup because it is handled in the OneWire library, instead of DallasTemperature.

Anyway parasite mode won't work with my cheap sensors. Doing some digging I found this detailed article: https://github.com/cpetrich/counterfeit_DS18B20. The sensors I have now definitely belong to family D2.

The description is very accurate: ROM patterns matches, initial temperature reading is 25 °C, default alarm register settings were 0x55 and 0x05. And as mentioned in issue #110  takes almost 500ms for temp conversion regardless of measurement resolution. I didn't test the other properties, but the rest is so accurate I'm sure the statement Parasitic Power won't work is correct too.

So, thank you again! Can someone else do some testing in parasitic mode?

gitkomodo, 2020-02-21:
I picked up some genuine sensors (as far as I know now) and tested the new functionality in non-parasite and parasite modes. Tests in both modes succeeded.

This only leaves saving in parasite mode _with strong pullup_ untested (specification doesn't require strong pullup when recalling). But I expect no problems because of similarity to strong pullup during temperature conversion _and_ same behavior was previously part of _writeScratchPad_.

Reviewing the code I added a _return false_ to both _ByIndex_ functions when _getAddress_ returns false (device not found). I also added the new functions to keywords.txt (updated other functions too) and simplified the example as the first version was way too complex.

I think this is about all I can do on this functionality. I'll push these changes to https://github.com/gitkomodo/Arduino-Temperature-Control-Library/tree/gitkomodo-eeprom soon and make a PR.

RobTillaart, 2020-02-21:
had a look and code seems to be ok, no questions so make the PR.

PR #152 (2020-06-30): Save to / recall from EEPROM functionality
Resolves #127. This PR (also) includes commits present in PR #150.
closed 2020-06-30 by milesburton