milesburton/Arduino-Temperature-Control-Library #129: no data when atmega328P on 8mhz clock

fuco809, 2019-03-25:
i use the lib successfully with my arduino uno board, and it works very nice with atmega328p at 16mhz clocks. but for battery saving reasons and lower voltage (3,6 to 4,2) i switched in my project to 8Mhz clocks, now i get mostly no tempature data, just a few measurements are working.

i think its a timing problem, i tried to read data directly with that guide:
https://www.arduinoforum.de/arduino-Thread-ATTINY85-mit-DS18B20-Temperatursensor-und-echtem-Analogausgang
this works, but only supports one device.

what have to be changed in your lib to get it working with slower clocks too?

best regards
hannes

RobTillaart, 2019-03-25:
The critical timing is in the oneWire library.
check the functions 
- reset()
- write_bit()
- read_bit()

some of the calls to delayMicroseconds are more than critical in 8 Mhz, 
especially the short ones need some tuning I expect

by using   #if F_CPU == 8000000    you could make it working for both  8 and 16 MHz

mmithril, 2019-04-04:
Try lower voltage and 16 MHz clock. If problem persists, it is a low voltage × weak pull-up × high capacitance problem.

In my greenhouse setup with multiple sensors and long cables (high capacitance), the weak pull-up resistor (4k7) was unable to pull-up the line fast enough for 1-wire bus stay in sync. It happened especially when temperature was low and humidity high in the greenhouse (cables got wet, rising capacitance).

My ultimate solution was [ds18b20-controller](https://github.com/mmithril/ds18b20-controller) for the greenhouse application and #131 for another setup (3.3V ESP8266).

closed 2019-06-28 by milesburton