milesburton/Arduino-Temperature-Control-Library #137: UserData

vangalvin, 2019-09-20:
I have a string of 40 sensors, After sorting them and slodering them I realised I had messed up so they are out of order. Ideally i would like to be able to refer to the top sensor as sensor 0 and the last one as sensor 40. currently sensor 0 is about the 12 sensor in the chain and 40 is about the 3rd sensor on the chain if I obtain temperatures by address (or index).

I set the UsedData to have the correct numerical sequence however the example seems to be ignoring this and when the temperature from getTempByID(0) its still returning the temperature form the incorrect sensor and not the one with 0 set in UserData. 

Any ideas how to sort this out, or another option to sort the sensor readings in to a logical order?

RobTillaart, 2019-09-20:
check https://github.com/milesburton/Arduino-Temperature-Control-Library/tree/master/examples/UserDataDemo
change the size of the (struct) array from 4 to 40 

Can you post the output?

vangalvin, 2019-09-20:
Hi @RobTillaart I altered the outout so you can better see the results of the userID, ID and the address.

#devices: 39
UserID: 30 address: 28FF6419F06D3A73
UserID: 38 address: 28FF6419F230F128
UserID: 37 address: 28FF6419F230053D
UserID: 0 address: 28FF6419F2285881
UserID: 31 address: 28FF6419F218A799
UserID: 20 address: 28FF6419F2386470
UserID: 14 address: 28FF6419F224A8B8
UserID: 18 address: 28FF6419F224A398
UserID: 15 address: 28FF6419F214B4AB
UserID: 29 address: 28FF6419F22C89B3
UserID: 8 address: 28FF6419F22CB90D
UserID: 4 address: 28FF6419F23CE1F8
UserID: 12 address: 28FF6419F21A5C5C
UserID: 26 address: 28FF6419F206221A
UserID: 25 address: 28FF6419F20ED727
UserID: 27 address: 28FF6419F22EB000
UserID: 24 address: 28FF6419F2459E45
UserID: 7 address: 28FF6419F20D73BC
UserID: 6 address: 28FF6419F23D39A9
UserID: 3 address: 28FF6419F203824A
UserID: 9 address: 28FF6419F243F5AA
UserID: 10 address: 28FF6419F2433DA2
UserID: 16 address: 28FF6419F25347C8
UserID: 2 address: 28FF6419F21BF856
UserID: 1 address: 28FF6419F21B7479
UserID: 5 address: 28FF6419F21B3561
UserID: 28 address: 28FF6419F21B9BD1
UserID: 11 address: 28FF6419F23B6F05
UserID: 19 address: 28FF6419F20F42CD
UserID: 22 address: 28FF6419EE6A269D
UserID: 21 address: 28FF6419F1971319
UserID: 17 address: 28FF6419F3903245
UserID: 23 address: 28FF6419F3942682
UserID: 32 address: 28FF6419F3947918
UserID: 36 address: 28FF6419F3915439
UserID: 34 address: 28FF6419F391A6F1
UserID: 35 address: 28FF6419F39136E0
UserID: 13 address: 28FF6419EFD6C04D
UserID: 33 address: 28FF6419EF8EA8EB

13672	Get temp by address
index#: 0 UserID: 30 address: 28FF6419F06D3A73temp: 16.56
index#: 1 UserID: 38 address: 28FF6419F230F128temp: 16.69
index#: 2 UserID: 37 address: 28FF6419F230053Dtemp: 16.94
index#: 3 UserID: 0 address: 28FF6419F2285881temp: 16.69
index#: 4 UserID: 31 address: 28FF6419F218A799temp: 16.87
index#: 5 UserID: 20 address: 28FF6419F2386470temp: 16.94
index#: 6 UserID: 14 address: 28FF6419F224A8B8temp: 17.06
index#: 7 UserID: 18 address: 28FF6419F224A398temp: 16.62
index#: 8 UserID: 15 address: 28FF6419F214B4ABtemp: 16.81
index#: 9 UserID: 29 address: 28FF6419F22C89B3temp: 16.75
index#: 10 UserID: 8 address: 28FF6419F22CB90Dtemp: 16.87
index#: 11 UserID: 4 address: 28FF6419F23CE1F8temp: 16.94
index#: 12 UserID: 12 address: 28FF6419F21A5C5Ctemp: 17.06
index#: 13 UserID: 26 address: 28FF6419F206221Atemp: 16.94
index#: 14 UserID: 25 address: 28FF6419F20ED727temp: 16.75
index#: 15 UserID: 27 address: 28FF6419F22EB000temp: 16.94
index#: 16 UserID: 24 address: 28FF6419F2459E45temp: 16.75
index#: 17 UserID: 7 address: 28FF6419F20D73BCtemp: 16.87
index#: 18 UserID: 6 address: 28FF6419F23D39A9temp: 16.81
index#: 19 UserID: 3 address: 28FF6419F203824Atemp: 17.06
index#: 20 UserID: 9 address: 28FF6419F243F5AAtemp: 16.56
index#: 21 UserID: 10 address: 28FF6419F2433DA2temp: 16.87
index#: 22 UserID: 16 address: 28FF6419F25347C8temp: 17.06
index#: 23 UserID: 2 address: 28FF6419F21BF856temp: 16.75
index#: 24 UserID: 1 address: 28FF6419F21B7479temp: 16.94
index#: 25 UserID: 5 address: 28FF6419F21B3561temp: 17.19
[... 28 lines cut ...]
loop index#: 14 UserID: 25 address: 28FF6419F20ED727	 temp:	17.06
loop index#: 15 UserID: 27 address: 28FF6419F22EB000	 temp:	16.81
loop index#: 16 UserID: 24 address: 28FF6419F2459E45	 temp:	17.06
loop index#: 17 UserID: 7 address: 28FF6419F20D73BC	 temp:	16.69
loop index#: 18 UserID: 6 address: 28FF6419F23D39A9	 temp:	16.62
loop index#: 19 UserID: 3 address: 28FF6419F203824A	 temp:	16.75
loop index#: 20 UserID: 9 address: 28FF6419F243F5AA	 temp:	16.94
loop index#: 21 UserID: 10 address: 28FF6419F2433DA2	 temp:	16.75
loop index#: 22 UserID: 16 address: 28FF6419F25347C8	 temp:	16.81
loop index#: 23 UserID: 2 address: 28FF6419F21BF856	 temp:	16.87
loop index#: 24 UserID: 1 address: 28FF6419F21B7479	 temp:	16.75
loop index#: 25 UserID: 5 address: 28FF6419F21B3561	 temp:	16.75
loop index#: 26 UserID: 28 address: 28FF6419F21B9BD1	 temp:	16.94
loop index#: 27 UserID: 11 address: 28FF6419F23B6F05	 temp:	16.94
loop index#: 28 UserID: 19 address: 28FF6419F20F42CD	 temp:	16.81
loop index#: 29 UserID: 22 address: 28FF6419EE6A269D	 temp:	16.75
loop index#: 30 UserID: 21 address: 28FF6419F1971319	 temp:	16.56
loop index#: 31 UserID: 17 address: 28FF6419F3903245	 temp:	16.87
loop index#: 32 UserID: 23 address: 28FF6419F3942682	 temp:	17.44
loop index#: 33 UserID: 32 address: 28FF6419F3947918	 temp:	16.94
loop index#: 34 UserID: 36 address: 28FF6419F3915439	 temp:	16.69
loop index#: 35 UserID: 34 address: 28FF6419F391A6F1	 temp:	16.75
loop index#: 36 UserID: 35 address: 28FF6419F39136E0	 temp:	16.75
loop index#: 37 UserID: 13 address: 28FF6419EFD6C04D	 temp:	16.94
loop index#: 38 UserID: 33 address: 28FF6419EF8EA8EB	 temp:	16.69

vangalvin, 2019-09-21:
I found the issue, in setup the T values were not being stored based on the ID stored in the UserData area. instead it was using the index to store the data in the array.

My code is a little messy and needs some refining but its working now and I am able to get the sensors stored in order. Probibly the only thing I may need to do is set the T array so its size if not fixed incase I end up with a system that has more or less sensors.

`
DeviceAddress tempDeviceAddress;
struct
{
  int id;
  DeviceAddress addr;
} T[39];

void setup(void)
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  Serial.println("Dallas Temperature Demo");

  sensors.begin();
  
  // count devices
  deviceCount = sensors.getDeviceCount();
  Serial.print("#devices: ");
  Serial.println(deviceCount);

  // Read ID's per sensor
  // and put them in T array
  for (uint8_t index = 0; index < deviceCount; index++)
  {
    // go through sensors
    Serial.print("Address: ");
    sensors.getAddress(tempDeviceAddress, index);
//    Serial.print(sensors.getAddress(tempDeviceAddress, index));
    printAddress(tempDeviceAddress);
    Serial.print(" UserID: ");
    Serial.println(sensors.getUserData(tempDeviceAddress));
    //sensors.getAddress(T[index].addr, index);
    //T[index].id = sensors.getUserData(T[index].addr);
    //Populate the T Structure with the data using the user data
    int uID = sensors.getUserData(tempDeviceAddress);
    T[uID].id = sensors.getUserData(tempDeviceAddress);
//    T[uID].addr = tempDeviceAddress[index];
    sensors.getAddress(T[uID].addr, index);
  }
  Serial.println("T Values Populated");
  Serial.println("Check T Values");

  // Check all 4 sensors are set
  for (uint8_t index = 0; index < deviceCount; index++)
  {
    Serial.print("UserID: ");
    Serial.print(T[index].id);
    Serial.print(" Address: ");
    printAddress(T[index].addr);
    Serial.println();
  }
//  Serial.println();

}`

RobTillaart, 2019-09-21:
Please close issue

milesburton, 2019-09-21:
Closed per Rob's request

closed 2019-09-21 by milesburton