milesburton/Arduino-Temperature-Control-Library #138: Is there a way to output the temperature only as a + int?

vangalvin, 2019-09-27:
I have 40 sensors and need to encode/compress the temperatures then fire them over a satellite link. The temperature readings are going to be well in to the -'s at this site and some will be well in to the + deg C.

Having a look thought the code I thought that getTemp() returned a + int value but it appears that it does not after I tested the sensor at -20degC and received a negative output.

Other than that I could not find anything that either presented a raw + value unless I missed it (due to lack of sleep). 

The SAT modem requires a binary string with the sensor readings so what I was thinking was to take the reading for each sensor then convert that in to binary and do a Densely packed decimal routine to trim them down. 

Thanks in advance for any advice.

RobTillaart, 2019-09-27:
Just add 60.0 to the value and the range of the sensor moves from -55..125  to  +5..185
a float takes 4 bytes in native binary format.  As you have 1800 possible values with the given range (using 1 decimal) you could pack the value in 12 bits, 

```
float temp = readTempbyID(s);
uint16_t value = round( temp + 55) *10; // 0..1800
send_over_modem(value);
```
As you use only 12 bits it is even possible to pack 2 values in 3 bytes;

```
float temp1 = readTempbyID(s);
float temp2 = readTempbyID(s+1);
uint16_t x1 = round( temp1 + 55) *10; // 0..1800
uint16_t x2 = round( temp2 + 55) *10; // 0..1800
uint8_t bytes[3];
bytes[0] = x1 & 0x00FF;
bytes[1] = (x1 & 0xFF00)>> 8;
bytes[1] = bytes[1] | (x2 & 0xFF00)>> 4;
bytes[2] = (x2 & 0xFF00)>> 4;

send_over_modem(bytes 1..3); // for loop whatever
```
It would be easiest if your range only would have 256 distinct values, 
than they could be directly packed into a byte.

vangalvin, 2019-09-27:
Thank you @RobTillaart I will give that a try shortly. That looks perfect and thanks for adding that packing part. I have been trying to work out what the best method to handle this was going to be. Even contemplated just sending deltas however from what I understand doing that on a sat network can be problematic as the packets can be received out of sequence.

vangalvin, 2019-09-28:
@RobTillaart when converting the bytes back to temperatures is this the correct method?
sorry its in C as have been using https://wandbox.org/permlink/79C85El7mp9r9ruj

int tx1 = ((short)(bytes[0])) + ((short)(bytes[1])<<8);
int tx2 = ((short)(bytes[1])<<4) + ((short)(bytes[2])<<4);

I seem to be getting the incorrect numbers back and not what I would have expected.

RobTillaart, 2019-09-28:
There might be a typo in my code as I did not test it. Lets check.

RobTillaart, 2019-09-28:
Change this on the sending side
```
bytes[0] = x1 & 0x00FF;       // LSB of x1  (least significant bits)
bytes[1] = (x1 & 0x0F00)>> 8;    // MSB of x1  (most significant bits)
bytes[1] = bytes[1] | (x2 & 0x0F00)>> 4;    // MSB of x2
bytes[2] = x2 & 0x00FF;      // LSB of x1
```

assuming you get the bytes in order, the receiving side should look lik

```
uint8_t bytes[3];
receive by modem(bytes 0 1 2 ); 

int tx1 = (((int)bytes[1] & 0x0F) << 8)   |   bytes[0];  
int tx2 = (((int)bytes[1] & 0xF0) << 4)   |   bytes[2];

float f1 = (tx1 * 0.1 ) - 55;
float f2 = (tx2 * 0.1 ) - 55;
```

Again not tested ...

vangalvin, 2019-09-29:
@RobTillaart that worked a treat thanks! 
Something just occurred to me as well. all the sensors are in a chain and will be lowered down a hole. The variation of each sensor is not likely to be huge and ideally I need the best resolution I can get to log the temperature changes. 
If i take the temperature form say Sensor1 then perform a delta on all the following sensors the data saving should be HUGE as if sensor one is say 12.458 degC and the next sensor is 13.127 degrees I only need to transmit a difference of 0.669 :) 

I really appreciate the assistance you have given me in this project.

RobTillaart, 2019-09-29:
Compression is a two edged sword so allways  be carefull. 
Some thoughts about your project (as far as  I understand)

1) You could add a getRaw() to the library which returns a 12 bit value. Would save some conversion which could be done at the receiving end.

2) Delta coding works till the reference sensor fails. That is relative easy to handle, just use every time a different one, so all sensors will eventually send a FULL value. In the list the ref sensor will have a delta of 0.0.  But you need to think about how to indicate failed sensors (come to that later)

3) Maybe better, use the lowest temperature as reference. Then all offsets are positive. Saves several (average 20?) sign bits, so worth might be worth the effort.
If all deltas are between 0.0 and 1.5 degrees you have 16 distinct values ==> fit in 4 bits.
40 values could be packed in 10 bytes (+ 2 bytes for reference value)

3) Delta coding could be relative to sensor itself. Send every e.g. 10 measurements a full value of every sensor and 9 times a delta, If temp changes slowly you should have very small values. However if a delta becomes too large it is time to resend all sensors at FULL

4) What precission do you need, if 10 bits is sufficient, why use 12?

5) Idea
Only send the average of all sensors if largest delta <= 0.25 from the average (or from median). So only when differences come up the details are send. This will only work if the temperature are all similar. That would safe a factor 20. You need an indicator to indicate this average.

6) Send changes only
Only send the ID of the sensor and its new value. or a NOCHANCE indicator. drawback is that this introduces a variable length encoding but if conditions are stable this safes a lot communication.

7) Subsets
Define different subsets and send the subset that changes the most more often. A subset indicator is needed. Or even odd alternating, whatever suits your project

To combine all the above your packet would look like 
1 byte - compression indicator = { TEXT, FLOAT, PACK, PACKED, DELTA, CHANGED, NOCHANGE, AVERAGE, SUB, ALARM, FAIL ... }
depending on compression indicator, it is followed by 
TEXT         40 comma separated ascii values
FLOAT      40 binary represented floats (4 bytes/float)
PACK        40 2 byte values (convert to ints)
PACK23 - 40 packed byte values (2 values in 3 bytes
DELTA    - reference value, and 40  4 bit values.
CHANGED - 1 byte for count of changes, and count (id, value) pairs.
NOCHANGE - needs no additional info
AVERAGE - 1 value for all sensor
SUB - subset-id + n values
ALARM - count, + count (id, value) pairs
FAIL - 5 bytes that give a bit array of failed sensors.
...

What do these compression gain: (1st order estimate)

TEXT is the worst as it needs roughly 250-300 bytes for all.  (definitely not)
FLOAT would use 40x4 + 1 = 161 bytes  (definitely not)
PACK would use 81 bytes       (acceptable)
**PACK23** would use 61 bytes   (better)
**DELTA** would use 23 bytes  and e.g. 1 in 10 it would need a PACK to send all in full glory
CHANGED - depends, only usable when there is very little change
NOCHANGE - would use only 1 byte and e.g. 1 in 10 it would need a PACK 
AVERAGE - 2 bytes, usability depends very much on project
SUB - depends, worth the effort if you definitely have high prio sensors and low prio sensors
ALARM - depends, 
**FAIL** - would definitely implement this one. 

BY allowing different compression techniques one can runtime decide which uses minimal bytes to send. I would implement the bold ones above, which average to

        (9x23 bytes + 61bytes)/10 = 26.8 bytes

If you could only implement one, choose PACK23 which is ~2.5x better than sending the 40 floats in their native format. 

so far, as this goes beyond the temp sensor you can mail me at rob dot tillaart at gmail

RobTillaart, 2019-11-27:
This issue can be closed

RobTillaart, 2020-02-10:
Please close

vangalvin, 2020-02-10:
Thanks for the help on this Rob.

closed 2020-02-10 by vangalvin