milesburton/Arduino-Temperature-Control-Library #141: getDeviceCount does not work with ESP32 boards

jackthese, 2019-11-29:
When I use the .getDeviceCount method with a ESP8266, it works correctly, but the same code & setup running on multiple various ESP32 boards always returns zero. At the same time, if I use a loop with the .getAddress method to count the sensors, it works correctly on both types of boards. For demo, I have a breakout board with three Dallas sensors on a common one wire (not parasitic). Code and sample output follows...

`#include <OneWire.h>
#include <DallasTemperature.h>
#define NUM_DAL 5
#define ONE_WIRE_BUS 26                  // ESP32 OneWire pin
//#define ONE_WIRE_BUS 2                   // ESP8266 OneWire pin
int NumDallasActive;
OneWire myOneWire(ONE_WIRE_BUS);         // Setup onewire & Dallas
DallasTemperature mySensors(&myOneWire);

void setup() {
  Serial.begin(9600);
  mySensors.begin();                            // initialize Dallas temp sensor use

  Serial.println("getDeviceCount method...");         // Use getDeviceCount method
  uint8_t mycount = mySensors.getDeviceCount();
  Serial.print("My Device Count:");
  Serial.println(mycount, DEC);

  Serial.println("Get Dallas addresses by index..."); // Use getAddress method to ID sensors
  NumDallasActive = 0;
  Serial.println(" Connected temp sensor addresses:");
  for (uint8_t j = 0; j < NUM_DAL; j++) {             // loop to count found sensors
    DeviceAddress tempaddr;
    if (mySensors.getAddress(tempaddr, j)) {          // true if sensor found for index j
      NumDallasActive++;
      Serial.print("Sensor "); Serial.print(j);
      Serial.print(" address : ");
      printAddress(tempaddr); Serial.println();
    }
  }
  Serial.print("My Device Count:");
  Serial.println(NumDallasActive, DEC);
}

void loop() {
}

void printAddress(DeviceAddress devAddr) {  // Print sensor address in format to copy
  Serial.print("0x");
  for (uint8_t i = 0; i < 8; i++) {
    if (devAddr[i] < 16) Serial.print("0"); Serial.print(devAddr[i], HEX);
    if (i < 7) Serial.print(", 0x");
  }
}`

Output with ESP32 is
"getDeviceCount method...
My Device Count:0
Get Dallas addresses by index...
 Connected temp sensor addresses:
Sensor 0 address : 0x28, 0xAA, 0x88, 0xBE, 0x1C, 0x13, 0x02, 0x08
Sensor 1 address : 0x28, 0xAA, 0xA8, 0xC0, 0x1C, 0x13, 0x02, 0x4D
Sensor 2 address : 0x28, 0x63, 0xB9, 0x19, 0x31, 0x14, 0x01, 0x6F
My Device Count:3"

atanisoft, 2019-12-03:
@jackthese which OneWire version are you using? There are a couple bugs in the library on the esp32 which have been fixed here: https://github.com/stickbreaker/OneWire

jackthese, 2019-12-04:
My previous testing was with oneWire 2.3.5 installed, according to the library manager. So here's what I did:
(1) I went to the link you provided and hit the clone button, then extracted the download to the libraries folder inside my Arduno folder in Documents. That created a folder called OneWire-master.
(2) There was also a folder in libraries called just OneWire, and I removed it, and restarted the Arduino IDE and opened the test sketch. The library manager listed OneWire 2.3.3 as installed, and I did not update it to 2.3.5, which was listed as available.
(3) I ran my test sketch with an ESP32, and it worked correctly! 
(4) I then used the library manager to install 2.3.5, which it did, and then recompiled and ran, and the error happened, just like before. Now, here's where it gets weird...
(5) I tried to use the library manager to reinstall 2.3.3, but it did not change the version, nor did it give me an error message. Closing and restarting the IDE changed nothing, and the program still had the error.
(6) I then tried to install v 2.3.0, but it installed 2.3.3 instead, and the program worked again!

Please note that I'm new at all this, and the library manager is somewhat of a mystery for me.... Thanks for getting this to work--now I just have to remember not to upgrade to 2.3.5...Jack

atanisoft, 2019-12-04:
@jackthese glad that the forked version fixed it for you. I'd suggest stick with the forked version, the IDE will not auto-upgrade the library so you should be safe from that aspect.

The problem on the esp32 comes down to bit timing. Since the esp32 has multiple tasks running concurrently the timing can (and usually does) get skewed leading to sensors not being seen or reporting erratically. The forked OneWire library addresses most of these issues with task concurrency mutex blocks, these steps are not necessary on AVR or esp8266 platforms that are single task/thread platforms.

jackthese, 2019-12-04:
Mike…Thanks for your help on this. The ESP32s I’ve tried have been very inconsistent with what works and what doesn’t. For instance, the IRremote.h library is basically unusable with SPIFFS. I’m still searching for a faster  controller that isn’t buggy, but perhaps that comes with multithreaded implementations. I’m just a rookie user, so I rely on guys like you to help me out, and I appreciate it. …Jack

atanisoft, 2019-12-04:
> For instance, the IRremote.h library is basically unusable with SPIFFS.

That is a rather ugly bug in how the IRRemote library is implemented with it's ISR usage. The ESP32 has a few restrictions on how things are managed when it comes to ISRs and flash. There is unfortunately no good replacement for that library that I'm aware of.

As for performance aspects, the ESP32 is more than capable of almost anything you can throw at it.

jackthese, 2019-12-05:
I was disappointed to learn that there was no suspend or disable command in the IR library. Once you enable it, it’s on forever. Perhaps simplistically, I thought that might be an avenue to work around the memory conflict. I believe I had success with SPIFFS activity before enabling IR, but that’s a very limited usage, typically. Sounds like this will never be fixed. I presume there are bug reports in the Espressif developers forum, and they have chosen to ignore it….

atanisoft, 2019-12-05:
> I presume there are bug reports in the Espressif developers forum, and they have chosen to ignore it…

It is intended behavior to disable flash access while an ISR is active. It is a bug in the IRRemote library implementation, there are other options out there. Also we shouldn't hijack this issue for IRRemote further, feel free to hop on https://gitter.im/espressif/arduino-esp32 and we can discuss further on the IRRemote issues.

jackthese, 2019-12-08:
So, I found an easy solution to getting IRremote library to work with SPIFFS. It's an easy modification (about 5 lines total) to IRremote.h and, in my case, irRecv.cpp. Basically it implements a disable command for the library, turning off the IR timers while you do SPIFFS stuff, then turns them back on. Seems to work fine...

closed 2019-12-08 by jackthese

sparkplug23, 2021-05-04:
> My previous testing was with oneWire 2.3.5 installed, according to the library manager. So here's what I did:
> (1) I went to the link you provided and hit the clone button, then extracted the [...]

Sorry to revive this old post. I was wondering since its been some time if 1) This still worked long term or 2) you found a better solution to the problem? I only tried switching to esp32 with my db18s20 sensors and just ran into this issue. Many thanks.

jackthese, 2021-05-05:
Michael… I just hooked up two Dallas sensors to a ESP32 Wrover module and ran the “Multiple” example. .getDeviceCount() did not find any sensors, but .getAddress(Name,index) did find the sensors, and they read correctly. Using OneWire 2.3.5 and Dallas 3.9.0. So the work around of searching by index is the way to go…Jack

omski, 2021-10-06:
I ran into this problem also and solved it. Maybe this could help someone: https://github.com/omski/DS18B20_TTGO