milesburton/Arduino-Temperature-Control-Library #145: [Read Power Supply]: Skip Rom instead of Search should be used

hasenradball, 2020-02-07:
In the method read power supply the device address is used.

But in the datasheet of the DS18S20 Sensor ist described that the ReadPowerSupply searches for any Device on the Bus which uses parasite power.

```
bool DallasTemperature::readPowerSupply(const uint8_t* deviceAddress) {

	bool ret = false;
	_wire->reset();
	_wire->select(deviceAddress);
	_wire->write(READPOWERSUPPLY);
	if (_wire->read_bit() == 0)
		ret = true;
	_wire->reset();
	return ret;

}
```

So the correct behavior would be to skip() instead of select(deviceAdress)

```
bool DallasTemperature::readPowerSupply(const uint8_t* deviceAddress) {

	bool ret = false;
	_wire->reset();
	_wire->skip();
	_wire->write(READPOWERSUPPLY);
	if (_wire->read_bit() == 0)
		ret = true;
	_wire->reset();
	return ret;

}
```

RobTillaart, 2020-02-07:
The flow diagram in figure 13 and 14 of the datasheet allows both paths, one with a specific address and one which uses skip ROM (0xCC)

So the method does not need to be changed but needs a second implementation.
The new readPowerSupply() does not need an address, so code should be
```
bool DallasTemperature::readPowerSupply()
{
	bool ret = false;
	_wire->reset();
	_wire->skip();
	_wire->write(READPOWERSUPPLY);
	if (_wire->read_bit() == 0)
		ret = true;
	_wire->reset();
	return ret;
}
```

As 90% of the implementation is identical, both could be implemented as a wrapper around a private worker
```
bool DallasTemperature::_readPowerSupply(const bool skip, const uint8_t* deviceAddress) 
```

Another (my preferred) implementation does the SKIP ROM when no address is given.
Although it uses a null pointer as a flag I expect it minimizes extra code, while being 
backwards compatible. Not tested.
```
bool DallasTemperature::readPowerSupply(const uint8_t* deviceAddress == ((uint8_t*) 0) )
{
	bool ret = false;
	_wire->reset();
	if (deviceAddress == ((uint8_t*) 0))
		_wire->skip();
	else
		_wire->select(deviceAddress);
	_wire->write(READPOWERSUPPLY);
	if (_wire->read_bit() == 0)
		ret = true;
	_wire->reset();
	return ret;
}
```

hasenradball, 2020-02-07:
Thanks for your fast reply and Proposal.
Your right both path can be combined.

No question regarding the return value. If true it means that any or the specific Sensor uses parasite power. Ist this just for information, or is this information further used in the code?

RobTillaart, 2020-02-07:
I do not recall that the library does use the info.
However if you use parasitic mode, you should supply a strong pull up for some time to give enough power to the chip to do the temperature conversion. There is a schema and timing in the datasheet.

hasenradball, 2020-02-07:
One remark did you mean like this?
Maybee small mistakes in the parameter "==" instead of "=".

```
bool DallasTemperature::readPowerSupply(const uint8_t* deviceAddress = nullptr)
{
	bool ret = false;
	_wire->reset();
	if (deviceAddress == nullptr)
		_wire->skip();
	else
		_wire->select(deviceAddress);
	_wire->write(READPOWERSUPPLY);
	if (_wire->read_bit() == 0)
		ret = true;
	_wire->reset();
	return ret;
}
```

RobTillaart, 2020-02-07:
Definitely a mistake, good catch :)   and yes nullptr is better too.
Have you verified that this code works?

hasenradball, 2020-02-07:
Hi,
no not verified.
I will do it and give you feedback.

hasenradball, 2020-02-10:
Hi, I tested the following code on the Digispark.
It needed liitle adaptions like nullptr -> NULL and method call.

```
bool readPowerStatus(const uint8_t* deviceAddress = NULL)
{
  bool ret = false;
  ds.reset();
  if (deviceAddress == NULL)
    ds.skip();
  else
    ds.select(deviceAddress);
  ds.write(READPOWERSUPPLY);
  if (ds.read_bit() == 0)
    ret = true;
  ds.reset();
  return ret;
}
```
But on the ESP8266 and Arduino it works with `nullptr` and so I think the following code would be fine:

```
bool DallasTemperature::readPowerSupply(const uint8_t* deviceAddress = nullptr)
{
	bool ret = false;
	_wire->reset();
	if (deviceAddress == nullptr)
		_wire->skip();
	else
		_wire->select(deviceAddress);
	_wire->write(READPOWERSUPPLY);
	if (_wire->read_bit() == 0)
		ret = true;
	_wire->reset();
	return ret;
}
```

RobTillaart, 2020-02-10:
Maybe add 

#ifndef nullptr
#define nullptr. NULL
#endif

hasenradball, 2020-02-10:
Ok Yes fine qorks fine :-)

RobTillaart, 2020-02-10:
Can be closed now

closed 2020-02-10 by milesburton