milesburton/Arduino-Temperature-Control-Library #156: invalid conversion in AlarmHandler.pde exemple

stef-ladefense, 2020-03-07:
the exemple AlarmHandler.pde is very old and not compile

C:\Users\stephane\Documents\Arduino\sketch_mar07a\sketch_mar07a.ino: In function 'void setup()':

sketch_mar07a:113:43: error: invalid conversion from 'void (*)(uint8_t*) {aka void (*)(unsigned char*)}' to 'void (*)(const uint8_t*) {aka void (*)(const unsigned char*)}' [-fpermissive]

   sensors.setAlarmHandler(&newAlarmHandler);

                                           ^

In file included from C:\Users\stephane\Documents\Arduino\sketch_mar07a\sketch_mar07a.ino:2:0:

C:\Users\stephane\Documents\Arduino\libraries\Temperature-Control-Library/DallasTemperature.h:181:7: error:   initializing argument 1 of 'void DallasTemperature::setAlarmHandler(void (*)(const uint8_t*))' [-fpermissive]

  void setAlarmHandler(const AlarmHandler *);

       ^

exit status 1
invalid conversion from 'void (*)(uint8_t*) {aka void (*)(unsigned char*)}' to 'void (*)(const uint8_t*) {aka void (*)(const unsigned char*)}' [-fpermissive]

RobTillaart, 2020-03-07:
Which platform (UNO ESP, ...) are you using?
Which version of the library are you using?

stef-ladefense, 2020-03-07:
arduino 1.8.12
ESP8266 2.6.3
3.8.0 for this library

RobTillaart, 2020-03-16:
Have no ESP 8266 installed, which tools path do you use?

RobTillaart, 2020-03-16:
ESP32 ==> same problem

RobTillaart, 2020-03-16:
I changed the code so it compiles for ESP32, can you try it for ESP8266?

Mainly it was adding const to address parameter of some functions.

[AlarmHandler.ino.zip](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/4339501/AlarmHandler.ino.zip)

RobTillaart, 2020-03-16:
Tried to compile all the example sketches for ESP32 and they all succeeded. 
So alarmHandler.pde was the only with this problem.

stef-ladefense, 2020-03-16:
test your ino, 
compile with ESP8266 2.4.2 and 2.6.3

it's ok for me, and just little 'const' before DeviceAddress

RobTillaart, 2020-03-16:
@milesburton  please close issue

closed 2020-03-16 by milesburton