milesburton/Arduino-Temperature-Control-Library #159: WaitForConversion example gives wrong temperature with DS18B20 but not DS18S20

marcmerlin, 2020-03-26:
I have dallastemperature 3.8.0 on an ESP8266
Just like Paul's OneWire library, this library decodes correctly on DS18S20, but not DS18B20 (other bug https://github.com/PaulStoffregen/OneWire/issues/91 )

Using WaitForConversion, with a single DS18S20 sensor, I see:
```
Before blocking requestForConversion
After blocking requestForConversion
Time used: 752
Temperature: 19.44
Before NON-blocking/async requestForConversion
After NON-blocking/async requestForConversion
Time used: 2
Temperature: 19.50
```
This is the correct temperature for the room as measured by an DS18B20 plugged into my linux PC via a DS2490

However, if I plug a DS18B20, I get wrong temperatures:
```
Before blocking requestForConversion
After blocking requestForConversion
Time used: 752
Temperature: 27.69
Before NON-blocking/async requestForConversion
After NON-blocking/async requestForConversion
Time used: 2
Temperature: 27.69
```
Any ideas?

For good measure, I tried to change the resolution between 9 and 12 and it didn't help
```
  sensors.begin();
  sensors.setResolution(resolution);
  Serial.print("Resolution: ");
  Serial.println(sensors.getResolution());
```

marcmerlin, 2020-03-26:
According to the other bug, this seems to be ESP8266 specific. On ESP32, I'm getting correct enough data with the DS18B20 sensor.
Before blocking requestForConversion
After blocking requestForConversion
Time used: 752
Temperature: 21.87
Before NON-blocking/async requestForConversion
After NON-blocking/async requestForConversion
Time used: 2
Temperature: 21.87

I assume both libraries are probably using the same underlying code or algorithm to read the temp, and that code fails for unknown reasons on ESP8266, but only for DS18B20 ?

marcmerlin, 2020-03-26:
Last updates are in https://github.com/PaulStoffregen/OneWire/issues/91 given that it works with multiple sensors at the same time and handily logs the type, serial, and temp of each sensor.
If you read the updates, the problems seem worse on ESP8266 and with DS18B20.
If you are not able to reproduce, I won't get offended if you close this :)

RobTillaart, 2020-03-26:
@marcmerlin @PaulStoffregen

Dallas Temperature uses a onewire library to communicate to the sensors, see line 25 of DallasTemperature.h  So your conclusion could be right, however there are other possible explanations.

1. Do you have right pull up resistor?

2. Have you checked for counterfeit sensors? These are discussed here  - https://github.com/cpetrich/counterfeit_DS18B20/

3. The ESP8266 is a 3V3 device and the DS18B20 should be able to operate at 2V8. However I've heard of a voltage problem in the past, so if you try a voltage converter in between?

Please let us know

PaulStoffregen, 2020-03-26:
As I mentioned on https://github.com/PaulStoffregen/OneWire/issues/91, I want to see this reproducible on a non-ESP board.

marcmerlin, 2020-03-26:
@RobTillaart thanks for https://github.com/cpetrich/counterfeit_DS18B20/ 
I had a read, and given that my ROM numbers are
ROM = 28 D8 13 6 2 0 0 F
ROM = 28 25 F6 AF 2 0 0 5
ROM = 28 95 7 B0 2 0 0 CB
they do follow the '28-xx-xx-xx-xx-00-00-xx ' pattern, so they are hopefully not clones?

RobTillaart, 2020-03-26:
ROM numbers are easiest to clone I guess. Have you checked the timing of your sensors for 9..12 bit resolution? That is often far harder to clone correctly.

marcmerlin, 2020-03-26:
@RobTillaart 
I forgot to add that I bought those 10 years ago (the dates on them at 2009 and 2010), so maybe they were fewer clones then. I did try setting resolution with your lib (see my original comment) and they returned the resolution number I wrote.

marcmerlin, 2020-03-26:
@RobTillaart about the voltage, I assume it would be a very bad idea to power the sensors on the 3rd pin (non parasitic) using 5V on a 3.3V chip, because it might return 5V level data back?
(ESP8266 can withstand that, but other chips could likely be damaged I assume)

RobTillaart, 2020-03-26:
That's why you should use a bidirectional level converter.

marcmerlin, 2020-03-27:
@RobTillaart  as you probably saw on Paul's bug, my reliability problems came from having he wrong resistor (needed 2.7K for 3.3V instead of 4.7k).
Those sensors are rated to work with 3V, so you don't need level converters.

I got your code working now, but please note:
1) there is no error checking if you put the wrong pin, you just get a bad value 127, with no clue that there is no sensor on that pin

2) it works with one sensor, but if you have more than one on the bus, it fails to return anything
```
Before blocking requestForConversion
After blocking requestForConversion
Time used: 752
Temperature: 21.56

Before NON-blocking/async requestForConversion
After NON-blocking/async requestForConversion
Time used: 2
Temperature: 21.56
```
add 2nd sensor
```
Before blocking requestForConversion
After blocking requestForConversion
Time used: 752
Temperature: 127.94

Before NON-blocking/async requestForConversion
After NON-blocking/async requestForConversion
Time used: 2
Temperature: 127.94
```
Feel free to close if it's working as intended though

RobTillaart, 2020-03-28:
@milesburton   Please close

RobTillaart, 2020-05-03:
@milesburton  please close this issue

closed 2020-05-03 by milesburton