milesburton/Arduino-Temperature-Control-Library #161: Error compiling for board NodeMCU 1.0 (ESP-12E Module)

mudassar791, 2020-04-17:
facing issue following error while compiling
Using library DallasTemperature-3.8.0 at version 3.8.0 in folder: C:\Program Files (x86)\Arduino\libraries\DallasTemperature-3.8.0 
Error
 Error compiling for board NodeMCU 1.0 (ESP-12E Module)

IDE1.8.12
NodeMCU Board Version 2.0.6
[image]

RobTillaart, 2020-04-17:
Can you compile the example sketches that are provided with the library?

castillo92, 2020-05-26:
No one is going to be able to help you if you do not show more code or expand the console panel to see what could happened.

RobTillaart, 2020-05-26:
@milesburton 
As no more info in 35+ days the issue is probably solved, so it can be closed

closed 2020-05-26 by milesburton

mudassar791, 2020-05-26:
solved,