milesburton/Arduino-Temperature-Control-Library #162: requestTemperatures() hangs for no obvious reason

Cubox-, 2020-06-19:
I'm having a very weird issue with the lib. I don't know if that an issue with the lib itself, but I would gladly appreciate the help.
I have tried this behaviour with either only one DS18B20 or multiples. I'm using a Arduino Uno, with the following code:
`#include <Arduino.h>
#define REQUIRESALARMS false
#define REQUIRESNEW false
#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 3
#define TIMEBETWEENREADS 1000
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

const DeviceAddress addresses[9] = {
{0x28, 0x14, 0xef, 0x79, 0xa2, 0, 0x3, 0x4c},
{0x28, 0xb4, 0x70, 0x79, 0xa2, 0, 0x3, 0xc8},
{0x28, 0xd6, 0xba, 0x79, 0xa2, 0, 0x3, 0xfa},
{0x28, 0x1e, 0x8d, 0x79, 0xa2, 0, 0x3, 0x11},
{0x28, 0x3e, 0x83, 0x79, 0xa2, 0, 0x3, 0x5},
{0x28, 0xb1, 0x6e, 0x79, 0xa2, 0, 0x3, 0xfd},
{0x28, 0x9d, 0xd7, 0x79, 0xa2, 0, 0x3, 0x8a},
{0x28, 0xe7, 0xf7, 0x79, 0xa2, 0, 0x3, 0x25},
{0x28, 0x3f, 0x7b, 0x79, 0xa2, 0, 0x3, 0xaa}
};
const double offsets[9] = {
    0.4,
    -0.35,
    0,
    0.2,
    -0.25,
    -0.2,
    0.3,
    0,
    -1.2
};
double temperature[9] = {0, 0, 0, 0, 0, 0, 0, 0, 0};
unsigned int temperatureCount[9] = {0, 0, 0, 0, 0, 0, 0, 0, 0};
unsigned long lastReading = 0;
unsigned long sentDataLast = 0;

void setup() {
    sensors.begin();
    sensors.setResolution(12);
    Serial.begin(115200);
    sentDataLast = millis();      
}

void loop() {
    sensors.requestTemperatures();
    Serial.println("loop");
    delay(250);
    for (unsigned int i = 0; i < 9; i++) {
        double reading = sensors.getTempC(addresses[i]);
        if (reading < 0 || isnan(reading)) {
            continue;
        }
        temperature[i] += reading;
        temperatureCount[i]++;
    }

    if (millis() - sentDataLast >= 10000) {
        for (unsigned int i = 0; i < 9; i++) {
            if (temperatureCount[i] > 0 && !isnan(temperature[i])) {
                Serial.print((temperature[i] / temperatureCount[i]) + offsets[i]);
                Serial.print(" ");
                temperature[i] = 0;
                temperatureCount[i] = 0;
            }
        }
        Serial.println("");
        sentDataLast = millis();
    }
}`

This, for me, will hang at the requestTemperature, and never execute the Serial.println after it.
If you **remove** the Serial.println line, the program works fine and the temperatures are listed fine.

I can't find why and it's turning me mad. I have checked the hardware side of things multiples times, tried to change the code more, but I could only reproduce the problem with that.

What the hell is happening?

RobTillaart, 2020-06-19:
Which version of the library is used?
Which version of the IDE?

**sensors.requestTemperatures();** normally blocks until a ready signal.

1) I see it is the first print statement. Can you add a print statement in setup - Serial.println("setup"); just to see print is working before the request call. I do not expect the problem to be there but excluding causes is important.

2) Comment these two lines - 
```cpp
#define REQUIRESALARMS false
#define REQUIRESNEW false
```
These two lines caused a lot of warnings when trying to compile your code

----
Markup trick: add cpp after the first three back quotes of your code block and you get syntax highlighting. Looks nicer and can help finding some errors.
```
```cpp
```

RobTillaart, 2020-06-19:
Seems this one that cause the compiler warnings
```cpp
#define REQUIRESALARMS false
```

Cubox-, 2020-06-19:
The versions of libraries:
```
|-- <DallasTemperature> 3.8.1
|   |-- <OneWire> 2.3.5
```

I am using VSCode with PlatformIO, which compiles the code.

Cubox-, 2020-06-19:
Alright, you found the issue.
I commented both the REQUIRES defines, and it worked. Something there must be needed for the program to work as intended.
Thanks, truly. I spent hours trying to find out WHY this would happen.

closed 2020-06-19 by Cubox-

RobTillaart, 2020-06-19:
Looked into the lib code and my quick analysis is that you need to setup a handler if you #define REQUIRESALARM . It doesn't matter if you set it to false, it just checks if it is defined or not.
By default it is set to  NULL so if it is called an UNO does sort of warm reboot. YOu could verify this with a print statement in setup followed by a delay(100) to give some time to print the text.

Cubox-, 2020-06-19:
if I use Serial.println in the setup() function, even without delay, it will print the message.