milesburton/Arduino-Temperature-Control-Library #163: get Hex address

guestisp, 2020-06-30:
Any example on how to get the HEX address for the sensor ?

I have the following code:

```
for(int i=0;i<numberOfDevices; i++){
    // Search the wire for address
    if ( TempSensors.getAddress(tempDeviceAddress, i) ) {
      float tempC = TempSensors.getTempC(tempDeviceAddress);

      Serial.print("Temperature for device ");
      Serial.println(i, DEC);

      jsonTemperatures[i]["sensor"] = i;
      jsonTemperatures[i]["temperature"] = tempC;
    }
  }
```

to create a JSON but I need something "fixed" and not an index, when sending the readings to the server, so I can't use the index. How can I get the sensor address, in hex?

RobTillaart, 2020-06-30:
Hi,

tempDeviceAddress is an uint8_t array so you can do create a helper function
to make a hex string of the deviceaddress array like

```cpp
for(int i=0;i<numberOfDevices; i++){
    // Search the wire for address
    if ( TempSensors.getAddress(tempDeviceAddress, i) ) {
      float tempC = TempSensors.getTempC(tempDeviceAddress);

      Serial.print("Temperature for device ");
      Serial.println(i, DEC);

      jsonTemperatures[i]["Address"] = hexString(tempDeviceAddress);  //  <<<<<
      jsonTemperatures[i]["sensor"] = i;
      jsonTemperatures[i]["temperature"] = tempC;
    }
  }

...

char * hexString(uint8_t * addr)
{
  static char txt[20];
  uint8_t j = 0;
  for (uint8_t i = 0; i < 8; i++)
  {
    uint8_t hi = deviceAddress[i] >> 4;
    uint8_t lo = deviceAddress[i] & 0x0F;
    txt[j++} = hi < 10 ? hi + '0' : hi + 'A';
    txt[j++} = lo < 10 ? lo + '0' : lo + 'A';
  }
  txt[j = 0;
  return txt;
}
```

Does this help?

RobTillaart, 2020-07-06:
You can also check the oneWireSearch.ino from the examples

RobTillaart, 2020-08-29:
@milesburton please close

closed 2020-08-29 by milesburton