milesburton/Arduino-Temperature-Control-Library #164: Should we move to an MIT licence?

milesburton, 2020-07-18:
A few weeks ago I received an email from a company in Europe who wanted to use this library in one of their products, however due to the licence LGPL being, and I quote, 'restrictive'

Here's an excerpt of the email I received: "If in general you don’t mind the library being used in commercial products changing the license to MIT might be a bit more appropriate as LGPL is quite restrictive and generally speaking “infectious”, meaning projects using the library need to be licensed under (L)GPL as well, unless explicitly exempted from that."

As this is an open source library with many contributors I am opening it to the floor, do we have any strong feelings one way or the other?

milesburton, 2020-07-18:
@RobTillaart I dropped you an email but it's been about ten years since we spoke outside of github so you may have missed the mail trail. Let me know your thoughts

RobTillaart, 2020-07-18:
Got you email July 7th and replied almost right away. It might have come into your spambox.
But here my reaction again, hope this answers your question.

Good morning MIles,  (10:04 AM here),

I use MIT for all my libraries because it is the most explicit for the not legally educated what is allowed 
and what not, e.g.  https://github.com/RobTillaart/DS18B20_RT/blob/master/LICENSE 

People are allowed to do anything with it, but don't get any warranty. 
The only condition is that they must add the license and copyright notice when distributed.
That last part prevents them from claiming it as their own work. 

MIT does not enforce people in any way to share their improvements / changes / evil additions whatever. 
If you want to "enforce" that GNU GPL3 seems the way to go. Question is how to enforce it? Taking legal action?

I am no expert but some licenses are difficult for commercial use. 
They do not allow you to make changes (even fixes) without permission.
Or you are enforced to open and distribute your own developed (closed) software.

Check - https://choosealicense.com/licenses/ - it explains quite well the difference in licenses for us mere mortals :)
To see them all - https://choosealicense.com/appendix/

I personally go for MIT - "feel free to use it at your own risk and mention the license and copyright (credits) "
In my copyrights I mention the year of start e.g for the DTCL that would be (c) 2012 - 2020 or even earlier?

milesburton, 2020-07-18:
Oh that's odd, yes the email must have been spammed @RobTillaart 

OK with your comments unless there's any strong objections I'll raise a PR to switch to MIT - I'll leave this issue open for a few days so people can comment either way.

RobTillaart, 2020-07-18:
Good move, in the mean time you could inform that company of the steps taken.

RobTillaart, 2020-07-18:
Sent you a test email to see if it arrives.

closed 2020-08-29 by milesburton

bobwolff68, 2021-04-16:
I love that this discussion happened and you decided to move to an MIT license last year. Can someone update the README.md to reflect this as well as the .cpp and .h file? I can submit a pull request if everyone is too busy. I'd love to see this documented as MIT.