milesburton/Arduino-Temperature-Control-Library #168: getDeviceCount still doesn't work with ESP32s (ie, Issue 85 is not really closed)

jackthese, 2020-08-28:
I circled back on Issue 85 (see title), and can verify that .getDeviceCount and the new .getDS18Count always returns zero when using a number of ESP32s (and maybe all of them). This occurs even when the connected sensors work correctly and return valid temps when polled by index or by address. I know there are workarounds, but basically this command does not work and it should, so I'm opening a new issue on the subject. Perhaps this should be remapped into a re-opened Issue 85. Perhaps this should really be an ESP32 issue, but I haven't found anything on their forums on this topic.

RobTillaart, 2020-08-28:
I have a DS18B20 and a ESP32 nearby so I can try to verify the problem #85  ( *note: NR with a # before it creates a link* )

RobTillaart, 2020-08-28:
Tested:
- 2x DS18B20 powered by 3.3 Volts from ESP32 
- used a 4K7 pull up resistor on the DQ pin - if no pull up I get failures
- used example multiple.ino
- changed pinnumber and baudrate in sketch ( to match reality )
- all on a breadboard, wirelength ~10 cm ( 4" )

```cpp
  // locate devices on the bus
  Serial.print("Locating devices...");
  Serial.print("Found ");
  Serial.print(sensors.getDeviceCount(), DEC);
  Serial.println(" devices.");
```

sketch worked well

```
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Dallas Temperature IC Control Library Demo
Locating devices...Found 2 devices.                  <<<<<<<<<<<<<<<<  👍
Parasite power is: OFF
Device 0 Address: 284BDB79020000C9
Device 1 Address: 28C7E379020000E4
Device 0 Resolution: 9
Device 1 Resolution: 9
Requesting temperatures...DONE
Device Address: 284BDB79020000C9 Temp C: 26.00 Temp F: 78.80
Device Address: 28C7E379020000E4 Temp C: 25.50 Temp F: 77.90
Requesting temperatures...DONE
Device Address: 284BDB79020000C9 Temp C: 26.00 Temp F: 78.80
Device Address: 28C7E379020000E4 Temp C: 25.50 Temp F: 77.90
Requesting temperatures...DONE
```

Using library OneWire at version 2.3.5 in folder: ....
Using library DallasTemperature at version 3.8.1 in folder: ....

So cannot reproduce it here. 
- Can you tell more about your setup, pull up , wire length, sketch used?
- Do the sensors work with an UNO?

RobTillaart, 2020-08-28:
redid the test with 1 and 0 devices.
All gave correct nr of devices found..

RobTillaart, 2020-08-28:
.getDS18Count -> added 2 lines in the sketch. Looks OK too

```
Locating devices...Found 2 devices.
getDS18Count: 	2
```

jackthese, 2020-08-28:
Thanks for the quick reply. I believe I found my problem: The only marking on my processor board is "ESP32 DEVKIT1", but I was using NodeMCU 32S as my board, because that is what I searched for and thought I purchased on Amazon recently:
https://www.amazon.com/gp/product/B086MJGFVV/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1
The Amazon listing claims what I bought was compatible with NodeMCU 32S, but clearly it is not fully compatible. When I changed the board type to "ESP Dev Module", I get the correct results, as you did. Still somewhat of a mystery to me why my sketch and hookup had the sensors working fine, but did not execute getDeviceCount correctly. 

So, lesson learned for me, and I appreciate your quick attention on this. Please close this issue if you would. Thanks...Jack

RobTillaart, 2020-08-28:
@jackthese 
you're welcome
As you opened the issue you should see a close button at the bottom

milesburton, 2020-08-28:
> @jackthese
> you're welcome
> As you opened the issue you should see a close button at the bottom

Done. Thanks again Rob

closed 2020-08-28 by milesburton

defl, 2021-01-19:
Same situation here, slight difference though to get it to work. If after search I manually search using the onewire search example, I get the right number of devices. This can be achieved simpler by just calling begin() twice, not very nice but everything beats hardcoding for me :)

sanya7901, 2021-02-22:
I had the same issue with getDeviceCount() on ESP32 and the solution posted [here](https://github.com/PaulStoffregen/OneWire/issues/57#issuecomment-450151098) worked for me.