milesburton/Arduino-Temperature-Control-Library #176: Error while compiling for Seeed Wio LTE Cat 1.

salmanfarisvp, 2020-09-27:
Getting an error while compiling to Seeed Wio LTE Cat 1, In the first, I got error like **‘yield’ was not declared in this scope** , the I commented the **yield()** on the Arduino\libraries\DallasTemperature\DallasTemperature.cpp" lin 543 and 446. 

The try to recompile and got another error. 

```
Using previously compiled file: C:\Users\Faris\AppData\Local\Temp\arduino_build_795906\libraries\OneWire\OneWire.cpp.o
Compiling library "DallasTemperature"
"C:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_discovery_f407 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOD -DERROR_LED_PIN=14 -mcpu=cortex-m4 -DF_CPU=168000000L -DARDUINO=10813 -DARDUINO_WioGpsM4 -DARDUINO_ARCH_SEEED_STM32F4 -std=gnu++11 -DMCU_STM32F406VG -mthumb -DSTM32_HIGH_DENSITY -DSTM32F2 -DSTM32F4 -DBOARD_discovery_f4 -DARDUINO_STM32F4_WIO_GPS -DARDUINO_ARCH_STM32F4 -mthumb -D__STM32F4__ "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino/libmaple" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino/libmaple/usbF4" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino/libmaple/usbF4/STM32_USB_Device_Library/Core/inc" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino/libmaple/usbF4/STM32_USB_Device_Library/Class/cdc/inc" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino/libmaple/usbF4/STM32_USB_OTG_Driver/inc" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino/libmaple/usbF4/VCP" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\cores\\arduino" "-IC:\\Users\\Faris\\AppData\\Local\\Arduino15\\packages\\Seeeduino101\\hardware\\Seeed_STM32F4\\1.2.5\\variants\\discovery_f407" "-IC:\\Users\\Faris\\Documents\\Arduino\\libraries\\OneWire" "-IC:\\Users\\Faris\\Documents\\Arduino\\libraries\\DallasTemperature" "C:\\Users\\Faris\\Documents\\Arduino\\libraries\\DallasTemperature\\DallasTemperature.cpp" -o "C:\\Users\\Faris\\AppData\\Local\\Temp\\arduino_build_795906\\libraries\\DallasTemperature\\DallasTemperature.cpp.o"
C:\Users\Faris\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp: In member function 'void DallasTemperature::blockTillConversionComplete(uint8_t)':
C:\Users\Faris\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:447:3: error: expected primary-expression before '}' token
   } else {
   ^
C:\Users\Faris\Documents\Arduino\libraries\DallasTemperature\DallasTemperature.cpp:447:3: error: expected ';' before '}' token
Using library OneWire at version 2.3.5 in folder: C:\Users\Faris\Documents\Arduino\libraries\OneWire 
Using library DallasTemperature at version 3.9.0 in folder: C:\Users\Faris\Documents\Arduino\libraries\DallasTemperature 
exit status 1
Error compiling for board Wio Tracker LTE
```.

salmanfarisvp, 2020-09-27:
Compilation error is solved by adding a { }

```
// Continue to check if the IC has responded with a temperature
void DallasTemperature::blockTillConversionComplete(uint8_t bitResolution)
{

	if (checkForConversion && !parasite)
	{
		unsigned long start = millis();
		while (!isConversionComplete() && (millis() - start < MAX_CONVERSION_TIMEOUT))
		{
			
		}
		//yield();
	}
	else
	{
		unsigned long delms = millisToWaitForConversion(bitResolution);
		activateExternalPullup();
		delay(delms);
		deactivateExternalPullup();
	}
}
```

closed 2020-09-27 by salmanfarisvp

RobTillaart, 2020-09-27:
You can add an empty function 
```cpp
Void yield()
{}
```

Yield() is used to switch tasks on some boards. If that is not supported by your board the empty function will catch the calls. Yield is used in many libs so expect to see it more often.