milesburton/Arduino-Temperature-Control-Library #179: Please check this tip, it might be interesting to update the library.

rtek1000, 2020-10-18:
`The DS18X20 B7 chip die (found in the DS18B20, DS1820S, and DS1822 products) can experience EEPROM data corruption failures during power on reset.`

`Since the EEPROM holds internal trim values (in addition to the user data in the TH and TL registers) that control the conversion process of the DS18X20 this may show up as inaccuracy of temperature readings. And can cause temperature measurement errors of up to ±60°C.`

Source: [[DS18B20, DS18S20, and DS1822] Potential corruption issues!?](https://forum.arduino.cc/index.php?topic=709615.0)

RobTillaart, 2020-10-18:
@rtek1000 
Looks important
Not read all details yet but will do coming week
- what do propose for the library?
- do you have a sketch to identify 'bad' sensors?
- do you have a sketch to fix a 'bad' sensor?

RobTillaart, 2020-10-18:
found small time slot but no sensor nearby, 
first version to check if the die of a sensor is a B7  (bullet 2)

```cpp
//
//    FILE: test_POR_state.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
// PURPOSE: test for possible Power On Reset bug
//    DATE: 2020-10-18
//     URL: https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/179

/*
  The DS18X20 B7 chip die (found in the DS18B20, DS1820S, and DS1822 products)
  can experience EEPROM data corruption failures during power on reset.

  Since the EEPROM holds internal trim values (in addition to the user
  data in the TH and TL registers) that control the conversion process
  of the DS18X20 this may show up as inaccuracy of temperature readings.
  And can cause temperature measurement errors of up to ±60°C.

  https://forum.arduino.cc/index.php?topic=709615.0
*/

#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS 2
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);

int devCount = 0;
DeviceAddress devAddr;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);
  sensors.begin();

  devCount = sensors.getDeviceCount();
  for (int d = 0; d < devCount; d++)
  {
    Serial.print(d);
    Serial.print('\t');
    if (sensors.getAddress(devAddr, d))
    {
      Serial.println("FAIL");
      continue;
    }
    Serial.print("OK\t");
    printAddress(devAddr);
    Serial.print('\t');
    bool b = isB7die(devAddr);
    Serial.println( b ? "B7" : "C2");

  }
}

void loop()
{
}

///////////////////////////////////////////////////

void printAddress(DeviceAddress deviceAddress)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (deviceAddress[i] < 0x10) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}

bool isB7die(DeviceAddress deviceAddress)
{
  uint32_t val = serial2uint32_t(deviceAddress);
  uint32_t start = 0, end = 0;
  if (deviceAddress[0] == 0x10)  // DS18S20
  {
    start = 0x08978A;
    end   = 0x0CB819;
    return ((start <= val) && (val <= end));
  }
  if (deviceAddress[0] == 0x22)  // DS1822
  {
    start = 0x591D20;
    end   = 0x80885F;
    return ((start <= val) && (val <= end));
  }
  if (deviceAddress[0] == 0x28)  // DS18B20  
  {
    start = 0x545010;
    end   = 0x662B4F;
    return ((start <= val) && (val <= end));
  }
  Serial.println("UNKNOWN TYPE");
  return false;
}

uint32_t serial2uint32_t(DeviceAddress deviceAddress)
{
  uint32_t rv = 0;
  for (uint8_t i = 3; i > 0 ; i--)  // note we need reverse order
  {
    rv <<= 8;
    rv += deviceAddress[i];
  }
  return rv;
}

// -- END OF FILE --

```

RobTillaart, 2020-10-18:
> 3. If on bad list, read the scratch pad (8 bytes)
  a. This tells you if device is in power on reset state, like this:
    Byte 0 = 50h
    Byte 1 = 05h
    Byte 2 = TH
    Byte 3 = TL
    Byte 4 = Config Register
    Byte 5 = FFh
    Byte 6 = 0Ch
    Byte 7 = 10h
    Byte 8 = CRC
4. If device is not in POR state, skip to 8

@rtek1000 
Do I understand correctly that the only way to see if the sensor is in POR mode is that the temperature is 85°C  (50h + 05h)

RobTillaart, 2020-10-18:
Found this link describing trimming process - https://files.elektroda.pl/54966,ds18x20_error.html
it is Polish but there is a text dump of the document.

but the document seems to be from 2004 April 14th so this problem is rather old? 
If so I think the library should not provide a fix (at most a detection of the B7 die and POR state.

rtek1000, 2020-10-18:
I have not yet implemented any treatment for this type of problem. I am making a datalogger, studying how to make a thermostat for a noisy environment. But one of my sensors has already died, in a project with PID to control a gas heater. And recently I bought about 10 sensors, and there are some of them with die B7.

rtek1000, 2020-10-18:
> 
> 
> found small time slot but no sensor nearby,
> first version to check if the die of a sensor is a B7 (bullet 2)
> 
> ```c++
> //
> //    FILE: test_POR_state.ino
> //  AUTHOR: Rob Tillaart
> // VERSION: 0.1.0
> // PURPOSE: test for possible Power On Reset bug
> //    DATE: 2020-10-18
> //     URL: https://github.com/milesburton/Arduino-Temperature-Control-Library/issues/179
> 
> /*
>   The DS18X20 B7 chip die (found in the DS18B20, DS1820S, and DS1822 products)
>   can experience EEPROM data corruption failures during power on reset.
> 
>   Since the EEPROM holds internal trim values (in addition to the user
>   data in the TH and TL registers) that control the conversion process
>   of the DS18X20 this may show up as inaccuracy of temperature readings.
>   And can cause temperature measurement errors of up to ±60°C.
> 
>   https://forum.arduino.cc/index.php?topic=709615.0
> */
> 
> #include <OneWire.h>
> #include <DallasTemperature.h>
> 
> #define ONE_WIRE_BUS 2
> OneWire oneWire(ONE_WIRE_BUS);
> DallasTemperature sensors(&oneWire);
> 
> int devCount = 0;
> DeviceAddress devAddr;
> 
> void setup()
> {
>   Serial.begin(115200);
>   Serial.println(__FILE__);
>   sensors.begin();
> 
>   devCount = sensors.getDeviceCount();
>   for (int d = 0; d < devCount; d++)
>   {
>     Serial.print(d);
>     Serial.print('\t');
>     if (sensors.getAddress(devAddr, d))
>     {
>       Serial.println("FAIL");
>       continue;
>     }
>     Serial.print("OK\t");
>     printAddress(devAddr);
>     Serial.print('\t');
>     bool b = isB7die(devAddr);
>     Serial.println( b ? "B7" : "C2");
> 
>   }
> }
> 
> void loop()
> {
> }
> 
> ///////////////////////////////////////////////////
> 
> void printAddress(DeviceAddress deviceAddress)
> {
>   for (uint8_t i = 0; i < 8; i++)
>   {
>     if (deviceAddress[i] < 0x10) Serial.print("0");
>     Serial.print(deviceAddress[i], HEX);
>   }
> }
> 
> bool isB7die(DeviceAddress deviceAddress)
> {
>   uint32_t val = serial2uint32_t(deviceAddress);
>   uint32_t start = 0, end = 0;
>   if (deviceAddress[0] == 0x10)  // DS18S20
>   {
>     start = 0x08978A;
>     end   = 0x0CB819;
>     return ((start <= val) && (val <= end));
>   }
>   if (deviceAddress[0] == 0x22)  // DS1822
>   {
>     start = 0x591D20;
>     end   = 0x80885F;
>     return ((start <= val) && (val <= end));
>   }
>   if (deviceAddress[0] == 0x28)  // DS18B20  
>   {
>     start = 0x545010;
>     end   = 0x662B4F;
>     return ((start <= val) && (val <= end));
>   }
>   Serial.println("UNKNOWN TYPE");
>   return false;
> }
> 
> uint32_t serial2uint32_t(DeviceAddress deviceAddress)
> {
>   uint32_t rv = 0;
>   for (uint8_t i = 3; i > 0 ; i--)  // note we need reverse order
>   {
>     rv <<= 8;
>     rv += deviceAddress[i];
>   }
>   return rv;
> }
> 
> // -- END OF FILE --
> ```

Good job! It may be interesting that the code performs the CRC check as well. Does the DallasTemperature library already perform this CRC check to validate the data?

RobTillaart, 2020-10-18:
> Good job! It may be interesting that the code performs the CRC check as well. Does the DallasTemperature library already perform this CRC check to validate the data?

Thanks

Yes, **getAddress()** calls **validAddresss()** which does the CRC check, see below.

```cpp
// returns true if address is valid
bool DallasTemperature::validAddress(const uint8_t* deviceAddress) {
	return (_wire->crc8(deviceAddress, 7) == deviceAddress[DSROM_CRC]);
}

// finds an address at a given index on the bus
// returns true if the device was found
bool DallasTemperature::getAddress(uint8_t* deviceAddress, uint8_t index) {
	uint8_t depth = 0;
	_wire->reset_search();
	while (depth <= index && _wire->search(deviceAddress)) {
		if (depth == index && validAddress(deviceAddress))
			return true;
		depth++;
	}
	return false;
}
```

in my implementation **isB7die()** I use only 3 bytes as serial signature
Technically I could use 6 and use uint64_t values. (no big change)
The serial is 48 bit according to datasheet.

opinion?

rtek1000, 2020-10-18:
> Yes, **getAddress()** calls **validAddresss()** which does the CRC check, see below.

Nice!

As to how this problem is handled, I was also a little confused. From what I understand, it is suggested that the sensor data be saved for future comparison, that means having to save it on the Arduino's EEPROM (or other uC) which can complicate things a bit.

> ...recommended solution is to read known good trim values once and then store in either EEPROM or in an initialization file for later use. During actual operations the procedure would then be something like the following:
> 
> Normal operational flow procedure for handling B7 versions of the DS18X20
> 1. Compare serial number of device with bad list
> 
> 2. If device serial number not on list, code normally... skip to number 8
> 
> 3. If on bad list, read the scratch pad (8 bytes)
>   a. This tells you if device is in power on reset state, like this:
>     Byte 0 = 50h
>     Byte 1 = 05h
>     Byte 2 = TH
>     Byte 3 = TL
>     Byte 4 = Config Register
>     Byte 5 = FFh
>     Byte 6 = 0Ch
>     Byte 7 = 10h
>     Byte 8 = CRC
> 
> 4. If device is not in POR state, skip to 8
> 
> 5. Read the DS18X20 Trim values
> 
> 6. Check Trim data against known good data for the particular chip
> 
> 7. If not same as saved data, Write new Trim values, etc., (you may wish to keep track of number of times you have updated the EEPROM so if needed you could replace chip when exceeding the lifetime of the EEPROM.)
> 
> 8. Do regular temperature conversion sequence

rtek1000, 2020-10-18:
I thought about working more radically and rejecting the sensor with die B7 in my projects. But that implies Russian roulette, because I can't choose which sensor the vendor is selling.

RobTillaart, 2020-10-18:
> I thought about working more radically and rejecting the sensor with die B7 in my projects. But that implies Russian roulette, because I can't choose which sensor the vendor is selling.

Use multiple resellers could be an option, (Purchase approach)
Or if 40% of the sensors is B7 on average you could just increase the price to pay for the B7 sensors. (Sales approach)
Or use triple sensors so if you detect POR on one sensor, you still have two fallback (NASA approach)

RobTillaart, 2020-10-18:
Some thoughts about POR detection

- detect 85°C after powerup
- write to TH or TL and read it back 
- change resolution and read it back?

assumption of the latter two is that the sensor is "frozen" @ 85°C   so cant be read back correctly? Would that work?

@rtek1000 
do you have a sensor that failed due to this bug to test :  
- detection, recognition
- solutions.

RobTillaart, 2020-10-18:
sort of POR checking - not tested 
```cpp
int checkSensors(int expectedDevices)
{
  int error = 0;
  sensors.requestTemperatures();
  delay(750);
  devCount = sensors.getDeviceCount();
  if (devCount != expectedDevices)
  {
    return -1;  // not all sensors seen...
  }
  
  for (int d = 0; d < devCount; d++)
  {
    if (sensors.getAddress(devAddr, d))
    {
      error |= 0x01;  // not reachable 
      continue;
    }
    if (isB7die(devAddr))
    {
      if (sensors.getTempC(devAddr) == 85.00) // not 100% sure as it may be 85°C
      {
        error |= 0x02;  // POR flag
      }
    }
  }
  return error;
}
```

rtek1000, 2020-10-18:
I'm going to look for the damaged sensor, but I think I already discarded that sensor.

I contacted the manufacturer at that time (Fev/20), I will quote the conversation below:

Support:
> Thank you for contacting with Maxim Technical Support.
> 
> According to your description, it seems that you can read the ROM code, that means DS18B20 received the Read ROM command correctly and sent out the ROM code even if the ROM code you get is not correct.  Do you power DS18B20 with 3.3V power supply or using parasite power? if you are using parasite power, I recommend you to test the failure part with 3.3V power supply to power the part and check if this can make it work normal? if it still can not work normal, then the device should be failure, you can contact with your supplier to request failure analysis, Maxim can provide the failure analysis for the devices from Maxim.
> 
> the reserved registers can not cause the device failure, ROM ID is constant and never changed, if you can not read the ROM ID correctly, that means there is something wrong during communication, alternatively, you can remove the 100ohm protection resistor and change the 4k7 pull up to 2.2k or 1k to check if this can read ROM ID correctly on the failure part? what you can reconfigure the DS18B20 is to write the configuration register to change the resolution  and this will affect the conversion time.

Me:
> I changed the pullup resistor from 4k7 to 2k2 and replaced the power supply from 3v3 to 5v. The uC was replaced from the STM32 to the Atmega328P (Arduino). The sample code firmware was found at the link: https://www.instructables.com/id/Get-the-Serial-Number-of-a-DS18B20-With-an-Arduino/.
> 
> The result was the same as the STM32 uc has received (same ID data). Including the temperature that was different from a good sensor. The other 8 sensors are around 24°C and the bad sensor indicates 33.5°C. When holding two sensors (one good and one bad) by hand, the good sensors register the temperature of 33°C and the bad sensor indicates 42.5°C.
> 
> Arduino + 'Single' sketch:
> 
>     Parasite power is: OFF
>     Address: 7B7C38287DFF9FAF
>     Resolution: 9
>     Temp C: 33.50 Temp F: 92.30
>     ...
>     Temp C: 42.50 Temp F: 108.50
> 
> 
>     Resolution: 9; Temp C: 31.50 Temp F: 88.70
>     Resolution: 10; Temp C: 31.75 Temp F: 89.15
>     Resolution: 11; Temp C: 31.62 Temp F: 88.93
>     Resolution: 12; Temp C: 31.69 Temp F: 89.04 
> 
> 
> Arduino + 'WaitForConversion2' sketch:
> 
>     Dallas Temperature Control Library - Async Demo
>     Library Version: 3.7.9
>       Resolution: 9
>     Idle counter: 92 (ms)
>       Resolution: 10
>     Idle counter: 185 (ms)
>       Resolution: 11
>     Idle counter: 369 (ms)
>       Resolution: 12
>     Idle counter: 738 (ms)
> 
> 
> It really looks like this sensor has had its core damaged. The question is how it happened to prevent this from happening to other sensors in the future.

Support:
> According to your test result, it seems the communication section and temperature conversion section are active but get wrong data, so I would like to know if you use the same code on STM32 and on ATmel MCU?  As you can get the ROM ID but the return value is not correct, so I suggest you to  write the TH or TL with any value and read it back to verify that the read and write operation are correct on the failure part, if the readback value is differ from the write value, that maybe the timing issue for this DS18B20, then you can adjust the read time instant to see if this can help to resolve the issue. also you can monitor the waveform of readback data to verify if the response waveform is exact the same as the data readback?
> 
> If you still have questions on this, please feel free let me know.

As I didn't have the time, I ended up thinking it was an isolated case and focused on the other parts of the project, so I don't have any more data on this case.

rtek1000, 2020-10-19:
I was reviewing the codes of my sensors and noticed that I was observing incorrectly, It might be interesting to reverse the order of the data in the example sketch.

[https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Single/Single.ino](https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Single/Single.ino)

From (28 275C799 71203 6F):
```C++
// function to print a device address
void printAddress(DeviceAddress deviceAddress)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (deviceAddress[i] < 16) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}
```
To (6F 03129 7795C27 28):
```C++
// function to print a device address
void printAddress(DeviceAddress deviceAddress)
{
  for (int8_t i = 7; i >= 0; i--)
  {
    if (deviceAddress[i] < 16) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}
```

> 0xCC00000SSSSSSSFF
> where CC is the CRC 00000 is the customer code SSSSSSS is the serial number FF is the family code

RobTillaart, 2020-10-19:
The order is discussed more than once - here and on arduino.cc forum. 
When reading the unique ROM code (a.k.a. address) you get the LSB first and the CRC as last one.
Feel free to print them in reverse or right or whatever order but I expect this library won't change

Solution is to have a flag MSBFIRST LSBFIRST to determine the printing of the address.

RobTillaart, 2020-10-19:
Think it is better to have the test for POR state function just work on one address, as that makes it more explicit.

```cpp
bool testPOR(DeviceAddress addr)
{
  if  (sensors.getTempC(addr) == 85.00) // not 100% sure as it may be 85°C
  {
    if  (idB7die(addr) )
    {
      // additional tests here?
      return true;
    }
  }
return false;
}
```

RobTillaart, 2020-10-22:
@rtek1000 
Do you think there are ways to improve the **testPOR()** function?

rtek1000, 2020-10-29:
> 3\. If on bad list, read the scratch pad (8 bytes)
>         a. This tells you if device is in power on reset state, like this:
>         Byte 0 = 50h
>         Byte 1 = 05h
>         Byte 2 = TH [...]

```
**Solution to the 85 °C-Problem

There is a simple, undocumented, way to discriminate between the power-up 85 °C-reading and a genuie temperature reading of 85 °C in DS18B20 of Family A [5]: <byte 6> of the scratchpad register. If it is 0x0c, then the 85 °C-reading is a power-up reading, otherwise it is a true temperature measurement.**

Source: [https://github.com/cpetrich/counterfeit_DS18B20](https://github.com/cpetrich/counterfeit_DS18B20)
```

rtek1000, 2020-10-29:
It may be interesting to have a way to identify a fake chip and a potentially true chip, see the information in the link mentioned above.

(https://github.com/cpetrich/counterfeit_DS18B20)

rtek1000, 2020-10-29:
> Family A1 is the authentic Maxim-produced DS18B20 (C4 die). All other families are clones. Note the similarities between the dies of Families D1 and D2 (consistent with their similarity in software) and the signifcant differences between Families B1 and B2 (as opposed to their similarity in software).
> Family A1: Authentic Maxim DS18B20
> 
> Obtained no probes containing these chips on ebay or AliExpress in 2019, but obtained chips from a few vendors in 2019
> 
>     ROM pattern [5]: 28-tt-tt-ss-ss-00-00-crc
>     Scratchpad register: (<byte 0> + <byte 6>) & 0x0f == 0 after all successful temperature conversions, and 0x00 < <byte 6> <= 0x10 [2,3,5]. I.e., <byte 6> = 0x10 – (<byte 0> & 0x0f).
>     According to current behavior [5] and early datasheets [9], the power-up state of reserved <byte 6> in the Scratchpad register is 0x0c.
>     Returns "Trim1" and "Trim2" values if queried with function codes 0x93 and 0x68, respectively [4]. The bit patterns are very similar to each other within a production run [4]. Trim2 is currently less likely to equal 0xff than Trim1 [5]. Trim2 was 0xDB or 0xDC since at least 2009, and has been 0x73/0x74 since the fall of 2016 (all with C4 die) [5]. (In parasitic power-only chips, Trim2 is 0xDB or 0xDC as of 2020.)
>         Trim1 and Trim2 encode two parameters [5]. Let the bit pattern of Trim1 be [t17, t16, t15, t14, t13, t12, t11, t10] (MSB to LSB) and Trim2 be [t27, t26, t25, t24, t23, t22, t21, t20]. Then,
>             offset parameter = [t22, t21, t20, t10, t11, t12, t13, t14, t15, t16, t17] (unsigned 11 bit-value) [5], and
>             curve parameter = [t27, t26, t25, t24, t23] (unsigned 5 bit-value) [5].
>         Within a batch, the offset parameter seems to spread over 20 to 30 units while all sensors within the batch share the same curve parameter [5].
>         The offset parameter shifts the temperature output over a range of approx. 100 °C (0.053 °C per unit), while the curve parameter shifts the temperature over a range of 3.88 °C (0.12 °C per unit), at least in current versions of the chip [5]. Example values of 2019 are offset = 0x420 and curve = 0x0E, i.e. they lie pretty central within their respective ranges.
>     Temperature offset of current batches (2019) is as shown on the Maxim FAQ page, i.e. approx. +0.1 °C at 0 °C [6] (i.e., not as shown on the datasheet [1,9]. The plot on the datasheet stems from measurements at the time of introduction of the sensor 10+ years ago [5,10].). Very little if any temperature discretization noise [5].
>     Polling after function code 0x44 indicates a spread of 584-615 ms between sensors for a 12-bit temperature conversion at room temperature [5]. Conversion time is easily repeatable for individual chips. Lower resolutions cut the time in proportion, i.e. 11 bit-conversions take half the time. The trim parameters affect the conversion time.
>     Returns a power-up temperature of 85 °C if scratchpad register is read before temperature conversion has completed (in both normal and parasitic power modes). [5].
>     It appears the chip returns a temperature of 127.94 °C (=0x07FF / 16.0) if a temperature conversion was unsuccessful [5] (e.g. due to power stability issues which arise reproducibly in "parasitic power" mode with multiple DS18B20 if Vcc is left floating rather than tied to ground. Note that the datasheet clearly states that Vcc is to be tied to GND in parasitic mode.).
> 
>     Example ROM: 28-13-9B-BB-0B -00-00- 1F
>     Initial Scratchpad: 50/05/4B/46/7F/FF/0C/10/1C
>     Example topmark: DALLAS 18B20 1932C4 +786AB
>     Indent mark: P (date codes 1150 through 2019)
>     Indent mark: possibly options besides P since 2020 (cf. Issue 21) (2020)
> 
> Family A1-Fishy: Authentic but Stolen?
> 
> Obtained no probes containing these chips on ebay or AliExpress in 2019, but obtained chips from one vendor in 2019
> 
> If I were to make a wild guess I would say these chips were diverted somewhere toward the end of the Maxim production pipeline (stolen?) [5]. Fun fact: the vendor advertised these chips as QT18B20, so as sold these were actually counterfeits of DS18B20 clones. These chips are marked as produced in Thailand rather than Philippines.
> 
>     ROM pattern [5]: 28-tt-tt-Cs-03-00-00-crc
> 
> The chips follow the description of Family A1 above with the following exceptions [5]:
> 
>     Both alarm registers are set to 0x00 (scratchpad bytes 2 and 3).
>     The conversion resolution is set to 9 bits (i.e., both configuartion bits are 0).
>     Both trim values are 0x00, resulting in wrong temperatures (i.e., very low) and conversion times in the range of 400 to 500 ms.
>         Once trim values are set to something reasonable, the time for temperature conversion is within the range specified for Family A1 above.
> 
>     Example ROM: 28-9B-9E-CB-03 -00-00- 1F
>     Initial Scratchpad: 50/05/00/00/1F/FF/0C/10/74
>     Example topmark: DALLAS 18B20 1136C4 +957AE
>     Example topmark: DALLAS 18B20 1136C4 +957AF
>     Example topmark: DALLAS 18B20 1136C4 +152AE
>     Example topmark: DALLAS 18B20 1136C4 +152AF
>     Example topmark: DALLAS 18B20 1136C4 +152AG
>     Example topmark: DALLAS 18B20 1136C4 +152AI
>     Indent mark: THAI <letter>
> 
> Family A2: Good Clone with Poor Temperature Performance
> 
> Obtained no probes containing these chips on ebay or AliExpress in 2019, but obtained chips from one vendor in 2019
> 
> These chips are not Maxim-produced.
> 
[... cut ...]
>     Example ROM: 28-DF-54-56 -B5- 01-3C-F5 (2020)
>     Example ROM: 28-AF-EC-07 -D6- 01-3C-0A (2020)
>     Initial Scratchpad: 90/01/55/05/7F/xx/xx/66/xx
>     Example topmark: DALLAS 18B20 1812C4 +051AG
>     Example topmark: DALLAS 18B20 1827C4 +051AG
>     Example topmark: DALLAS 18B20 1916C4 +051AG
>     Example topmark: DALLAS 18B20 1923C4 +051AG
>     Example topmark: DALLAS 18B20 1943C4 +051AG
>     Example topmark: DALLAS 18B20 1828C4 +233AA
>     Example topmark: DALLAS 18B20 2008C4 +817AB (2020)
>     Indent mark: none
>

rtek1000, 2020-10-29:
[image]
https://raw.githubusercontent.com/cpetrich/counterfeit_DS18B20/master/images/DS18B20_die_collage_authentic_and_clones_200405.jpg

rtek1000, 2020-10-29:
[image]

milesburton, 2020-10-29:
Now that is rather interesting, great discovery. I’ll have to have a dig
when I get some free time.

RobTillaart, 2020-10-29:
Indeed lots of information, no time to dive into the details soon. 
I propose that these tools / analysis need a library of its own. That way the tools can be developed / matured in its own pace without "disturbing" this library.

Your opinion?

RobTillaart, 2020-10-29:
> ....
>     Byte 0 = 50h
>     Byte 1 = 05h
> ...
>     Byte 6 = 0Ch

POR detection should be a part of a more generic **getDeviceStatus()** function that can be extended over time if needed.

```cpp
int getDeviceStatus(DeviceAddress addr)
{
  ScratchPad sp;
  if (!isConnected(addr, sp)) return DS18_NOT_CONNECTED;
  
  if ((sp[0] == 0x50) && (sp[1] == 0x05) && (sp[6] == 0x0C))
    return DS18_POWER_ON_RESET;

  return DS18_OK;
}
```

As this is lightweight it does not affect the size/complexity of the library much.

milesburton, 2020-10-29:
I've had a brief read of @cpetrich's article and I do wonder if this curious quirk is out of scope for this library, or at least as @RobTillaart suggested maybe some sort of fork/superset. We know this library is used commercially so there is merit that we detect anomalies, however if that capability is added to the core library it'll eat up valuable rom for those hobbyist's who may never use that code.

...On that note, wouldn't tree shaking be so useful in the MCU world. 

Maybe there's value creating a 'pro' version of the library?

RobTillaart, 2020-10-29:
> ... hobbyist's who may never use that code.

good point, hobbyist will indeed seldom use this and I do not recall that this problem discussed a lot e.g. on the Arduino forum.

> Maybe there's value creating a 'pro' version of the library?

The functionality we are talking about is "a set of diagnostic functions to detect problems with sensors and functions to fix these". 

Question is how easy people can use these functionality?

Some thoughts:
1. build new functions into current lib, 
	- compiler decides what is included
	- maintain 1 lib
		- much iterations esp during initial phase.
	- interface of the lib grows a lot (problem for beginner / medium users ?)
1. build new functions into a PRO lib
	- user decides which version to include
	- maintain 2 libs - keeping them 100% compatible needs constant attention
		- PRO as derived class from BASIC ?
	- current BASIC lib stays stable - only PRO lib interface will grow / be dynamic initially
1. a separate (disjunct) library
	- user decides to include extra lib e.g.  "dallasHelpers.h", 
	- maintain 2 libs
		- no (100%) compatibility requirement
		- lib could also have some other supportive functions like **printAddress(addr)**
	- current BASIC lib stays stable - only new lib interface will grow / be dynamic initially

Development wise I prefer option 3 as that is the easiest. 
Some functions may be implemented too in the BASIC lib if they are stable and add value 

to be continued...

milesburton, 2020-10-29:
#3 is actually the most appealing, I was going to suggest a macro but that's a bit hacky. 

We should reach out to the maintainer and see whether they mind us incorporating their code into ours. Hopefully they saw the tag I added above

RobTillaart, 2020-10-29:
https://github.com/cpetrich/counterfeit_DS18B20/blob/master/LICENSE 
Apache 2 license ==> , so one need to reference the original source, copyright, license, and tell what are the changes made. 

see also - https://choosealicense.com/licenses/

rtek1000, 2020-10-30:
🤔 "3- A separate (disjunct) library"?

Interesting. I think it would be ideal if the additional library functions as an external library dependent on the pre-existing one.

Just like this library (DallasTemperature) depends on the Wire library.

> // Include the libraries we need
> #include <OneWire.h>
> #include <DallasTemperature.h>

As for the license issue, I don't know much about it, so I don't have much to add, sorry. 😑

rtek1000, 2020-11-06:
Hi, I was looking at the setResolution function, and I would like to confirm, does this function read the current status before writing the EEPROM? Or does it just burn the EEPROM?

If I'm not mistaken, in the Arduino EEPROM library, there is a function called Update, which avoids overwriting the memory, if it is already with the same value.

I find it interesting to be careful not to overwrite the EEPROM too often without need, if the user leaves the code in an infinite loop, what do you think?

rtek1000, 2020-11-06:
> EEPROM Update
> The microcontroller on the Arduino boards have 512 bytes of EEPROM: memory whose values are kept when the board is turned off (like a tiny hard drive).
> 
> The purpose of this example is to show the EEPROM.update() method that writes data only if it is different from the previous content of the locations to be written. This solution may save execution time because every write operation takes 3.3 ms; the EEPROM has also a limit of 100,000 write cycles per single location, therefore avoiding rewriting the same value in any location will increase the EEPROM overall life.

https://www.arduino.cc/en/Tutorial/LibraryExamples/EEPROMUpdate

RobTillaart, 2020-11-06:
@rtek1000 
The Dallas library does not use the Arduino EEPROM functions afaik, (can you indicate which line?)
so I do not understand how it is related to this topic,
If it is a separate topic, that need to be fixed, please open a new issue for it

wmarkow, 2021-02-28:
>Found this link describing trimming process - https://files.elektroda.pl/54966,ds18x20_error.html
it is Polish but there is a text dump of the AN247 document.

Attaching the extracted PDF version. Maybe it will help.
[AN247_DS18x20_EEPROM_Corruption_Issue.pdf](https://github.com/milesburton/Arduino-Temperature-Control-Library/files/6056605/AN247_DS18x20_EEPROM_Corruption_Issue.pdf)

closed 2022-06-03 by milesburton