milesburton/Arduino-Temperature-Control-Library #180: yield function

NvMfgEngr, 2020-10-31:
Hi:

Just started working with the Arduino UNO PCB.  
Today's task was to get the 18B20 working.  
But compiling generates the error :

File  :  DallasTemperature.cpp
Function  :    void  DallasTemperature()
"yield()     was not declared"

Error occurs in 2 location:
Lines   446:13   and    543:11

Library was just downloaded off of github.

What is your fix?

Doug
NV Mfg Engr
nvmfgengr@gmail.com

RobTillaart, 2020-11-01:
Hi solution is discussed in issue #177

NvMfgEngr, 2020-11-02:
Installing the new version of Arduino IDE fixed the "yield()" issue.
Fix was stated in Issue #107. 
But what was misleading the IDE I was using was version 2.1.5.   Where I acquired that IDE I do not know.  But version 1.8.5  dated 10/6/2020 is working.
Doug

RobTillaart, 2020-11-02:
@NvMfgEngr 
you may close the issue

milesburton, 2020-11-02:
I'll do the honours @RobTillaart

closed 2020-11-02 by milesburton